openwrt/openwrt #10532: Move packages to openwrt/openwrt to fix warnings?

PolynomialDivision, 2022-08-28:
Currently OpenWrt experiences warnings:
```
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
```

- busybox and policycoreutils depend on libpam resulting in warnings (cc @stintel)
- kexec-tools depend liblzma (no Maintainer)
- lldpd depend on libnetsnmp (cc @stintel)

ynezz, 2022-08-28:
> Move packages to openwrt/openwrt to fix warnings?

IMO this is not a proper fix. What about fixing `scripts/package-metadata.pl` (or some other relevant parts) instead? That warning message is little bit misleading, there is no such dependency unless you've that specific config knob enabled.

PolynomialDivision, 2022-08-28:
> That warning message is little bit misleading, there is no such dependency unless you've that specific config knob enabled.

But if you enable it, you have the dependency from openwrt/openwrt to openwrt/packages. I thought openwrt main repository would be structured like a hierarchical tree that has no dependencies to its leafs.

ynezz, 2022-08-28:
> I thought openwrt main repository would be structured like a hierarchical tree that has no dependencies to its leafs.

Indeed, and AFAIK it works like that, you don't need any external dependency for default and core configuration. It builds and its usable without any other feeds.

> But if you enable it, you have the dependency from openwrt/openwrt to openwrt/packages.

IMO this is acceptable given current build system architecture and layout.

PolynomialDivision, 2022-08-28:
Okay, than lets fix the scripts. :)