openwrt/openwrt #10534: OpenWrt 22.03.0-rc6  DHCPv6 client bug

f4nff, 2022-08-28:
[https://openwrt.anycast.xyz/releases/22.03.0-rc6/targets/rockchip/armv8/openwrt-22.03.0-rc6-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz](https://openwrt.anycast.xyz/releases/22.03.0-rc6/targets/rockchip/armv8/openwrt-22.03.0-rc6-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz)

Obtain ip through slaac, but often encounter the situation of ipv6 empty route.**Do nothing, wait a few minutes and it will automatically recover.**
However, windows10 on the same network segment has always been normal, and there is no link to ipv6.

I tried adding ipv6 static routing, but that didn't solve the problem.

[image]

In addition, windows 10 can obtain two ipv6 addresses and one temporary address, but the WAN6 DHCPv6 client can only obtain one ipv6 address.

f4nff, 2022-08-28:
[image]

Use default gateway

If unchecked, no default route is configured

This seems to be missing intermittently.

f4nff, 2022-08-30:
The problem routing table is obtained by tracing as follows:

```
root@OpenWrt:~# ip -6 route
default dev tun0  metric 1024
2409:8220:9a14:b0b1::/64 dev eth0  metric 256
unreachable 2409:8220:9a14:b0b1::/64 dev lo  metric 2147483647
unreachable 2409:8220:9a15:281::/64 dev lo  metric 2147483647
unreachable 2409:8220:9a17:f691::/64 dev lo  metric 2147483647
unreachable 2409:8220:9a18:5641::/64 dev lo  metric 2147483647
unreachable 2409:8220:9a18:5ab1::/64 dev lo  metric 2147483647
unreachable 2409:8220:9a18:5d61::/64 dev lo  metric 2147483647
fe80::/64 dev br-lan  metric 256
fe80::/64 dev eth0  metric 256
fe80::/64 dev tun0  metric 256
default via fe80::68b5:b7ff:fe09:5ce4 dev eth0  metric 512
anycast fe80:: dev br-lan  metric 0
anycast fe80:: dev eth0  metric 0
anycast fe80:: dev tun0  metric 0
multicast ff00::/8 dev br-lan  metric 256
multicast ff00::/8 dev eth0  metric 256
multicast ff00::/8 dev tun0  metric 256
```

The normal routing table that can be connected to the Internet is as follows:
```

root@OpenWrt:~# ip -6 route
default dev tun0  metric 1024
unreachable 2409::8220:9a15:281::/64 dev lo  metric 2147483647
2409::8220:9a17:f691::/64 dev eth0  metric 256
unreachable 2409::8220:9a17:f691::/64 dev lo  metric 2147483647
unreachable 2409::8220:9a18:5641::/64 dev lo  metric 2147483647
unreachable 2409::8220:9a18:5ab1::/64 dev lo  metric 2147483647
unreachable 2409::8220:9a18:5d61::/64 dev lo  metric 2147483647
fe80::/64 dev br-lan  metric 256
fe80::/64 dev eth0  metric 256
fe80::/64 dev tun0  metric 256
default via fe80::68b5:b7ff:fe09:5ce4 dev eth0  metric 512
anycast 2409::8220:9a17:f691:: dev eth0  metric 0
anycast fe80:: dev br-lan  metric 0
anycast fe80:: dev eth0  metric 0
anycast fe80:: dev tun0  metric 0
multicast ff00::/8 dev br-lan  metric 256
multicast ff00::/8 dev eth0  metric 256
multicast ff00::/8 dev tun0  metric 256
```

By comparison, it can be concluded that when the network cannot be connected, it is lost:
`anycast 2409::8220:9a17:f691:: dev eth0  metric 0`

zorun, 2022-09-08:
Please describe in clear terms what problem you are facing. "ipv6 empty route" is not clear. Is your OpenWrt device itself losing ipv6 connectivity? What are the symptoms? Are other devices connected to the LAN of your OpenWrt device losing ipv6 connectivity?

Also, your setup looks very custom: you have a lot of different IPv6 prefixes, and two default routes. Make sure to reproduce the problem with a more "standard" configuration.