openwrt/openwrt #10538: dsa cannot use for bridge vlan changer

ptpt52, 2022-08-29:
top:
```
client---vlan20----Router(bridge forward)----vlan30----Server
```

on swconfig platform, I could do this by setup an `br-lan(eth0.20 eth0.30)` and set vlan on switch to archive this.

but how to archive the same goal on dsa platform?

ptpt52, 2022-08-29:
on dsa:
`br-lan( lan1.20 lan2.30)` not works

ptpt52, 2022-08-29:
I sent this issue to show some shortcomings of dsa
I think dsa implements a subset of switch device functionality
But what swconfig driver implements is a complete set

john-tho, 2022-09-03:
> on dsa: `br-lan( lan1.20 lan2.30)` not works

Are you not able to create an additional bridge with the vlans of those two ports included?
```
config interface 'translate'
        option type 'bridge'
        option ifname 'lan1.20 lan2.30'
        option bridge_empty '1'
```

ptpt52, 2022-09-03:
> > on dsa: `br-lan( lan1.20 lan2.30)` not works
> 
> Are you not able to create an additional bridge with the vlans of those two ports included?
> 
> ```
> config interface 'translate'
>         option type 'bridge'
>         option ifname 'lan1.20 lan2.30'
>         option bridge_empty '1'
> ```

bridge create ok, but it would not work.

john-tho, 2022-09-03:
> bridge create ok, but it would not work.

Okay, yes, with that config, the bridge is created, with the correct members, but no traffic through it. I see on tcpdump ping reaches the inbound port.vlan, and bridge, but not the outbound port.vlan

---

I poked around with a bridge-vlan config and think I got the inter-vlan / vlan translation I wanted.

Moved from multiple bridges, to vlan-filtering aware single bridge, with a translation bridge atop bridge vlans, rather than the individual ports.
This is my full `/etc/config/network`:
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fdad:8aa3:610b::/48'

config interface 'lan'
	option device 'br0.lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'br0.wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'br-wan'
	option proto 'dhcpv6'

config device
	option name 'br0'
	option type 'bridge'
	option vlan_filtering '1'
	list ports 'wan'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'lan5'
	list ports 'sfp'

config bridge-vlan
	option device 'br0'
	option vlan '1'
	option alias 'lan'
	list ports 'lan2:*'
	list ports 'lan3:*'
	list ports 'lan4:*'
	list ports 'lan5:*'

config bridge-vlan
	option device 'br0'
	option vlan '2'
	option alias 'wan'
	list ports 'sfp:*'
	list ports 'wan:*'

config bridge-vlan
	option device 'br0'
	option vlan '20'
	list ports 'wan:t'

config bridge-vlan
	option device 'br0'
	option vlan '30'
	list ports 'lan2:t'

# make netifd bringup br0.20
#config interface 'v20'
#	option device 'br0.20'
#	option proto 'none'

# make netifd bringup br0.30
#config interface 'v30'
#	option device 'br0.30'
#	option proto 'none'

# classic bridge for vlan translation
config interface 'translate'
	option type 'bridge'
	option ifname 'br0.20 br0.30'
	option bridge_empty '1'
```

See if _something like that_ works?
I only used the `#config interface 'br0.20'` in testing, and seem to be working fine after reboot without them now.

---

I had `ip-full`, `ip-bridge`, and `tcpdump` installed to poke around with:

```
brctl show
bridge name	bridge id		STP enabled	interfaces
br0		7fff.7a18639dd783	no		lan4
							lan2
							wan
							lan5
							lan3
							sfp
br-translate		7fff.7a18639dd783	no		br0.30
							br0.20
```
```
bridge vlan show
port              vlan-id  
sfp               2 PVID Egress Untagged
wan               2 PVID Egress Untagged
                  20
lan2              1 PVID Egress Untagged
                  30
lan3              1 PVID Egress Untagged
lan4              1 PVID Egress Untagged
lan5              1 PVID Egress Untagged
br0               1
                  2
                  20
                  30
br-translate      1 PVID Egress Untagged
br0.20            1 PVID Egress Untagged
br0.30            1 PVID Egress Untagged

```

ptpt52, 2022-09-03:
@john-tho  great!
this looks like work, and I create my own config, it works!
my network config:
```

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fd5f:546d:852a::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	list ports 'eth1'
	list ports 'eth2'
	list ports 'eth3'
	list ports 'eth4'
	list ports 'eth5'

config interface 'lan'
	option proto 'static'
	option ipaddr '192.168.15.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option device 'br-lan.1'

config interface 'wan'
	option proto 'dhcp'
	option device 'br-lan.2'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'eth1:u*'
	list ports 'eth2:u*'
	list ports 'eth3:u*'
	list ports 'eth4:u*'
	list ports 'eth5:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '2'
	list ports 'eth0:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '30'
	list ports 'eth0:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '20'
	list ports 'eth1:t'

config device
	option type 'bridge'
	option name 'br-translate'
	list ports 'br-lan.20'
	list ports 'br-lan.30'
	option bridge_empty '1'

config interface 'translate'
	option proto 'none'
	option device 'br-translate'

```