openwrt/openwrt #10555: Tools: broken xxd download link

KernelErr, 2022-08-30:
Hi, while trying to build Openwrt, I found that download link of xxd seems to be broken. Should we update the link or host the file by the official?

[image]

https://github.com/openwrt/openwrt/blob/c6d6306827e9296faad26981996825ce3e90259d/tools/xxd/Makefile#L8

I found one online with a matching SHA256 hash.

[xxd-1.10.tar.gz](https://github.com/openwrt/openwrt/files/9448781/xxd-1.10.tar.gz)

closed 2022-09-06 by jow-

wasdqwerdf, 2022-09-14:
in tag v21.03.0，the error is still there

ynezz, 2022-09-14:
> in tag v21.03.0，the error is still there

Thanks for reminder, I've just prepared following backports:

* 21.02 https://github.com/openwrt/openwrt/pull/10711
* 22.03 https://github.com/openwrt/openwrt/pull/10709

ynezz, 2022-09-21:
> Thanks for reminder, I've just prepared following backports:

Both are merged now.

thagabe, 2022-09-24:
Hopefully this will be merged soon and new releases for 21.02 and 22.03 will be out soon.

ynezz, 2022-09-24:
> Hopefully this will be merged soon

Should be merged, so the fix is just one `git pull` away.

> and new releases for 21.02 and 22.03 will be out soon.

AFAIK this affects only folks compiling from sources, so they shouldn't have a problem to do `git pull` or `git cherry-pick` to get the fix. Or do I miss something here?

xiaopang0v0, 2022-11-15:
In fact, this source is not invalid, it just blocks http access.
Just change http to https to download normally

commit 06e01e81 (2022-09-06): scripts: xxdi.pl: add xxd -i compat mode

So it can serve as a standalone drop in replacement for xxd utility used
currently mostly in U-Boot packages with `xxd -i` mode which outputs C
include file style, with aim for byte to byte identical output, so the
eventual difference in the generated output is easily spottable.

Fixes: #10555
commit 88c9056a (2022-09-06): tools: remove xxd package

It shouldn't be needed anymore as we've now `scripts/xxdi.pl`, which
should be self contained and fully compatible `xxd -i` replacement.

Fixes: #10555
commit eae2fb80 (2022-09-06): build: provide xxd -i with scripts/xxdi.pl

Dependency on xxd was added in commit c4dd2441e787 ("tools: add xxd
(from vim)") as U-Boot requires xxd to create the default environment
from an external file.

Later in commit 2b94aac7a128 ("tools: xxd: use more convenient source
tarball"), xxd from another source was used instead, but that source is
currently unavailable, so let's fix it by using simple xxdi.pl Perl
script instead.

Fixes: #10555
commit 542447b9 (2022-09-16): scripts: xxdi.pl: add xxd -i compat mode

So it can serve as a standalone drop in replacement for xxd utility used
currently mostly in U-Boot packages with `xxd -i` mode which outputs C
include file style, with aim for byte to byte identical output, so the
eventual difference in the generated output is easily spottable.

Fixes: #10555

(cherry picked from commit 06e01e817ec6643a35beb9e6946689e9cc7d020a)