openwrt/openwrt #10563: 6rd doesn't work until after manual ifup

kberanek, 2022-08-31:
I'm running 22.03.0-rc6 on a Linksys WRT3200ACM with CenturyLink fiber which requires pppoe for ipv4 and 6rd for ipv6.  I recently upgraded to 22.03 and I've been having problems with 6rd.  I have not successfully set up 6rd on a previous release so I'm not sure if this only happens on 22.03 or not.

When I configure 6rd everything looks fine to me in `ip addr`, `ifstatus wan6`, etc., but I'm unable to send any ipv6 traffic over the 6rd-wan6 interface until I manually run `ifup wan6`.  `/etc/init.d/network restart` does NOT work for some reason.  This happens every time I reboot.  The 6rd-wan6 interface seems to come up fine, but will not work until after I run `ifup wan6`.

I do have mwan3 installed, but I can consistently reproduce this after disabling mwan3 via `/etc/init.d/mwan3 disable` and rebooting.  The commands below were run with mwan3 disabled.

I've included the output of some commands I ran after rebooting and waiting a few minutes to make sure everything had stabilized.  You can see I was assigned 2602:62:3ba1:9900::1/24, but I'm unable to send traffic over the interface.  I then run `ifup wan6` and get a new address of 2602:61:7101:a800::1/24.  Seconds later I can successfully send ipv6 traffic over this interface.  As best as I can tell, nothing besides running `ifup wan6` fixes things.  I can wait a really long time, restart the network, etc., but nothing else seems to do it.

There's not really anything interesting in my config files except 1) I removed lan1 and lan2 from br-lan and made them wan interfaces 2) I added a firewall rule to `ALLOW` `option proto 41` for 6rd.

```
root@OpenWrt:~# uptime
 23:31:17 up 4 min,  load average: 0.02, 0.10, 0.05

root@OpenWrt:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1508 qdisc mq state UP group default qlen 1024
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3223:3ff:fee0:12b0/64 scope link
       valid_lft forever preferred_lft forever
3: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
4: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.12.184/24 brd 192.168.12.255 scope global lan2
       valid_lft forever preferred_lft forever
    inet6 2607:fb91:152a:aac0:3223:3ff:fee0:12b0/64 scope global dynamic noprefixroute
       valid_lft 86163sec preferred_lft 14163sec
    inet6 2607:fb91:152a:aac0:64a1:56ca:5a53:3aaa/128 scope global dynamic noprefixroute
       valid_lft 3362sec preferred_lft 2012sec
    inet6 fe80::3223:3ff:fee0:12b0/64 scope link
       valid_lft forever preferred_lft forever
6: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
    inet 98.xx.xx.xx/23 brd 98.xx.xx.xx scope global lan1
       valid_lft forever preferred_lft forever
    inet6 2001:558:600a:8:9c71:5bad:5d5c:a3fb/128 scope global dynamic noprefixroute
       valid_lft 5520sec preferred_lft 5520sec
    inet6 fe80::3223:3ff:fee0:12b0/64 scope link
       valid_lft forever preferred_lft forever
7: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 32:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff permaddr 30:23:03:e0:12:b0
    inet6 fe80::3023:3ff:fee0:12b0/64 scope link
       valid_lft forever preferred_lft forever
8: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: mlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 30:23:03:e0:12:b3 brd ff:ff:ff:ff:ff:ff
12: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 30:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 2601:602:8180:16d4::1/64 scope global dynamic noprefixroute
       valid_lft 5520sec preferred_lft 5520sec
    inet6 2602:62:3ba1:9900::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fd7f:xxxx:xxxx::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::3223:3ff:fee0:12b0/64 scope link
       valid_lft forever preferred_lft forever
13: wan.201@wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 32:23:03:e0:12:b0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3023:3ff:fee0:12b0/64 scope link
[... 566 lines cut ...]
    option family     ipv6
    option target     ACCEPT

config rule
    option name       Allow-IPSec-ESP
    option src        wan
    option dest       lan
    option proto      esp
    option target     ACCEPT

config rule
    option name       Allow-ISAKMP
    option src        wan
    option dest       lan
    option dest_port  500
    option proto      udp
    option target     ACCEPT

config rule
    option name       Allow-6rd
    option src        wan
    option proto      41
    option target     ACCEPT
    option family     ipv4
```

jow-, 2022-08-31:
You have multiple wan interfaces with multiple default routes. It is likely that the 6rd interface picks the wrong one (not the one with the lowest metric) during boot.

Also note that `option device` is not specified for 6rd interfaces (your config also contains `option device` twice for `config interface wan6` which is an error).

Judging from the `option device wan.201` I assume the 6rd interface is meant to bind to `wan` (not `wanb` or `wanc`). Try removing both `option device` lines from it and replace them with a single `option tunlink wan` - this instructs the 6rd proto handler to route the tunnel via `wan` and to deduce the local IPv4 endpoint address from it.

kberanek, 2022-08-31:
Thank you.  That was it.  Using `option tunlink wan` seems to have fixed the problem.  It also resulted in the 6rd interface name changing to `6rd-wan6@pppoe-wan`.

I don't think I came across `option tunlink` when I was trying to figure out how to use 6rd.  Looking right now, it doesn't seem to be mentioned as a possible option for 6rd as far as I can find or really mentioned as a generic option for tunnels that I see.  I'd suggest adding it to a wiki page like https://openwrt.org/docs/guide-user/network/ipv6_ipv4_transitioning or maybe a more general page about tunnels like https://openwrt.org/docs/guide-user/network/tunneling_interface_protocols.

I discovered I also don't actually need the firewall rule to allow protocol 41.  So, for anyone reading this in the future with a similar issue, here's the only config I ended up needing:

```
config interface 'wan6'
        option tunlink wan
        option proto '6rd'
        option peeraddr '205.171.2.64'
        option ip6prefix '2602::'
        option ip6prefixlen '24'
        option ip4prefixlen '0'
```

The multiple `option device` entries looks like it was a typo.  I failed to fully remove one of the lines that was actually commented out when I created this issue.

closed 2022-08-31 by kberanek