openwrt/openwrt #10569: is it possible to use ucidef_set_interface() for proto qmi?

bam80, 2022-08-31:
I use following uci configuration for my QMI modem, is it possible to set it with [ucidef_set_interface()](https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/functions/uci-defaults.sh#L28) or similar function at build-time?
```
set network.wwan='interface'
set network.wwan.proto='qmi'
set network.wwan.device='/dev/cdc-wdm0'
set network.wwan.pdptype='ipv4'
```

MaxS0niX, 2022-08-31:
Hi all, Can it be done like this?
ucidef_set_interface "wwan" device "/dev/cdc-wdm0" protocol "qmi"

Linaro1985, 2022-09-07:
You can add support for this here https://github.com/openwrt/openwrt/blob/master/package/base-files/files/bin/config_generate#L208
by adding something like this:
```
qmi|\
mbim)
    uci -q batch <<-EOF
            set network.$1.proto='${protocol}'
            set network.$1.pdptype='ipv4'
    EOF
    ;;
```

bam80, 2022-09-07:
Thanks @Linaro1985, I think I'll be able to provide the PR.

For now, I tried the following patch which presumably should also work, but for some reason the interface didn't show up:
```
$ git diff
diff --git a/target/linux/ramips/mt76x8/base-files/etc/board.d/02_network b/target/linux/ramips/mt76x8/base-files/etc/board.d/02_network
index ae6870d4a9..f966c766a9 100644
--- a/target/linux/ramips/mt76x8/base-files/etc/board.d/02_network
+++ b/target/linux/ramips/mt76x8/base-files/etc/board.d/02_network
@@ -14,7 +14,6 @@ ramips_setup_interfaces()
        glinet,microuter-n300|\
        glinet,vixmini|\
        hak5,wifi-pineapple-mk7|\
-       kroks,kndrt31r19|\
        mediatek,linkit-smart-7688|\
        minew,g1-c|\
        onion,omega2p|\
@@ -114,6 +113,12 @@ ramips_setup_interfaces()
                ucidef_add_switch "switch0" \
                        "0:lan" "1:lan" "2:lan" "6@eth0"
                ;;
+       kroks,kndrt31r19)
+               uci set network.wan='interface'
+               uci set network.wan.proto='qmi'
+               uci set network.wan.device='/dev/cdc-wdm0'
+               uci set network.wan.pdptype='ipv4'
+               ;;
        motorola,mwr03)
                ucidef_add_switch "switch0" \
                        "1:lan" "2:lan" "3:lan" "0:wan" "6@eth0"

bam80, 2022-09-07:
> I think I'll be able to provide the PR

https://github.com/openwrt/openwrt/pull/10632