openwrt/openwrt #10570: build fails when using ssh to invoke make

Steve-Newcomb, 2022-08-31:
When I use ssh to get to the build host "babs" by running `ssh babs 'cd <openwrt build dir>; make -j1 V=s`, the build always fails quite near the end of the process, and always in the same place.  Here's what that looks like:
```
cat /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/image-qca9563_tplink_archer-a7-v5.dtb >> /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin
/persist/home/srn/OpenWRT/openwrt/staging_dir/host/bin/lzma e /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin -lc1 -lp2 -pb2  /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin.new
mkimage -A mips -O linux -T kernel -C lzma -a 0x80060000 -e 0x80060000 -n 'MIPS OpenWrt Linux-5.10.138'   -d /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin.new
bash: line 1: mkimage: command not found
make[5]: *** [Makefile:104: /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin] Error 127
make[5]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux/ath79/image'
make[4]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux/ath79'
make[3]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux'
make[4]: *** [Makefile:19: install] Error 2
make[3]: *** [Makefile:11: install] Error 2
time: target/linux/install#70.59#10.93#73.85
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/persist/home/srn/OpenWRT/openwrt'
make[1]: *** [target/Makefile:24: /persist/home/srn/OpenWRT/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/persist/home/srn/OpenWRT/openwrt'
make: *** [/persist/home/srn/OpenWRT/openwrt/include/toplevel.mk:230: world] Error 2
220831 10:52 /persist/home/srn root@nutkin# 
```
As you can see, the `mkimage` command is not found, although it's present as `<build-dir>/tools/mkimage`.

When I interactively ssh to the build host (babs) as the same user and I issue exactly the same command, the make process completes normally.   

It doesn't help to tweak the PATH environment variable; the following yielded the same build failure:

```
ssh srn@babs- bash -c 'export PATH="${PATH}:/home/srn/OpenWRT/openwrt/tools"; cd /home/srn/OpenWRT/openwrt; make -j1'
```

I'm using Debian Bullseye (stable) on the build host.

jow-, 2022-08-31:
The issue might not be `$PATH` but the shell being spawned. Does the same happen when you do `ssh babs 'cd <openwrt build dir>; bash -c "make -j1 V=s"'` ?

ynezz, 2022-08-31:
Can you try following fix?

```diff
diff --git a/include/image.mk b/include/image.mk
index bdcf20d30c70..3a3a2f6d0ed5 100644
--- a/include/image.mk
+++ b/include/image.mk
@@ -133,7 +133,7 @@ define Image/BuildKernel/Initramfs
 endef
 
 define Image/BuildKernel/MkuImage
-	mkimage -A $(ARCH) -O linux -T kernel -C $(1) -a $(2) -e $(3) \
+	$(STAGING_DIR_HOST)/bin/mkimage -A $(ARCH) -O linux -T kernel -C $(1) -a $(2) -e $(3) \
 		-n '$(call toupper,$(ARCH)) $(VERSION_DIST) Linux-$(LINUX_VERSION)' -d $(4) $(5)
 endef
```

Steve-Newcomb, 2022-09-01:
Same failure as before.  Curiously, the expanded mkimage path doesn't show up in the verbose output; just "mkimage".  Viz:

```/persist/home/srn/OpenWRT/openwrt/staging_dir/host/bin/lzma e /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin -lc1 -lp2 -pb2  /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin.new
mkimage -A mips -O linux -T kernel -C lzma -a 0x80060000 -e 0x80060000 -n 'MIPS OpenWrt Linux-5.10.138'   -d /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin.new
bash: line 1: mkimage: command not found
make[5]: *** [Makefile:104: /persist/home/srn/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-initramfs-kernel.bin] Error 127
make[5]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux/ath79/image'
make[4]: *** [Makefile:19: install] Error 2
make[4]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux/ath79'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/persist/home/srn/OpenWRT/openwrt/target/linux'
time: target/linux/install#71.27#11.47#74.79
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/persist/home/srn/OpenWRT/openwrt'
make[1]: *** [target/Makefile:24: /persist/home/srn/OpenWRT/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/persist/home/srn/OpenWRT/openwrt'
make: *** [/persist/home/srn/OpenWRT/openwrt/include/toplevel.mk:230: world] Error 2
220901 08:26 ~ root@nutkin# 
```
On the plus side, I suppose, is the fact that when I did a `make -j13` interactively on the build host (rather than remotely and non-interactively via ssh), it still completed successfully, even with your change.  I did not precede with a `make clean` so something may yet have been broken by your change and I wouldn't know it yet.  But it looks OK so far.  Alas, the change doesn't fix the problem.

Steve-Newcomb, 2022-09-01:
jow-: I compared the output of `env` in each situation.  There's no significant difference that I can see, but here's the output of a `diff-u` on the two, just in case you can see something there that I can't:

```
--- /tmp/env-interactive	2022-09-01 09:02:38.312123856 -0400
+++ /tmp/env-ssh	2022-09-01 09:03:55.048304920 -0400
@@ -2,72 +2,53 @@
 DEFAULT_TMLOOM_YEAR=2018
 ch_fs=small
 ch_defaultXtermBackgroundColor_root=wheat
-TERMCAP=
 JAVA_HOME=/usr/java/jdk1.8.0_333
-INSIDE_EMACS=27.1,comint
-WINDOW=0
-ch_backupDefaultEmacsFontFamily4xwin_nonroot=DejaVu Sans Mono
 ch_defaultEmacsColorThemeCommand4xwin_root=%s
-NO_AT_BRIDGE=1
 EDITOR=emacs
 browserPath=/usr/bin/
 tmlirssharePath=/home/srn/irs/2018_irsshare
-PWD=/persist/home/srn/OpenWRT/openwrt
+PWD=/persist/home/srn
 LOGNAME=srn
 ch_defaultXtermBackgroundColor_nonroot=white
 ch_defaultEmacsFontFamily4xwin_nonroot=DejaVu Sans Mono
 XDG_SESSION_TYPE=tty
-ch_backupDefaultEmacsBackgroundColor4xwin_root=Wheat
 ch_defaultEmacsBackgroundColor4xwin_root=Wheat
 ch_defaultEmacsBackgroundColor4xwin_nonroot=White
 MOTD_SHOWN=pam
 HOME=/home/srn
 ch_defaultXtermResourceString_nonroot=*scaleHeight: 1.01
 LANG=en_US.UTF-8
-ch_backupDefaultEmacsFontFamily4xwin_root=DejaVu Sans Mono
-COLUMNS=79
 ch_defaultCh_fs=smaller
 ch_defaultXtermFontFace_nonroot=dejavu
 ch_defaultEmacsFontFamily4xwin_root=DejaVu Sans Mono
 browser=mozilla
-SSH_CONNECTION=192.168.4.1 42978 192.168.2.39 22
+SSH_CONNECTION=127.0.0.1 45276 127.0.1.1 2239
 tmloomPath=/home/srn/tmloom/
 ch_defaultXtermResourceString_root=*scaleHeight: 1.01
-NUTCH_HEAPSIZE=6000
-LESSCLOSE=/usr/bin/lesspipe %s %s
 XDG_SESSION_CLASS=user
-default_printer_name=bend
-TERM=dumb
 ch_defaultXtermFontFace_root=dejavu
-LESSOPEN=| /usr/bin/lesspipe %s
 USER=srn
 VISUAL=emacs
 SUDO_ASKPASS=/usr/bin/ssh-askpass-fullscreen
 DISPLAY=localhost:14.0
 multiprocess_ch_urinorm2=true
 SP_CHARSET_FIXED=yes
-SHLVL=3
+SHLVL=1
 sgmlsPath=/usr/bin/
-XDG_SESSION_ID=4741
+XDG_SESSION_ID=12847
 ch_defaultFontSizeName=smaller
 ch_defaultEmacsColorThemeCommand4xterm_root=%s
-ch_backupDefaultFontSizeName=small
 LD_LIBRARY_PATH=/usr/local/lib:/usr/lib
 XDG_RUNTIME_DIR=/run/user/1118
-ch_backupDefaultEmacsBackgroundColor4xwin_nonroot=White
-PS1=`date "+%y%m%d %H:%M "`${debian_chroot:+($debian_chroot)}\w \u@\h`makePromptEnding \\\$` 
 SP_ENCODING=UTF-8
-SSH_CLIENT=192.168.4.1 42978 22
+SSH_CLIENT=127.0.0.1 45276 2239
 tmllibPath=/home/srn/tmloom/lib
 ch_defaultEmacsColorThemeCommand4xwin_nonroot=%s
 LC_ALL=
 ch_defaultEmacsColorThemeCommand4xterm_nonroot=%s
 PATH=/run/_ch-priv3/ch-priv3:/usr/local/ch-tools3:/opt/schily/bin:/opt/google/chrome:/usr/local/sbin:/usr/sbin:/sbin:/usr/local/bin:/usr/bin:/bin:/usr/bin/X11:/home/srn/tmloom
-STY=2339254.pts-4.babs
 DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1118/bus
-SSH_TTY=/dev/pts/4
 hhcPath=/
 sxPath=/usr/bin/
 DEBIAN_FRONTEND=noninteractive
-OLDPWD=/persist/home/srn
 _=/usr/bin/env
```

ynezz, 2022-09-01:
> ynezz: Same failure as before. Curiously, the expanded mkimage path doesn't show up in the verbose output; just "mkimage". Viz:

It's likely different place, if it weren't expanded, you would see `/bin/mkimage` in that output.

ynezz, 2022-09-02:
Another [fix attempt](http://sprunge.us/umxiH2).

Steve-Newcomb, 2022-09-02:
Thank you for this.  There's good news and bad news.  The bad news is 
that it's not fixed yet; the good news is that we're making progress.

(1) The build still completes interactively, but not via `ssh 
<build-host> "make..."`.  Evidently your patch doesn't break anything 
that wasn't already broken, at least not for me.

(2) When using `ssh <build-host> "make..."` the formerly reported 
"mkimage: command not found" has now disappeared, so I guess maybe 
that's fixed.  However, other commands are now being reported as not 
found: `usign`, `ucert`, `fwtool`:

```echo '{  "metadata_version": "1.1", "compat_version": "1.0", 
"supported_devices":["tplink,archer-a7-v5"], "version": { "dist": 
"OpenWrt", "version": "SNAPSHOT", "revision": "r20448-ab31ffc425", 
"target": "ath79/generic", "board": "tplink_archer-a7-v5" } }' | fwtool 
-I - 
/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin
bash: line 1: fwtool: command not found
make[5]: [Makefile:105: 
/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin] 
Error 127 (ignored)
sha256sum 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin" 
| cut -d" " -f1 > 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.sha256sum"
[ ! -s "/persist/home/srn/OpenWRT/openwrt-21.02.1-current/key-build" -o 
! -s "/persist/home/srn/OpenWRT/openwrt-21.02.1-current/key-build.ucert" 
-o ! -s 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin" 
] || { cp 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/key-build.ucert" 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.ucert" 
; usign -S -m 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin" 
-s "/persist/home/srn/OpenWRT/openwrt-21.02.1-current/key-build" -x 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.sig" 
; ucert -A -c 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.ucert" 
-x 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.sig" 
; fwtool -S 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin.ucert" 
"/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin" 
; }
bash: line 1: usign: command not found
bash: line 1: ucert: command not found
bash: line 1: fwtool: command not found
make[5]: *** [Makefile:105: 
/persist/home/srn/OpenWRT/openwrt-21.02.1-current/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-tplink_archer-a7-v5-squashfs-sysupgrade.bin] 
Error 127
make[5]: Leaving directory 
'/persist/home/srn/OpenWRT/openwrt-21.02.1-current/target/linux/ath79/image'
make[4]: *** [Makefile:19: install] Error 2
make[4]: Leaving directory 
'/persist/home/srn/OpenWRT/openwrt-21.02.1-current/target/linux/ath79'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory 
'/persist/home/srn/OpenWRT/openwrt-21.02.1-current/target/linux'
time: target/linux/install#76.37#9.80#80.76
     ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory 
'/persist/home/srn/OpenWRT/openwrt-21.02.1-current'
make[1]: *** [target/Makefile:24: 
/persist/home/srn/OpenWRT/openwrt-21.02.1-current/staging_dir/target-mips_24kc_musl/stamp/.target_install] 
Error 2
make[1]: Leaving directory 
'/persist/home/srn/OpenWRT/openwrt-21.02.1-current'
make: *** 
[/persist/home/srn/OpenWRT/openwrt-21.02.1-current/include/toplevel.mk:230: 
world] Error 2
220902 10:05 /home/srn/OpenWRT/openwrt ***@***.***{1}#

```

ynezz, 2022-09-02:
```diff
diff --git a/include/image-commands.mk b/include/image-commands.mk
index 65e8c33dd100..d859286f4010 100644
--- a/include/image-commands.mk
+++ b/include/image-commands.mk
@@ -94,9 +94,9 @@ define Build/append-metadata
 	sha256sum "$@" | cut -d" " -f1 > "$@.sha256sum"
 	[ ! -s "$(BUILD_KEY)" -o ! -s "$(BUILD_KEY).ucert" -o ! -s "$@" ] || { \
 		cp "$(BUILD_KEY).ucert" "$@.ucert" ;\
-		usign -S -m "$@" -s "$(BUILD_KEY)" -x "$@.sig" ;\
-		ucert -A -c "$@.ucert" -x "$@.sig" ;\
-		fwtool -S "$@.ucert" "$@" ;\
+		$(STAGING_DIR_HOST)/bin/usign -S -m "$@" -s "$(BUILD_KEY)" -x "$@.sig" ;\
+		$(STAGING_DIR_HOST)/bin/ucert -A -c "$@.ucert" -x "$@.sig" ;\
+		$(STAGING_DIR_HOST)/bin/fwtool -S "$@.ucert" "$@" ;\
 	}
 endef
```

ynezz, 2022-09-02:
```diff
diff --git a/include/image-commands.mk b/include/image-commands.mk
index 65e8c33dd100..46fdd0646333 100644
--- a/include/image-commands.mk
+++ b/include/image-commands.mk
@@ -92,6 +92,7 @@ metadata_json = \
 define Build/append-metadata
 	$(if $(SUPPORTED_DEVICES),-echo $(call metadata_json) | fwtool -I - $@)
 	sha256sum "$@" | cut -d" " -f1 > "$@.sha256sum"
+	env | sort LC_ALL=C > $(TOPDIR)-append-metadata.env
 	[ ! -s "$(BUILD_KEY)" -o ! -s "$(BUILD_KEY).ucert" -o ! -s "$@" ] || { \
 		cp "$(BUILD_KEY).ucert" "$@.ucert" ;\
 		usign -S -m "$@" -s "$(BUILD_KEY)" -x "$@.sig" ;\
```

Steve-Newcomb, 2022-09-03:
Thanks for this suggestion, Petr.  A comparison of the value of PATH (after removing redundant entries) shows that there is one directory that's specified when running `make` interactively that is *not* specified when running `make` noninteractively via an ssh command line:

`/persist/home/srn/OpenWRT/openwrt-21.02.1-current/staging_dir/host/bin`

The directory must be added somehow in the build process when running interactively, and not added when running non-interactively.  I can only wonder why.  Interesting, isn't it?

champtar, 2022-09-03:
can you try `ssh -t` or `ssh -tt` ?

Steve-Newcomb, 2022-09-04:
I'm closing this because my problem is solved, thanks to ynezz's much deeper understanding of the make process.  

However, the absence of `.../staging_dir/host/bin` from the operative PATH when an ssh command line is used to invoke `make` seems to me possibly worthy of some attention.  I can't even imagine how that happens.  The rest of the operative PATH is essentially equivalent either way, if we ignore repetitions.

Steve-Newcomb, 2022-09-04:
I did try `ssh -t` and `ssh -tt` (thanks for the idea, which I forgot to 
acknowledge earlier!).  It didn't help.

ynezz, 2022-09-05:
> I'm closing this because my problem is solved, thanks to ynezz's much deeper understanding of the make process.

Well, the problem is not properly fixed, I've provided just some workaround for you.

> The directory must be added somehow in the build process when running interactively, and not added when running non-interactively. I can only wonder why. Interesting, isn't it?

It should be set properly here https://github.com/openwrt/openwrt/blob/f1802b0db7e22f98e1718e2c8bec29b2842f5aea/Makefile#L18 but for some reason it's not, maybe some wrong recursive Make usage in that target is in play?

RED4911, 2022-09-07:
[image]

Steve-Newcomb, 2022-09-08:
> We should check the PATH in each sub-make invocation to find out where that path gets mangled.

It seems reasonable for me to do that, since I've got a working demo of the bug, and now that my immediate emergency is resolved (thanks again for that).  How?

dhewg, 2023-07-06:
Works without any issues here, so it has to be a problem on your end

Steve-Newcomb, 2023-07-06:
It works here without any issues, too, _as long as I don't use a noninteractive ssh session to run_ `make`.  If I do that, though, `make` consistently fails.  So: I can't script the make; I can't automate it in such a way as to run it on another machine.  I really don't believe it's a problem on my end; I'm running Debian 11 on the script-running machine and on the compiling machine.  I'm compiling the current snapshot, and all snapshots I've tried (as documented above) have failed for quite a while now.  To show what's happening, I'm attaching the following terminal outputs.

(1) `ssh srn@babs 'cd OpenWRT/openwrt; make V=sc -j1' `([make_via_ssh.txt](https://github.com/openwrt/openwrt/files/11968607/make_via_ssh.txt)).  **It fails**.

(2) `ssh srn@babs` and *then* the same command sequence, `cd OpenWRT/openwrt; make V=sc -j1` ([make_on_local_machine.txt](https://github.com/openwrt/openwrt/files/11968617/make_on_local_machine.txt)).  **It works well**.

(3) `diff -u make_via_ssh.txt make_on_local_machine.txt > make-diffs` ([make_diffs.txt](https://github.com/openwrt/openwrt/files/11968635/make_diffs.txt)).  (One wonders why there are so many diffs.)

(4) `scripts/diffconfig` ([diffconfig_output.txt](https://github.com/openwrt/openwrt/files/11968629/diffconfig_output.txt))

I have learned that I cannot update physically-inaccessible mesh nodes without building new sysupgrade binaries for them, with each binary already containing the configuration details that are unique to its target node.  That's pretty tedious, so it's important to automate the process.  As things stand, I have to make each binary by hand.  It's a suboptimal situation.  I raise the issue because it makes me think something may be amiss in the build process itself.  Perhaps there's an occult dependency of some kind?.

dhewg, 2023-07-06:
> It works here without any issues, too, _as long as I don't use a noninteractive ssh session to run_ `make`

I get that, and you described that just fine. Which is why I tested exactly your cmdline, and that is precisely what works without issues here ;)

Steve-Newcomb, 2023-07-06:
Wow.  That's **interesting.**  Did you test it with my .config?  If not, and you're willing, please try it.  
[dotConfig.txt](https://github.com/openwrt/openwrt/files/11969139/dotConfig.txt)

dhewg, 2023-07-06:
I don't believe that your issue is .config related. If I had to guess it's probably session/permission/logind/pam related.
For the record: I ran the ssh cmdline from a debian bookworm arm64 box to access my debian sid amd64 workstation

Steve-Newcomb, 2023-07-06:
Your hunch about it not being .config-related is pretty well confirmed; it's still broken after reverting to the default config.
I see nothing obvious.  I experimented with duplicating the environment variables of the working session; no dice; still broken.  I don't see how pam or logind could be responsible, and I don't see how any permissions would be involved.
 
After a `make clean`, the first compile error is:
```
bison -o scripts/kconfig/parser.tab.c --defines=scripts/kconfig/parser.tab.h -t -l scripts/kconfig/parser.y
/persist/home/srn/OpenWRT/openwrt/staging_dir/host/share/bison/skeletons/yacc.c:420: error: invalid header value: b4_header_flag
/persist/home/srn/OpenWRT/openwrt/staging_dir/host/share/bison/skeletons/yacc.c:420: the top level

```
Pretty deep in the weeds.  I haven't a clue.  I guess this is going to stay broken.  (;^(.

dhewg, 2023-07-06:
I can only guess, but one idea is that it's filesystem related, where you, for whatever reason, have no exec perms.
Try a huge tmpfs and copy your openwrt tree there and try with that?

dhewg, 2023-07-06:
Or maybe it's related to your shell. try `/bin/bash -l -c "cd <openwrt build dir>; make -j1 V=s"`?

Steve-Newcomb, 2023-07-06:
I've tried the `bash -c` trick before, as documented above.  You just suggested I add the `-l` and so I tried that.  No luck.  

I do have exec perms on the filesystem, viz:

```
230706 13:13 /usr/local/ch-tools3 root@umesh{2}# ssh srn@babs 'cd OpenWRT/openwrt; echo "echo Hello,_World" > z; chmod +x z; ./z'
Hello,_World
230706 13:21 /usr/local/ch-tools3 root@umesh{2}# 
```