openwrt/openwrt #10577: kernel issue with qmi

listoczok, 2022-09-01:
I have a zyxel nr7101 router. I ran openwrt on it according to the openwrt instructions. I use qmi protocol. At higher speeds achieved in the lte network, the router sheds errors as below:

on Aug 29 12:42:16 2022 kern.warn kernel: [  297.091482] ------------[ cut here ]------------
Mon Aug 29 12:42:16 2022 kern.warn kernel: [  297.100701] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:467 0x80556bbc
Mon Aug 29 12:42:16 2022 kern.info kernel: [  297.114761] NETDEV WATCHDOG: wwan0 (qmi_wwan): transmit queue 0 timed out
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.128502] Modules linked in: pppoe ppp_async option nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt7603e mt76 mac80211 libchacha20poly1305 cfg80211 usbserial usbnet ts_kmp ts_fsm ts_bm slhc poly1305_mips nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c crc_ccitt compat chacha_mips cdc_wdm asn1_decoder ip6_udp_tunnel udp_tunnel
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.128860]  tun sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common mii crc32c_generic
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.339675] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.10.134 #0
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.351798] Stack : 00000000 8071f098 809a0000 80820000 00000000 00000000 00000000 00000000
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.368446]         00000000 00000000 00000000 00000000 00000000 00000001 81411d28 385b4d43
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.385087]         81411dc0 00000000 00000000 81411bd0 00000038 80399724 ffffffea 00000000
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.401728]         81411bdc 00000142 807c2868 ffffffff 81411d08 806ee0cc 00000000 806f0000
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.418370]         00000009 00000003 ffffffff 80820000 00000018 80403050 0000000c 8098000c
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.435017]         ...
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.439880] Call Trace:
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.439898] [<80399724>] 0x80399724
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.451711] [<80403050>] 0x80403050
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.458662] [<80007b08>] 0x80007b08
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.465595] [<80007b10>] 0x80007b10
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.472527] [<8037e83c>] 0x8037e83c
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.479453] [<800835d8>] 0x800835d8
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.486391] [<80556bbc>] 0x80556bbc
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.493325] [<8002ff28>] 0x8002ff28
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.500253] [<80556bbc>] 0x80556bbc
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.507181] [<80030020>] 0x80030020
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.514162] [<80556bbc>] 0x80556bbc
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.521089] [<8006b290>] 0x8006b290
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.528022] [<805568d0>] 0x805568d0
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.534962] [<800a119c>] 0x800a119c
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.541898] [<800a1cb4>] 0x800a1cb4
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.548832] [<8006a330>] 0x8006a330
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.555766] [<804bcea8>] 0x804bcea8
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.562701] [<800a1d60>] 0x800a1d60
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.569628] [<80085db4>] 0x80085db4
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.576557] [<806d58cc>] 0x806d58cc
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.583495] [<8039a570>] 0x8039a570
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.590429] [<80034b14>] 0x80034b14
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.597358] [<80399d94>] 0x80399d94
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.604291] [<80003448>] 0x80003448
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.611216]
Mon Aug 29 12:42:17 2022 kern.warn kernel: [  297.614458] ---[ end trace 57d2f19a6e8c9511 ]---

What is the problem in the driver or what?

mrkiko, 2022-09-01:
Maybe @bmork can help ?

bmork, 2022-09-01:
I suspect the hardware isn't capable of more without either using QMAP (which the OEM firmware does) or MBIM.  Personally I am using the modem in MBIM mode.  QMAP is much harder to set up.

The modem is switched to MBIM like any other Quectel modem:
`AT+QCFG=“usbnet”,2`
or back to QMI with
`AT+QCFG=“usbnet”,0`

listoczok, 2022-09-01:
AT+QCFG=“usbnet”,2
ERROR
AT+QCFG=“usbnet”,2
ERROR
AT+QCFG=“usbnet”,0
ERROR
AT+QCFG=“usbnet”,0
ERROR
AT+QCFG=“usbnet”,0
ERROR
ati
Quectel
RG502Q-EA
Revision: RG502QEAAAR11A07M4G

Other command or mistake?

bmork, 2022-09-02:
Can't explain that.  Just tested on mine to be 107% sure, and got this running exactly the same modem firmware:

```
root@OpenWrt:/# picocom /dev/ttyUSB3
[..]

ATI
Quectel
RG502Q-EA
Revision: RG502QEAAAR11A07M4G

OK
AT+CFG=?

OK
AT+QCFG=?
+QCFG: "rrc",(0-5)
+QCFG: "hsdpacat",(6,8,10-24)
+QCFG: "hsupacat",(5,6)
+QCFG: "pdp/duplicatechk",(0,1)
+QCFG: "risignaltype",("respective","physical")
+QCFG: "lte/bandprior",(1-43),(1-43),(1-43)
+QCFG: "volte_disable",(0,1)
+QCFG: "diversity/config",(4,6),(1-4),(0)
+QCFG: "div_test_mode",(0,1)
+QCFG: "data_interface",(0,1),(0,1)
+QCFG: "pcie/mode",(0,1)
+QCFG: "pcie_mbim",(0,1)
+QCFG: "sms_control",(0,1),(0,1)
+QCFG: "call_control",(0,1),(0,1)
+QCFG: "usb/maxpower",(0-900)
+QCFG: "efratctl",(0,1)
+QCFG: "gatewayset",(0,1)[,<gateway>]
+QCFG: "netmaskset",(0,1)[,<netmask>]
+QCFG: "mmwave",ant_chip,ant_type
+QCFG: "clat",(0,1),(0,1),<prefix>,(0,32,40,48,56,64,96),<fqdn>,(0,1),(0,1,2,4,8),(0,1),(0,1),(0,1,2),(0,1,2)
+QCFG: "usage/apmem"
+QCFG: "enable_gea1"[,(0,1)]
+QCFG: "dhcppktfltr",(0,1)

OK
AT+QCFG="usbnet"

+QCFG: "usbnet",2

OK
AT+QCFG="usbnet",0
OK
AT+QCFG="usbnet"

+QCFG: "usbnet",0

OK
AT+CFUN=1,1
OK
[669915.955565] usb 2-1: USB disconnect, device number 2
[669915.966060] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[669915.982522] option 2-1:1.0: device disconnected
[669915.992733] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[669916.009288] option 2-1:1.1: device disconnected
[669916.019625] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[669916.036118] option 2-1:1.2: device disconnected

FATAL: [669916.047650] option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
read zero bytes [669916.065110] option 2-1:1.3: device disconnected
from port
term_[669916.077487] cdc_mbim 2-1:1.4 wwan0: unregister 'cdc_mbim' usb-1e1c0000.xhci-1, CDC MBIM
exitfunc: reset failed for dev UNKNOWN: Not a tty
[669916.247383] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep3in intr, mpkt:10, interval:9/32ms
[669916.266064] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep5in intr, mpkt:10, interval:9/32ms
[669916.284590] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep7in intr, mpkt:10, interval:9/32ms
[669916.302970] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep8in intr, mpkt:64, interval:9/32ms
root@OpenWrt:/# [669926.505534] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
[669926.552540] option 2-1:1.0: GSM modem (1-port) converter detected
[669926.565630] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
[669926.580510] option 2-1:1.1: GSM modem (1-port) converter detected
[669926.593444] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB1
[669926.608419] option 2-1:1.2: GSM modem (1-port) converter detected
[669926.621368] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB2
[669926.636342] option 2-1:1.3: GSM modem (1-port) converter detected
[669926.649350] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB3
[669926.664586] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep8in intr, mpkt:8, interval:9/32ms
[669926.684031] qmi_wwan 2-1:1.4: cdc-wdm0: USB WDM device
[669926.696531] qmi_wwan 2-1:1.4 wwan0: register 'qmi_wwan' at usb-1e1c0000.xhci-1, WWAN/QMI device, 8a:af:1e:79:6c:15

root@OpenWrt:/# picocom /dev/ttyUSB3
[..]

AT+QCFG="usbnet"

+QCFG: "usbnet",0

OK
AT+QCFG="usbnet",2
OK
AT+CFUN=1,1
OK
[670010.733315] option1 ttyUSB3: usb_wwan_write: submit urb 0 failed: -19

FATAL: write to port failed: Resource temporarily unavailable
[670010.855552] usb 2-1: USB disconnect, device number 3
[670010.866030] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[670010.882500] option 2-1:1.0: device disconnected
[670010.892666] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[670010.909230] option 2-1:1.1: device disconnected
[670010.919404] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[670010.935940] option 2-1:1.2: device disconnected
[670010.946493] option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
term_exitfunc: r[670010.963013] option 2-1:1.3: device disconnected
eset failed for dev UNKNOWN: Not[670010.980837] qmi_wwan 2-1:1.4 wwan0: unregister 'qmi_wwan' usb-1e1c0000.xhci-1, WWAN/QMI device
 a tty
[670011.146591] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep3in intr, mpkt:10, interval:9/32ms
[670011.166284] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep5in intr, mpkt:10, interval:9/32ms
[670011.184689] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep7in intr, mpkt:10, interval:9/32ms
root@OpenWrt:/# [670021.375539] usb 2-1: new SuperSpeed USB device number 4 using xhci-mtk
[670021.422925] option 2-1:1.0: GSM modem (1-port) converter detected
[670021.435927] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
[670021.450789] option 2-1:1.1: GSM modem (1-port) converter detected
[670021.463720] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB1
[670021.478825] option 2-1:1.2: GSM modem (1-port) converter detected
[670021.491762] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB2
[670021.506828] option 2-1:1.3: GSM modem (1-port) converter detected
[670021.519878] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB3
[670021.556180] cdc_mbim 2-1:1.4: setting rx_max = 16384
[670021.566981] cdc_mbim 2-1:1.4: cdc-wdm0: USB WDM device
[670021.579380] cdc_mbim 2-1:1.4 wwan0: register 'cdc_mbim' at usb-1e1c0000.xhci-1, CDC MBIM, 8a:af:1e:79:6c:15
[670021.676437] 8021q: adding VLAN 0 to HW filter on device wwan0

```

listoczok, 2022-09-02:
Okay, I found a problem. I used "instead" which is correct in this syntax. Now it actually connects as ppp0. I do not know if it still works at high speeds because I can't achieve such a fast speed. However, on mbim my router slowed down significantly if he wanted to openwrt. Commands in picocom are not visible at all, but you can not see typing. After returning to qmi everything returns to normal. It is a pity that this protocol cannot be refined qmi ... Thank you for your help.

I will ask if there is any way in this modem to save the order in which the technology is selected, e.g. 5RNG, LTE, WCDMA or WCDMA, LTE, 5RNG. Like if I type and then immediately verify with the command is ok. But it is enough to do ifdown wan, ifupwan immediately goes back to automatic "AUTO". There is a solution for this?