openwrt/openwrt #10580: Increase package partition size

castillofrancodamian, 2022-09-01:
Why is it so low? I can't install almost any big package like docker.

PolynomialDivision, 2022-09-02:
It would help if you would give us your device name.

castillofrancodamian, 2022-09-02:
I forgot to add that it is "strange" that x86 has so much space and is not used.

PolynomialDivision, 2022-09-02:
You have to resize the partition.

PolynomialDivision, 2022-09-02:
https://openwrt.org/docs/guide-user/installation/openwrt_x86#resizing_partitions

castillofrancodamian, 2022-09-02:
Yes but no. I meant the default image.

PolynomialDivision, 2022-09-02:
An alternative is to just use the imagebuilder and inster all packages into the squashfs directly.

castillofrancodamian, 2022-09-02:
That is the problem in practice. When I add packages to have an image ready, they don't fit because they don't have space.

ynezz, 2022-09-02:
> When I add packages to have an image ready, they don't fit because they don't have space.

Can't you simply set `CONFIG_TARGET_ROOTFS_PARTSIZE` so it fits your use case?

closed 2022-09-02 by ynezz

castillofrancodamian, 2022-09-09:
Is that modifying a configuration file or in the command to build the image?

bobafetthotmail, 2022-10-04:
> We can't simply please everyone with default settings and cover all use cases out there, but the system should be flexible enough to be somehow bend over as needed with little effort.

Default settings should still strive to cover the most common usecase, which is not what is happening here. You lowered the x86 partsize from its original 256MB (that was set in 2016) and I'm not aware of anybody that complained that partsize on x86 was too big. 
While there are plenty of people asking in forums and a wiki article about manually increasing partition size (linked by @PolynomialDivision above). And the people having issues only increased after your change. x86 is a target where space is a non-issue and people are more likely to install heavy packages that need the space since it is a high performance and RAM target.

As you stated in the commit:
https://github.com/openwrt/openwrt/commit/469ba337a73586573699be3999bfdf6f9ebe404f
"As we're now going to pad all images by default, lets decrease the default rootfs partition size from 256MiB to 128MiB in order to save some space."
Not sure what that means, the images are compressed anyway so padding isn't wasting any space.

As a historical reference, the commit that increased default rootfs size to 256 was in 2016.
https://github.com/openwrt/openwrt/commit/a1f83bad606411a561e8e60110c71232b1a28aa2

Can we revert that change and go back to at least 256MB (possibly to 512MB) or is there some other problem I'm not seeing?

bobafetthotmail, 2022-10-04:
> Is that modifying a configuration file or in the command to build the image?

if you build from source it's a setting, if you use the Image builder it's in the .config file, see 
this thread on the forum
https://forum.openwrt.org/t/how-to-set-root-filesystem-partition-size-on-x86-imabebuilder/4765

ynezz, 2022-10-05:
> Not sure what that means, the images are compressed anyway so padding isn't wasting any space.

It was 2109156b794f leading to that change.

> You lowered the x86 partsize from its original 256MB (that was set in 2016) 

Well, that `TARGET_ROOTFS_PARTSIZE` is not x86 specific config option, it's shared amongst targets and IIRC some targets didn't compressed images by default, so this new value was a form of compromise in order to remove the padding and have working combined images in QEMU out of the box.