openwrt/openwrt #10585: [RFC] Copy Upstream-Status from openembedded

PolynomialDivision, 2022-09-02:
As written on [openembedded.org](https://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines#Patch_Header_Recommendations:_Upstream-Status): `In order to keep track of patches and ultimately reduce the number of patches that are required to be maintained, we need to track the status of the patches that are created.` They do this by introducing `Upstream-Status: [STATUS] [WHERE]`. Here is an example:
```
  grub2: Fix CVE-2015-8370
  
  Back to 28; Grub2 Authentication
  
  Two functions suffer from integer underflow fault; the grub_username_get() and grub_password_get()located in
  grub-core/normal/auth.c and lib/crypto.c respectively. This can be exploited to obtain a Grub rescue shell.
  
  Upstream-Status: Accepted [http://git.savannah.gnu.org/cgit/grub.git/commit/?id=451d80e52d851432e109771bb8febafca7a5f1f2]
  CVE: CVE-2015-8370
  Signed-off-by: Joe Developer <joe.developer@example.com>
```
The wiki describes this very detailed:
https://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines#Patch_Header_Recommendations:_Upstream-Status

We already have some kind of upstream status, by looking at the [patch number](https://openwrt.org/docs/guide-developer/toolchain/use-patches-with-buildsystem#naming_patches):
```
0xx - upstream backports
1xx - code awaiting upstream merge
```
However, I see there a huge benefit having the upstream status with a link to a mailinglist in the patch header.
What do you think?

ynezz, 2022-09-02:
This is issue tracker, if you would like to discuss this, then the right place is [openwrt-devel|https://openwrt.org/contact#mailing_lists] mailing list.

closed 2022-09-02 by ynezz

PolynomialDivision, 2022-09-02:
Oh sorry, I will send this through the mailinglist.