openwrt/openwrt #10588: openssh-server closes immediately

Steve-Newcomb, 2022-09-02:
I have installed both dropbear and openssh-server/openssh-client, etc.  `dropbear` is listening at port 20022, and it works.  `sshd` is listening at 22 and a couple of other ports as well.  I do not understand why both client and server fail to communicate with each other, each blaming the other, but only AFTER the public key is accepted.  The agreement on the public key suggests (to me, anyway) that they are communicating OK.

Maybe the firewall isn't getting the message about the port that will be used afterward?  After I do an `/etc/init.d/firewall stop`, the problem persists, so I think not (which may be because I'm making a wrong assumption about the firewall).

The same configuration still works under a build of 21.02.1 that I made some time ago, but it doesn't work now.  I don't know where to look or what to try next.  The test router's hostname is `rpc151`, which may help you decipher the attached (verbose) files, `sshd_server_output.txt`, `ssh_client_output.txt`, `etc-ssh-ssh_config.txt` and `etc-ssh-sshd_config.txt`.  If there's a clue in any of it, I'm certainly not seeing it, but I'm not an openssh expert.

(Alas, the file attachment feature of this github  website keeps saying "Something went really wrong...", so I can only insert the files wholesale right here:)

--------------------------------------------------------
sshd_server_output.txt

root@rpc151:~# /usr/sbin/sshd -d
debug1: sshd version OpenSSH_9.0, OpenSSL 1.1.1q  5 Jul 2022
debug1: private host key #0: ssh-ed25519 SHA256:3pVU9cKYJFNgl+JxXEt6moFeudEuzH4fy7gbzhCT4SQ
debug1: private host key #1: ssh-rsa SHA256:Xf7HrvwwoH5CXObPT+/IKh8JSlGgUtIVWlWq3YCc+fg
debug1: rexec_argv[0]='/usr/sbin/sshd'
debug1: rexec_argv[1]='-d'
debug1: Set /proc/self/oom_score_adj from 0 to -1000
debug1: Bind to port 2751 on ::.
Server listening on :: port 2751.
debug1: Bind to port 2351 on ::.
Server listening on :: port 2351.
debug1: Bind to port 22 on ::.
Server listening on :: port 22.
debug1: Bind to port 2751 on 0.0.0.0.
Server listening on 0.0.0.0 port 2751.
debug1: Bind to port 2351 on 0.0.0.0.
Server listening on 0.0.0.0 port 2351.
debug1: Bind to port 22 on 0.0.0.0.
Server listening on 0.0.0.0 port 22.
debug1: Server will not fork when running in debugging mode.
debug1: rexec start in 9 out 9 newsock 9 pipe -1 sock 12
debug1: sshd version OpenSSH_9.0, OpenSSL 1.1.1q  5 Jul 2022
debug1: private host key #0: ssh-ed25519 SHA256:3pVU9cKYJFNgl+JxXEt6moFeudEuzH4fy7gbzhCT4SQ
debug1: private host key #1: ssh-rsa SHA256:Xf7HrvwwoH5CXObPT+/IKh8JSlGgUtIVWlWq3YCc+fg
debug1: inetd sockets after dupping: 3, 3
Connection from ::1 port 56074 on ::1 port 22 rdomain ""
debug1: Local version string SSH-2.0-OpenSSH_9.0
debug1: Remote protocol version 2.0, remote software version OpenSSH_9.0
debug1: compat_banner: match: OpenSSH_9.0 pat OpenSSH* compat 0x04000000
debug1: permanently_set_uid: 22/22 [preauth]
debug1: list_hostkey_types: ssh-ed25519,rsa-sha2-512,rsa-sha2-256 [preauth]
debug1: SSH2_MSG_KEXINIT sent [preauth]
debug1: SSH2_MSG_KEXINIT received [preauth]
debug1: kex: algorithm: curve25519-sha256 [preauth]
debug1: kex: host key algorithm: ssh-ed25519 [preauth]
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none [preauth]
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none [preauth]
debug1: expecting SSH2_MSG_KEX_ECDH_INIT [preauth]
debug1: SSH2_MSG_KEX_ECDH_INIT received [preauth]
debug1: rekey out after 134217728 blocks [preauth]
debug1: SSH2_MSG_NEWKEYS sent [preauth]
debug1: Sending SSH2_MSG_EXT_INFO [preauth]
debug1: expecting SSH2_MSG_NEWKEYS [preauth]
debug1: SSH2_MSG_NEWKEYS received [preauth]
debug1: rekey in after 134217728 blocks [preauth]
debug1: KEX done [preauth]
debug1: userauth-request for user root service ssh-connection method none [preauth]
debug1: attempt 0 failures 0 [preauth]
debug1: user root matched 'User root' at line 85
debug1: userauth-request for user root service ssh-connection method publickey [preauth]
debug1: attempt 1 failures 0 [preauth]
debug1: userauth_pubkey: publickey test pkalg rsa-sha2-512 pkblob RSA SHA256:C3EMnwJej8S6dYBvWrtXP9FrdA2u4C7H96r0MS217lA [preauth]
debug1: temporarily_use_uid: 0/0 (e=0/0)
debug1: trying public key file /root/.ssh/authorized_keys
debug1: fd 4 clearing O_NONBLOCK
debug1: /root/.ssh/authorized_keys:13: matching key found: RSA SHA256:C3EMnwJej8S6dYBvWrtXP9FrdA2u4C7H96r0MS217lA
debug1: /root/.ssh/authorized_keys:13: key options: agent-forwarding port-forwarding pty user-rc x11-forwarding
Accepted key RSA SHA256:C3EMnwJej8S6dYBvWrtXP9FrdA2u4C7H96r0MS217lA found at /root/.ssh/authorized_keys:13
debug1: restore_uid: 0/0
[... 219 lines cut ...]
###
#VersionAddendum none
X11DisplayOffset 10
X11Forwarding no
#X11UseLocalhost yes
Match User keepalive
    AllowTcpForwarding yes
    X11Forwarding yes
    PermitTunnel yes
    GatewayPorts yes
    AllowAgentForwarding yes
    PasswordAuthentication no
    AuthorizedKeysFile	/home/keepalive/.ssh/authorized_keys

Match User meshmaint
    AllowTcpForwarding yes
    X11Forwarding yes
    PermitTunnel yes
    GatewayPorts yes
    AllowAgentForwarding yes
    PasswordAuthentication no
    AuthorizedKeysFile	/home/meshmaint/.ssh/authorized_keys

###Match User root
###    PasswordAuthentication no

Hurricos, 2022-09-02:
`cat /etc/config/openssl`?

Steve-Newcomb, 2022-09-03:
There is no such file as /etc/config/openssl, nor is there in the earlier (working) build of 21.02.1.  (Interesting idea, though.)

Hurricos, 2022-09-03:
I was wondering if you'd perhaps been running hardware accelerated OpenSSL.
Martin

Steve-Newcomb, 2022-09-04:
Fixed.  After a lot of experimentation, it turned out that the problem was in `/etc/ssh/sshd_config`.

The problem was:

`ClientAliveInterval 30`

That specification no longer works in OpenWRT, although it worked a few months ago on a fresh build of 21.02.1.  Nowadays, the ClientAliveInterval specification, if it is uncommented, *must* be `ClientAliveInterval 0`.  I don't know why it stopped working.  If any value other than the default value 0 is used, public keys don't work.  In my configuration, the server falls back to giving a password prompt.  No reason for the public key failure is logged, even with `LogLevel DEBUG`.  

Who'd have guessed *that* spec was the source of the trouble?  In order to figure this out, I was reduced to a tedious line-by-line "does it still work?" process.

closed 2022-09-04 by Steve-Newcomb