openwrt/openwrt #10609: Build failure with 22.03.0: `Input/output error: 'selinuxenabled'`

bersbersbers, 2022-09-04:
```
Installing libpkgconf.a to /home/bers/openwrt/staging_dir/host/lib
Installing pkgconf to /home/bers/openwrt/staging_dir/host/bin
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/libpkgconf/libpkgconf.h to /home/bers/openwrt/staging_dir/host/include/pkgconf/libpkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/libpkgconf/stdinc.h to /home/bers/openwrt/staging_dir/host/include/pkgconf/libpkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/libpkgconf/iter.h to /home/bers/openwrt/staging_dir/host/include/pkgconf/libpkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/libpkgconf/bsdstubs.h to /home/bers/openwrt/staging_dir/host/include/pkgconf/libpkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/libpkgconf/libpkgconf-api.h to /home/bers/openwrt/staging_dir/host/include/pkgconf/libpkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/man/pkgconf.1 to /home/bers/openwrt/staging_dir/host/share/man/man1
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/man/pkg.m4.7 to /home/bers/openwrt/staging_dir/host/share/man/man7
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/man/pc.5 to /home/bers/openwrt/staging_dir/host/share/man/man5
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/man/pkgconf-personality.5 to /home/bers/openwrt/staging_dir/host/share/man/man5
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/openwrt-build/meson-private/libpkgconf.pc to /home/bers/openwrt/staging_dir/host/lib/pkgconfig
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/pkg.m4 to /home/bers/openwrt/staging_dir/host/share/aclocal
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/AUTHORS to /home/bers/openwrt/staging_dir/host/share/doc/pkgconf
Installing /home/bers/openwrt/build_dir/host/pkgconf-1.8.0/README.md to /home/bers/openwrt/staging_dir/host/share/doc/pkgconf
Traceback (most recent call last):
  File "/home/bers/openwrt/staging_dir/host/bin/meson.py/mesonbuild/mesonmain.py", line 146, in run
    return options.run_func(options)
  File "/home/bers/openwrt/staging_dir/host/bin/meson.py/mesonbuild/minstall.py", line 811, in run
    installer.do_install(datafilename)
  File "/home/bers/openwrt/staging_dir/host/bin/meson.py/mesonbuild/minstall.py", line 593, in do_install
    self.restore_selinux_contexts(destdir)
  File "/home/bers/openwrt/staging_dir/host/bin/meson.py/mesonbuild/minstall.py", line 385, in restore_selinux_contexts
    restore_selinux_contexts()
  File "/home/bers/openwrt/staging_dir/host/bin/meson.py/mesonbuild/minstall.py", line 231, in restore_selinux_contexts
    subprocess.check_call(['selinuxenabled'])
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 5] Input/output error: 'selinuxenabled'

ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
FAILED: meson-install
/home/bers/openwrt/staging_dir/host/bin/python3 /home/bers/openwrt/staging_dir/host/bin/meson.py install --no-rebuild
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:38: /home/bers/openwrt/staging_dir/host/stamp/.pkgconf_installed] Error 1
make[3]: Leaving directory '/home/bers/openwrt/tools/pkgconf'
time: tools/pkgconf/compile#2.62#0.25#3.04
    ERROR: tools/pkgconf failed to build.
make[2]: *** [tools/Makefile:168: tools/pkgconf/compile] Error 1
make[2]: Leaving directory '/home/bers/openwrt'
make[1]: *** [tools/Makefile:164: /home/bers/openwrt/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyynyyyyyyyyyyyyyyyyyyynynyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/bers/openwrt'
make: *** [/home/bers/openwrt/include/toplevel.mk:230: world] Error 2
```

bersbersbers, 2022-09-04:
`apt install selinux-utils` seems to fix this error, but I had expected one of the config steps before to catch that.

ynezz, 2022-09-05:
@bersbersbers can you share output of `./scripts/diffconfig.sh` command or ideally find out, which config option (or combination) triggers this issue? Maybe it's due to `CONFIG_SELINUX=y` enabled?

bersbersbers, 2022-09-05:
@ynezz sure, see below.! I don't see `CONFIG_SELINUX` anywhere, though. I followed the approach described in https://forum.openwrt.org/t/openwrt-19-07-x-ath79-tiny-luci/55458/121 with 22.03.0 and https://openwrt.ashus.net/21.02.3-targets-ath79-tiny-luci/config on cleanly installed Ubuntu 20.04 in WSL (after only `apt install python3`, and that's it - so it may very well be in the toolchain).

```
bers:~/openwrt$ ./scripts/diffconfig.sh
tmp/.config-package.in:53765:error: recursive dependency detected!
tmp/.config-package.in:53765:   symbol PACKAGE_libgd is selected by PACKAGE_libgd-full
tmp/.config-package.in:53799:   symbol PACKAGE_libgd-full depends on PACKAGE_libgd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:53765:error: recursive dependency detected!
tmp/.config-package.in:53765:   symbol PACKAGE_libgd is selected by PACKAGE_libgd-full
tmp/.config-package.in:53799:   symbol PACKAGE_libgd-full depends on PACKAGE_libgd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_tiny=y
CONFIG_TARGET_ath79_tiny_DEVICE_tplink_tl-wr841-v9=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_ASH_OPTIMIZE_FOR_SIZE=y
CONFIG_BUSYBOX_CONFIG_BUNZIP2=y
CONFIG_BUSYBOX_CONFIG_BZCAT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BZIP2_DECOMPRESS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_HTTPD_CGI=y
CONFIG_BUSYBOX_CONFIG_FEATURE_HTTPD_CONFIG_WITH_SCRIPT_INTERPR=y
CONFIG_BUSYBOX_CONFIG_FEATURE_HTTPD_ETAG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_HTTPD_PORT_DEFAULT=80
# CONFIG_BUSYBOX_CONFIG_FEATURE_VERBOSE_USAGE is not set
CONFIG_BUSYBOX_CONFIG_HTTPD=y
# CONFIG_BUSYBOX_CONFIG_LONG_OPTS is not set
# CONFIG_BUSYBOX_CONFIG_NSLOOKUP is not set
# CONFIG_BUSYBOX_CONFIG_SWAPOFF is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_PRINTK is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_firewall=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_igmpproxy=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-ipt-offload=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-phy-ath79-usb=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-usb-chipidea=y
CONFIG_PACKAGE_kmod-usb-chipidea2=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-gadget=y
CONFIG_PACKAGE_kmod-usb-ledtrig-usbport=y
CONFIG_PACKAGE_kmod-usb-phy-nop=y
CONFIG_PACKAGE_kmod-usb-roles=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
# CONFIG_PACKAGE_libpthread is not set
CONFIG_PACKAGE_libubus-lua=y
# CONFIG_PACKAGE_libustream-wolfssl is not set
# CONFIG_PACKAGE_libwolfssl is not set
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_uhttpd=y
# CONFIG_PACKAGE_usign is not set
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
CONFIG_PACKAGE_wpad-mini=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
CONFIG_STRIP_KERNEL_EXPORTS=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_USE_MKLIBS=y
# CONFIG_WPA_WOLFSSL is not set
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libnftnl=y
CONFIG_PACKAGE_libucode=y
CONFIG_PACKAGE_nftables-json=y
CONFIG_PACKAGE_ucode=y
CONFIG_PACKAGE_ucode-mod-fs=y
# CONFIG_PACKAGE_ucode-mod-math is not set
# CONFIG_PACKAGE_ucode-mod-nl80211 is not set
# CONFIG_PACKAGE_ucode-mod-resolv is not set
# CONFIG_PACKAGE_ucode-mod-rtnl is not set
# CONFIG_PACKAGE_ucode-mod-struct is not set
CONFIG_PACKAGE_ucode-mod-ubus=y
CONFIG_PACKAGE_ucode-mod-uci=y
# CONFIG_PACKAGE_ucode-mod-uloop is not set
```

Happy to help debugging further, but I don't see any immediate culprit and I don't really want to bisect this whole tree up there.

neheb, 2022-09-06:
Related: https://github.com/mesonbuild/meson/issues/10774

That is very strange.

neheb, 2022-09-06:
Found the issue:

https://github.com/mesonbuild/meson/blob/master/mesonbuild/minstall.py#L236

I think this is a problem with WSL. Input/Output Error is the wrong thing to report when the "selinuxenabled" binary is missing.

neheb, 2022-09-06:
I tried to reproduce on WSL1 18.04 and 20.04. Couldn't do it.

The only potential issues I can think of are: old Windows 10, old Ubuntu packages (sudo apt update;sudo apt upgrade), or some weird fs setup.

Are you running WSL1 or 2?

bersbersbers, 2022-09-06:
> I tried to reproduce on WSL1 18.04 and 20.04. Couldn't do it.

I still can, so let's see if I can help.

> The only potential issues I can think of are: old Windows 10

Nope - 21H2.

> old Ubuntu packages (sudo apt update;sudo apt upgrade)

TLDR: did not fix the issue.

Okay, so I went ahead to repro this quickly:

```
bers:~/openwrt/build_dir/host/pkgconf-1.8.0/openwrt-build$ sudo apt remove selinux-utils
bers:~/openwrt/build_dir/host/pkgconf-1.8.0/openwrt-build$ /home/bers/openwrt/staging_dir/host/bin/python3 /home/bers/openwrt/staging_dir/host/bin/meson.py install --no-rebuild
```

That now gives the same error again. Then:

```
bers:~/openwrt$ python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 5] Input/output error: 'selinuxenabled'
```

Same thing!

For the record:

```
bers:~/openwrt/build_dir/host/pkgconf-1.8.0/openwrt-build$ python2 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/subprocess.py", line 185, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 5] Input/output error
```

```
bers:~/openwrt$ selinuxenabled
-bash: selinuxenabled: command not found
```

Now, `sudo apt update; sudo apt upgrade`:

```
The following packages will be upgraded:
  base-files curl dirmngr git git-man gnupg gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm gpgv isc-dhcp-client isc-dhcp-common libcurl3-gnutls libcurl4 libcurl4-openssl-dev libfreetype-dev
  libfreetype6 libfreetype6-dev libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgnutls30 libharfbuzz0b libhttp-daemon-perl libnss3 libpython2.7-minimal libpython2.7-stdlib libssl-dev libssl1.1
  libtss2-esys0 libxml2 libxml2-dev linux-libc-dev motd-news-config openssl python2.7 python2.7-minimal r-base r-base-core r-base-dev r-base-html r-cran-cluster r-cran-mass r-cran-nlme r-cran-survival r-doc-html
  r-recommended rsync tzdata
```

Still, same `OSError` as above.

Related: https://github.com/python/cpython/blob/3.8/Lib/subprocess.py#L1678-L1704
(TLDR: an `OSError` does not seem unexpected.)

> or some weird fs setup.

I am not aware of anything.

> Are you running WSL1 or 2?

2: 

```
C:\Users\bers>wsl -l -v
  NAME      STATE           VERSION
* Ubuntu    Stopped         2
```

> edit: add a new file, tools/meson/patches/010-wsl.patch with the contents:

This looks like a useful given, given my output of `python3 -c ...` above. Is there any need to investigate further? What are you getting with `python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"` if `selinuxenabled` is not installed?

By the way, `IOError` is `OSError` now (https://stackoverflow.com/questions/29347790/difference-between-ioerror-and-oserror).

neheb, 2022-09-06:
```
mangix@DESKTOP-VIRSTQA ~> python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/subprocess.py", line 306, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 287, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'selinuxenabled': 'selinuxenabled'
```

yet I don't get the pkgconf error... weird.

I'm running WSL1 though. IIRC WSL2 was using some plan9 thing. Forgot the details...

Did the patch fix it?

bersbersbers, 2022-09-06:
> Did the patch fix it?

Looks like it, yes! `/home/bers/openwrt/staging_dir/host/bin/python3 /home/bers/openwrt/staging_dir/host/bin/meson.py install --no-rebuild` works now.

neheb, 2022-09-06:
I'm still curious. What's the filesystem of the directory? (cat /proc/mounts) IOError indicates a bad driver.

bersbersbers, 2022-09-06:
> What's the filesystem of the directory? (cat /proc/mounts) 

```
$ cat /proc/mounts
/dev/sdb / ext4 rw,relatime,discard,errors=remount-ro,data=ordered 0 0
tmpfs /mnt/wsl tmpfs rw,relatime 0 0
tools /init 9p ro,dirsync,relatime,aname=tools;fmask=022,loose,access=client,trans=fd,rfd=6,wfd=6 0 0
none /dev devtmpfs rw,nosuid,relatime,size=3202648k,nr_inodes=800662,mode=755 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,noatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,noatime 0 0
devpts /dev/pts devpts rw,nosuid,noexec,noatime,gid=5,mode=620,ptmxmode=000 0 0
none /run tmpfs rw,nosuid,noexec,noatime,mode=755 0 0
none /run/lock tmpfs rw,nosuid,nodev,noexec,noatime 0 0
none /run/shm tmpfs rw,nosuid,nodev,noatime 0 0
none /run/user tmpfs rw,nosuid,nodev,noexec,noatime,mode=755 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,relatime 0 0
tmpfs /sys/fs/cgroup tmpfs rw,nosuid,nodev,noexec,relatime,mode=755 0 0
drivers /usr/lib/wsl/drivers 9p ro,dirsync,nosuid,nodev,noatime,aname=drivers;fmask=222;dmask=222,mmap,access=client,msize=65536,trans=fd,rfd=4,wfd=4 0 0
lib /usr/lib/wsl/lib 9p ro,dirsync,nosuid,nodev,noatime,aname=lib;fmask=222;dmask=222,mmap,access=client,msize=65536,trans=fd,rfd=4,wfd=4 0 0
cgroup2 /sys/fs/cgroup/unified cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0
cgroup /sys/fs/cgroup/cpu cgroup rw,nosuid,nodev,noexec,relatime,cpu 0 0
cgroup /sys/fs/cgroup/cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpuacct 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/net_cls cgroup rw,nosuid,nodev,noexec,relatime,net_cls 0 0
cgroup /sys/fs/cgroup/perf_event cgroup rw,nosuid,nodev,noexec,relatime,perf_event 0 0
cgroup /sys/fs/cgroup/net_prio cgroup rw,nosuid,nodev,noexec,relatime,net_prio 0 0
cgroup /sys/fs/cgroup/hugetlb cgroup rw,nosuid,nodev,noexec,relatime,hugetlb 0 0
cgroup /sys/fs/cgroup/pids cgroup rw,nosuid,nodev,noexec,relatime,pids 0 0
cgroup /sys/fs/cgroup/rdma cgroup rw,nosuid,nodev,noexec,relatime,rdma 0 0
C:\134 /mnt/c 9p rw,dirsync,noatime,aname=drvfs;path=C:\;uid=1000;gid=1000;symlinkroot=/mnt/,mmap,access=client,msize=65536,trans=fd,rfd=8,wfd=8 0 0
```

Not sure what all of this is, but I guess these two are relevant:
```
/dev/sdb / ext4 rw,relatime,discard,errors=remount-ro,data=ordered 0 0
C:\134 /mnt/c 9p rw,dirsync,noatime,aname=drvfs;path=C:\;uid=1000;gid=1000;symlinkroot=/mnt/,mmap,access=client,msize=65536,trans=fd,rfd=8,wfd=8 0 0
```

Interestingly, today, I get only this, in the same directories I tried before:

```
$ python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'selinuxenabled'
```

Strange indeed.

neheb, 2022-09-06:
As in the upstream PR, I'm blaming 9p for this. WSL1 does not use it.

bersbersbers, 2022-09-06:
> Interestingly, today, I get only this, in the same directories I tried before:
>
> ```
> $ python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
> Traceback (most recent call last): [...]

Yet, after building once more, I see this - in the same WSL session as before:

```
 make[3] -C package/utils/busybox compile
 make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going onmake: *** [/home/bers/openwrt/include/toplevel.mk:230: world] Error 1

bers:~/openwrt$ python3 -c "import subprocess; subprocess.check_call(['selinuxenabled'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 5] Input/output error: 'selinuxenabled'
bers:~/openwrt$
```

Again, the `FileNotFoundError` error turned into an `OSError`.

neheb, 2022-09-22:
as upstream merged the fix, do you want this backported?

ynezz, 2022-09-23:
> as upstream merged the fix, do you want this backported?

is there any other way to fix this?

neheb, 2022-09-23:
Avoid WSL2?

Ansuel, 2022-09-27:
@neheb this is strange. wsl2 use plan9 from windows to expose wsl2 directory but internally it's just a plain ext4 in a vm... (as you can see from /proc/mounts)

Wonder if I can try to repro this and check if this bug is not fixed on new version of wsl2

neheb, 2022-10-02:
@Shine- that's not why it's slow: https://github.com/Microsoft/WSL/issues/873#issuecomment-425272829 . It's windows itself.

@Ansuel since this was merged this can be closed.