openwrt/openwrt #10612: sfp port of turris omnia was broken after commit aab466f422500842adb62ddb7b1357e2e123ebb2

wackejohn, 2022-09-05:
My sfp module stopped working after commit aab466f422500842adb62ddb7b1357e2e123ebb2:
The eth2 would never up, ang the ethtool showed:
```
Settings for eth2:
        Supported ports: [ FIBRE ]
        Supported link modes:   2500baseX/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  2500baseX/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Auto-negotiation: on
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: d
        Wake-on: d
        Link detected: no
```

I tried to remove the patches in that commit, then my sfp module working again:
```
Settings for eth2:
        Supported ports: [ FIBRE ]
        Supported link modes:   2500baseX/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  2500baseX/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes
```

ynezz, 2022-09-05:
@dangowrt it looks like a regression introduced in aab466f422 (`kernel: backport generic phylink validate`) via #10515

dangowrt, 2022-09-05:
Please share complete `dmesg` output, if possible of both, working and non-working boots.
Most likely we'll need to backport https://github.com/torvalds/linux/commit/2106be4fdf3223d9c5bd485e6ef094139e3197ba

wackejohn, 2022-09-05:
@dangowrt 

not working:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.64 (Wacke@HOME-Server) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.3.0 r20346-163e680e05) 11.3.0, GNU ld (GNU Binutils) 2.39) #0 SMP Sat Sep 3 01:27:59 2022
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Turris Omnia
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14092 r8192 d22772 u45056
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 522560
[    0.000000] Kernel command line: earlyprintk console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=auto rootwait
[    0.000000] Bootloader command line not present
[    0.000000] Unknown kernel command line parameters "earlyprintk mangled_fs=auto", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 2064584K/2097152K available (8192K kernel code, 631K rwdata, 1888K rodata, 1024K init, 232K bss, 32568K reserved, 0K cma-reserved, 1310720K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x06070000 -> 0x16070000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] sched_clock: 64 bits at 800MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000016] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xb881274fa3, max_idle_ns: 440795210636 ns
[    0.000032] Switching to timer-based delay loop, resolution 1ns
[    0.000168] Ignoring duplicate/late registration of read_current_timer delay
[    0.000175] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000307] Calibrating delay loop (skipped), value calculated using timer frequency.. 1600.00 BogoMIPS (lpj=8000000)
[    0.000316] pid_max: default: 32768 minimum: 301
[    0.000411] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000420] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000779] CPU: Testing write buffer coherency: ok
[    0.000799] CPU0: Spectre v2: using BPIALL workaround
[    0.000889] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001149] Setting up static identity map for 0x100000 - 0x100060
[    0.001230] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.001315] mvebu-pmsu: Initializing Power Management Service Unit
[    0.001373] rcu: Hierarchical SRCU implementation.
[    0.001444] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.001502] smp: Bringing up secondary CPUs ...
[    0.001656] Booting CPU 1
[    0.001779] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.001786] CPU1: Spectre v2: using BPIALL workaround
[    0.001841] smp: Brought up 1 node, 2 CPUs
[    0.001847] SMP: Total of 2 processors activated (3200.00 BogoMIPS).
[    0.001852] CPU: All CPU(s) started in SVC mode.
[    0.003715] VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
[    0.003770] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.003780] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.003846] pinctrl core: initialized pinctrl subsystem
[    0.004308] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.004817] DMA: preallocated 256 KiB pool for atomic coherent allocations
[... 879 lines cut ...]
[   45.488071] lan: port 7(wlan1) entered disabled state
[   45.493292] device wlan1 entered promiscuous mode
[   47.326502] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   47.332990] lan: port 7(wlan1) entered blocking state
[   47.338077] lan: port 7(wlan1) entered listening state
squid[10531]: Set Current Directory to /tmp/squid
squid[10758]: Set Current Directory to /tmp/squid
[   51.044682] lan: port 1(lan0) entered learning state
[   51.365360] pppoe-wan3: renamed from ppp0
[... cut ...]
[   59.369867] lan: topology change detected, propagating
[   59.380202] IPv6: ADDRCONF(NETDEV_CHANGE): lan: link becomes ready
[   59.389109] IPv6: ADDRCONF(NETDEV_CHANGE): lan.1: link becomes ready
[   59.395789] IPv6: ADDRCONF(NETDEV_CHANGE): lan.11: link becomes ready
[   61.284679] lan: port 6(wlan0) entered forwarding state
[   61.289940] lan: topology change detected, propagating
[   63.844673] lan: port 7(wlan1) entered forwarding state
[   63.849935] lan: topology change detected, propagating

```

dangowrt, 2022-09-05:
@wackejohn Please try building with #10616 and see if that really fixes the issue.

wackejohn, 2022-09-05:
@dangowrt Thanks, the issue is fixed.

PR #10616 (2022-09-05): kernel: fix mvneta Ethernet after generic phylink validate
Import patches from Linux v5.16 and v5.17 to get 2500Base-X SFP working again with mvneta driver after the generic phylink validate backport.

Fixes: aab466f422 ("kernel: backport generic phylink validate")

Fixes #10612
closed 2022-09-05 by wackejohn