openwrt/openwrt #10613: libnl3 bug

legale, 2022-09-05:
I found a problem related to 4.x linux kernel. iw utility which uses libnl3 `station dump` and `station del aa:bb:cc:dd:ee:ff` is not working. Same problem with my sample code for station delete. 

htpad which use same netlink cmd NL80211_DEL_STATION is working as expected.
iwinfo assoclist which use netlink cmd NL80211_GET_STATION via libnl-tiny is working good too. 

I've tried to find a diff between iw, htpad, and libiwinfo with no luck. It seems that netlink is used almost the same way everywhere. 

This is my sample code.
```
#include <stdio.h> /* printf() */
#include <stdlib.h> /* exit() */
#include <stdbool.h> /* bool, true, false macros */
#include <unistd.h> /* close() */

#include <errno.h> /* printf */
#include <string.h>
#include <time.h>
#include <arpa/inet.h> /* inet_ntop() */
#include <linux/nl80211.h> /* 802.11 netlink interface */
#include <linux/netlink.h> /*netlink macros and structures */
#include <linux/if_ether.h> /*netlink macros and structures */
#include <net/if.h>

/* libnl-3 */
#include <netlink/netlink.h>
#include <netlink/attr.h>
#include <netlink/msg.h>

/*libnl-gen-3*/
#include <netlink/genl/genl.h>
#include <netlink/genl/ctrl.h>

#include "station_del.h"

/* used macros */
#define NL_OWN_PORT (1<<2)
#define ETH_ALEN 6

#define WLAN_FC_STYPE_DISASSOC		10
#define WLAN_FC_STYPE_DEAUTH		12

struct ucred
{
    pid_t pid;			/* PID of sending process.  */
    uid_t uid;			/* UID of sending process.  */
    gid_t gid;			/* GID of sending process.  */
};

struct nl_sock {
    struct sockaddr_nl s_local;
    struct sockaddr_nl s_peer;
    int s_fd;
    int s_proto;
    unsigned int s_seq_next;
    unsigned int s_seq_expect;
    int s_flags;
    struct nl_cb *s_cb;
    size_t s_bufsize;
};

struct nl_msg
{
    int			nm_protocol;
    int			nm_flags;
    struct sockaddr_nl	nm_src;
    struct sockaddr_nl	nm_dst;
    struct ucred		nm_creds;
    struct nlmsghdr *	nm_nlh;
    size_t			nm_size;
    int			nm_refcnt;
};

struct nl80211_state {
    struct nl_sock *nl_sock;
    int nl80211_id;
} nl80211State = {
        .nl_sock = NULL,
        .nl80211_id = 0
};

static int nl80211_msg_error(struct sockaddr_nl *nla,
                             struct nlmsgerr *err, void *arg)
{
    int *ret = arg;
    *ret = err->error;
    return NL_STOP;
}

static int nl80211_msg_finish(struct nl_msg *msg, void *arg)
{
    int *ret = arg;
    *ret = 0;
    return NL_SKIP;
}

static int nl80211_msg_ack(struct nl_msg *msg, void *arg)
{
    int *ret = arg;
    *ret = 0;
    return NL_STOP;
}

static int nl80211_msg_response(struct nl_msg *msg, void *arg)
{
    return NL_SKIP;
}

/* convert hex mac address string FF:FF:FF:FF:FF:FF to 6 binary bytes */
static int mac_addr_atoi(uint8_t *mac, const char *hex) {
    if (strlen(hex) != sizeof("FF:FF:FF:FF:FF:FF") - 1) {
        printf("error: wrong len: %lu expected: %lu\n", strlen(hex), sizeof("FF:FF:FF:FF:FF:FF") - 1);
        return 0;
    }
    mac[0] = strtol(&hex[0], NULL, 16);
    mac[1] = strtol(&hex[3], NULL, 16);
    mac[2] = strtol(&hex[6], NULL, 16);
    mac[3] = strtol(&hex[9], NULL, 16);
    mac[4] = strtol(&hex[12], NULL, 16);
    mac[5] = strtol(&hex[15], NULL, 16);
    return 1;
}

static int nl_cb(struct nl_msg *msg, void *arg) {
    struct genlmsghdr *gnlh =  (void *)(unsigned char *) msg->nm_nlh + NLMSG_HDRLEN;
    static struct nlattr *tb_msg[NL80211_ATTR_MAX + 1];

    nla_parse(tb_msg, NL80211_ATTR_MAX, genlmsg_attrdata(gnlh, 0),
              genlmsg_attrlen(gnlh, 0), NULL);

    /*
    *  <------- NLA_HDRLEN ------> <-- NLA_ALIGN(payload)-->
    * +---------------------+- - -+- - - - - - - - - -+- - -+
    * |        Header       | Pad |     Payload       | Pad |
    * |   (struct nlattr)   | ing |                   | ing |
    * +---------------------+- - -+- - - - - - - - - -+- - -+
    *  <-------------- nlattr->nla_len -------------->
    */
    for (int i = 0; i < NL80211_ATTR_MAX; i++) {
        if (tb_msg[i] == NULL) continue;
        fprintf(stderr, "nla_type: %d nla_len: %d\n", tb_msg[i]->nla_type, tb_msg[i]->nla_len);
    }
    return 0;
}

int nl80211_cmd_del_station(const char *dev, const char *mac) {
    int ret; /* to store returning values */
    /* nl_socket_alloc(), genl_connect() replacement */
//    struct nl_sock sk = {
//            .s_fd = socket(AF_NETLINK, SOCK_RAW, NETLINK_GENERIC),
//            .s_cb = nl_cb_alloc(NL_CB_DEFAULT), /* callback */
//            .s_local.nl_family = AF_NETLINK,
//            .s_peer.nl_family = AF_NETLINK,
//            .s_seq_expect = time(NULL),
//            .s_seq_next = time(NULL),
//
//            /* the port is 0 (unspecified), meaning NL_OWN_PORT */
//            .s_flags = NL_OWN_PORT,
//    };
    struct nl_sock *sk = nl_socket_alloc();
    genl_connect(sk);

    //set nl socket
    nl80211State.nl_sock = sk;

    //find the nl80211 driver ID
    nl80211State.nl80211_id = genl_ctrl_resolve(sk, "nl80211");

    //attach a callback
    nl_cb_set(sk->s_cb, NL_CB_VALID, NL_CB_CUSTOM, nl_cb, NULL);

    //allocate a message
    struct nl_msg *msg = nlmsg_alloc();

    int if_index = if_nametoindex(dev);
    if (if_index == 0) if_index = -1;

    uint8_t mac_addr[ETH_ALEN];
    if (!mac_addr_atoi((uint8_t *) &mac_addr, mac)) {
        fprintf(stderr, "invalid mac address\n");
        return 2;
    }

    enum nl80211_commands cmd = NL80211_CMD_DEL_STATION;
    int flags = 0;

    // setup the message
    genlmsg_put(msg, 0, 0, nl80211State.nl80211_id, 0, flags, cmd, 0);

    //add message attributes
    NLA_PUT_U32(msg, NL80211_ATTR_IFINDEX, if_index);
    NLA_PUT_U8(msg, NL80211_ATTR_MGMT_SUBTYPE, WLAN_FC_STYPE_DEAUTH);

    NLA_PUT(msg, NL80211_ATTR_MAC, ETH_ALEN, mac_addr);

    //NLA_PUT_U8(msg, NL80211_ATTR_MGMT_SUBTYPE, WLAN_FC_STYPE_DISASSOC);

    //NLA_PUT_U16(msg, NL80211_ATTR_REASON_CODE, 244);

    //send the message (this frees it)
    ret = nl_send_auto_complete(sk, msg);
    if (ret < 0) {
        /* -ret bacause nl commands returns negative error code if false */
        fprintf(stderr, "nl_send_auto_complete: %d %s\n", ret, strerror(-ret));
        return (ret);
    }

    int err = 0;
    nl_cb_err(sk->s_cb,               NL_CB_CUSTOM, nl80211_msg_error,  &err);
    nl_cb_set(sk->s_cb, NL_CB_FINISH, NL_CB_CUSTOM, nl80211_msg_finish, &err);
    nl_cb_set(sk->s_cb, NL_CB_ACK,    NL_CB_CUSTOM, nl80211_msg_ack,    &err);

    //block for message to return
    //ret = nl_recvmsgs_default(&sk);
    ret = nl_recvmsgs(sk, sk->s_cb);

    if (ret < 0) {
        /* -ret bacause nl commands returns negative error code if false */
        fprintf(stderr, "nl_send_auto_complete: %d %s\n", ret, strerror(-ret));
    }
    fprintf(stderr, "err: %d\n", err);

    return (ret);

    nla_put_failure: /* this tag is used in NLA_PUT macros */
    nlmsg_free(msg);
    return -ENOBUFS;
}
```

pkgadd, 2022-09-05:
The oldest supported kernel would be v5.4.x in the openwrt-21.02 branch, so if this would only affect kernels older than that, there wouldn't be anything to worry about.

zorun, 2022-09-09:
`iw dev wlan0 station dump` works here on OpenWrt 22.03.0

Please provide more details: exact OpenWrt and kernel version, any custom build options or configuration...