openwrt/openwrt #10619: [22.03] kernel: `kmod-mmc` SD card is not recognized on x86-64 target

raidenii, 2022-09-06:
Hi,

I tried to put OpenWRT 22.03 on an SD card and boot from it. When boots, it stuck at `Waiting for root device PARTUUID=266b958d-769a-bcbd-aa7b-1bac46ab8602` step. Appears the kernel doesn't have MMC/SD support builtin, so I redeployed the image onto an USB flash drive and it boots fine. However, upon checking the kernel config:
```
user@debian:~/openwrt$ cat ../openwrt-imagebuilder-22.03.0-x86-64.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.138/.config | grep -i mmc
# CONFIG_PCI_MMCONFIG is not set
# MMC/SDIO DVB adapters
CONFIG_MMC=y
CONFIG_MMC_BLOCK=y
CONFIG_MMC_BLOCK_MINORS=8
# CONFIG_MMC_TEST is not set
# MMC/SD/SDIO Host Controller Drivers
# CONFIG_MMC_DEBUG is not set
CONFIG_MMC_SDHCI=y
CONFIG_MMC_SDHCI_IO_ACCESSORS=y
CONFIG_MMC_SDHCI_PCI=y
```
It appears that mmc support should be built-in (as opposed to a module). At any rate, I installed both `kmod-mmc` and `kmod-sdhci`, but still, when plugging in/ejecting an SD card, no events are showing up in `dmesg` and no devices of `/dev/mmcblk0*` are found.

`kmod-mmc` provides `/etc/modules.d/mmc` which says both modules `mmc_core` and `mmc_block` should be loaded at boot, however, `lsmod | grep mmc` returns nothing, nor those files exist in `/lib/modules/5.10.138`.

Yet, this folder does exist:
```
root@openwrt:/etc/modules-boot.d# ls -ld /sys/module/mmcblk/
drwxr-xr-x    3 root     root             0 Sep  5 20:23 /sys/module/mmcblk/
```

but its content are drastically different from `/sys/module/mmc_core` and `/sys/module/mmc_block` from another Linux machine I have. Not an expert on this so not sure if I am looking at wrong things. Any help will be much appreciated.

raidenii, 2022-09-06:
The SD card reader is an Intel one:
```
root@openwrt:~# lspci | grep -i sd
00:1b.0 SD Host controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SDXC/MMC Host Controller (rev 0b)
00:1c.0 SD Host controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series eMMC Controller (rev 0b)
00:1e.0 SD Host controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SDIO Controller (rev 0b)
```

which according to https://cateee.net/lkddb/web-lkddb/MMC_SDHCI_PCI.html it should be well supported in current kernel.
also, the SDHCI controller seems to be found, but just not bringing any `mmcblk0*` devices when a card is inserted:
```
root@openwrt:~# dmesg | grep -i sdhci
[    1.742577] sdhci: Secure Digital Host Controller Interface driver
[    1.743038] sdhci: Copyright(c) Pierre Ossman
[    1.743304] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    1.744217] sdhci-pci 0000:00:1c.0: SDHCI controller found [8086:5acc] (rev b)
[    1.745264] mmc0: SDHCI controller on PCI [0000:00:1c.0] using ADMA 64-bit
[    1.745679] sdhci-pci 0000:00:1e.0: SDHCI controller found [8086:5ad0] (rev b)
[    1.746607] mmc1: SDHCI controller on PCI [0000:00:1e.0] using ADMA 64-bit
[    1.756737] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    1.757658] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    2.218705] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    2.587312] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    2.923383] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    4.028856] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    4.276354] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    4.309782] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    4.342156] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    9.488273] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[    9.503570] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[   11.717135] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
[14924.970984] sdhci-pltfm: SDHCI platform and OF driver helper
```

Tested on the same host with a Debian live image and the SD card reader works perfectly, so not a hardware problem.

zorun, 2022-09-09:
Is this a regression, i.e. was it working in a previous release?

raidenii, 2022-09-10:
Good question, fairly sure that didnt work with 21.02 either, but not tested on 19.07 or earlier.

AiyionPrime, 2022-10-29:
Was looking for similar issues, but on the sunxi target.
My boot process is hanging at

`[    1.783044] Waiting for root device /dev/mmcblk0p2...`

Could this be related?

Full bootlog via `screen`:
http://aiyionpri.me/shared/sunxi_boot/screenlog_0.0