openwrt/openwrt #10620: `make defconfig` fails on OpenWrt 22.03.0 tree

Flisk, 2022-09-06:
I'm trying to build a custom firmware image using the following diffconfig:

```
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-c5-v1=y
CONFIG_PACKAGE_ath10k-firmware-qca988x=y
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

Following the instructions given on https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem, I run into this error at the `make defconfig` stage:

```
tmp/.config-package.in:137234: syntax error
tmp/.config-package.in:137233: unknown statement "---help---"
tmp/.config-package.in:137234: unknown statement "Ntfs-3g"
tmp/.config-package.in:137235:warning: ignoring unsupported character ','
tmp/.config-package.in:137235: unknown statement "If"
tmp/.config-package.in:137236: unknown statement "etc."
tmp/.config-package.in:137237: unknown statement "of"
make: *** [/build/openwrt/include/toplevel.mk:120: defconfig] Error 1
```

Am I doing something wrong, or is this a bug?

richard-scott, 2022-09-06:
Did you put your contents of the 'diff' into '.config' in the root of your cloned repo?  
  
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#using_diff_file

Flisk, 2022-09-06:
Here's exactly what I'm doing:
```
build@88c897bad73d:/build/openwrt$ git status; git rev-parse HEAD
HEAD detached at v22.03.0
nothing to commit, working tree clean
17bd6b04777d71d081ddf4ff8fbf44ccaf6f422e

build@88c897bad73d:/build/openwrt$ ./scripts/feeds update -a
[cut for brevity; succeeds]

build@88c897bad73d:/build/openwrt$ ./scripts/feeds install -a
[cut for brevity; succeeds]

build@88c897bad73d:/build/openwrt$ cp ../config/hertz.config .config

build@88c897bad73d:/build/openwrt$ make defconfig
Collecting package info: done
tmp/.config-package.in:137234: syntax error
tmp/.config-package.in:137233: unknown statement "---help---"
tmp/.config-package.in:137234: unknown statement "Ntfs-3g"
tmp/.config-package.in:137235:warning: ignoring unsupported character ','
tmp/.config-package.in:137235: unknown statement "If"
tmp/.config-package.in:137236: unknown statement "etc."
tmp/.config-package.in:137237: unknown statement "of"
make: *** [/build/openwrt/include/toplevel.mk:120: defconfig] Error 1
```
I've double-checked that the file I've copied into place contains exactly the snippet I've provided above.

richard-scott, 2022-09-06:
Can you upload your hertz.config file? or provide a link to it?

Flisk, 2022-09-06:
I already provided the contents of that file: it's the first snippet in my original post. I've made absolutely certain the contents match what I used to produce the output in my previous response.

Are you unable to reproduce the problem? I'm using a custom Docker-based build environment for this, you can find it here: https://github.com/Flisk/openwrt-build

That repo also includes an exact copy of `hertz.config`.

Flisk, 2022-09-06:
My package feeds were indeed still checked out on a v21 branch. I'm able to build now after deleting the entire feeds directory (`git clean -dxf` didn't catch those because it'll still skip git repos and say that it did, which I missed in the output). Thanks for helping!

Is it expected behavior that `scripts/feeds update -a` won't update to a newer version branch when invoked from a newer branch of the openwrt repo? This part looks like a bug – if it's not, the build system usage guide should probably reflect this behavior and advise users to update their feeds manually.

closed 2022-09-06 by Flisk

Flisk, 2022-09-06:
Ah, nevermind. I guess that's what the hint about `make distclean` is for. Sorry for the noise.