openwrt/openwrt #10623: Change `PKG_SOURCE_URL` for `tools/xxd`

jkool702, 2022-09-06:
When building from source, downloading the source for xxd fails. You seem to be aware of this, since you removed it from the required tools ~10 hours ago.

I was building an image slightly before this, and figured out it will download (on my network at least) if you change the `http://` to `https://`. 

i.e., in `tools/xxd/Makefile`, change `PKG_SOURCE_URL` to

`PKG_SOURCE_URL:=https://grail.cba.csuohio.edu/~somos/`

Should you still want to include xxd in required tools (with a working download) this might be a usable fix.

jospezial, 2022-09-06:
```
grail.cba.csuohio.edu verwendet ein ungültiges Sicherheitszertifikat.
 
Dem Zertifikat wird nicht vertraut, weil es vom Aussteller selbst signiert wurde.
 
Fehlercode: MOZILLA_PKIX_ERROR_SELF_SIGNED_CERT
```

ynezz, 2022-09-07:
> When building from source, downloading the source for xxd fails

See #10566, #10555, #10578

closed 2022-09-07 by ynezz

mpratt14, 2022-09-07:
> You seem to be aware of this, since you removed it from the required tools ~10 hours ago.

maybe don't make a pull request if you notice the problem getting fixed....

mpratt14, 2022-09-07:
sorry I mistook this as a PR haha