openwrt/openwrt #10625: dnsmasq cant load config folder in new openwrt release

rezad1393, 2022-09-06:
I upgraded to 22.03 today and got this error:
`daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory`
I have checked and folder is there.

jow-, 2022-09-06:
Please provide your `/etc/config/dhcp` and `/etc/dnsmasq.conf`

rezad1393, 2022-09-06:
```
config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option localservice '1'
        option ednspacket_max '1232'
        option confdir '/tmp/dnsmasq.d'
        option cachesize '1000'
        #list server '/.ir/8.8.8.8'
        list server '/.pool.ntp.org/8.8.8.8'
        #list server '8.8.8.8'
        list server '192.168.1.20#253'
        option rebind_protection '1'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        list ra_flags 'none'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'
        list ra_flags 'none'

```

```
quiet-dhcp
strict-order
log-facility=syslog
min-cache-ttl=3600
all-servers

# For debugging purposes, log each DNS query as it passes through
# dnsmasq.
#log-queries

# Log lots of extra information about DHCP transactions.
#log-dhcp

# Include another lot of configuration options.
#conf-file=/etc/dnsmasq.more.conf
#conf-dir=/etc/dnsmasq.d

# Include all the files in a directory except those ending in .bak
#conf-dir=/etc/dnsmasq.d,.bak

# Include all files in a directory which end in .conf
#conf-dir=/etc/dnsmasq.d/,*.conf

```

stintel, 2022-09-06:
option confdir '/tmp/dnsmasq.d,/etc/dnsmasq.d' ?

rezad1393, 2022-09-06:
??

> option confdir '/tmp/dnsmasq.d,/etc/dnsmasq.d' ?

??

rezad1393, 2022-09-06:
weird.
I just upgraded and it didnt work with new software so I I thought my config was fine.
I dont even remember why that line is there.
thank you.

rezad1393, 2022-09-06:
too soon.
I forgot to uncommment the dnsmasq.conf line before restart.
the bug is still there.

stintel, 2022-09-06:
Remove it from /etc/dnsmasq.conf and add it in /etc/config/dhcp as in my previous comment. If that doesn't work maybe we should change the UCI option to a list and loop over it, adding multiple --conf-dir arguments.

rezad1393, 2022-09-06:
I replaced option confdir '/tmp/dnsmasq.d' with option confdir '/etc/dnsmasq.d' and it works now.
is this a bug that I just am ignoring?
why did I have to change this from one version of openwrt to another?

stintel, 2022-09-06:
I'd rather call it a feature. Dnsmasq is running in ujail now. The init script will make sure the path in the UCI option is exposed in the jail, but we don't parse /etc/dnsmasq.conf to do that.

rezad1393, 2022-09-06:
> I'd rather call it a feature. Dnsmasq is running in ujail now. The init script will make sure the path in the UCI option is exposed in the jail, but we don't parse /etc/dnsmasq.conf to do that.

I was gonna suggest that it may have to do with jail but I didnt see anything in changelog of openwrt.

closed 2022-09-06 by rezad1393