openwrt/openwrt #10628: [OpenWrt 22.03.0 r19685-512e76967f] Slow upload on Linksys WRT3200ACM

goshoo, 2022-09-07:
Hi,

I first want to thank you for the good project!

Now let me describe the issue. 

Yesterday I upgraded OpenWrt from v21.02.3 to 22.03.0 and after the upgrade (I kept old configuration) the upload speed was slow. In normal condition my Internet speed is 900 Mbps download and 500-600 Mbps upload. After the upgrade my download speed was 900 Mbps (usual) but upload speed was only 9 Mbps.  I made reset and deleted my configuration on my router but this didn't change anything.

This is output of dmesg command from my router (Linksys WRT3200ACM).

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.138 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r19685-512e76967f) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sat Sep 3 02:55:34 2022
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Linksys WRT3200ACM
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 11 pages/cpu s16268 r8192 d20596 u45056
[    0.000000] pcpu-alloc: s16268 r8192 d20596 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock6 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs),80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] Bootloader command line (ignored): console=ttyS0,115200 root=/dev/mtdblock6 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs),80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 506940K/524288K available (8192K kernel code, 642K rwdata, 1104K rodata, 1024K init, 238K bss, 17348K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x06070000 -> 0x16070000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000006] sched_clock: 64 bits at 933MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000016] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x1ae5b571769, max_idle_ns: 881590513431 ns
[    0.000027] Switching to timer-based delay loop, resolution 1ns
[    0.000131] Ignoring duplicate/late registration of read_current_timer delay
[    0.000137] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000260] Calibrating delay loop (skipped), value calculated using timer frequency.. 1866.00 BogoMIPS (lpj=9330000)
[    0.000267] pid_max: default: 32768 minimum: 301
[    0.000327] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000332] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000626] CPU: Testing write buffer coherency: ok
[    0.000638] CPU0: Spectre v2: using BPIALL workaround
[    0.000720] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.000920] Setting up static identity map for 0x100000 - 0x100060
[    0.000983] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0xA
[    0.001046] mvebu-pmsu: Initializing Power Management Service Unit
[    0.001097] rcu: Hierarchical SRCU implementation.
[    0.001157] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.001212] smp: Bringing up secondary CPUs ...
[... 374 lines cut ...]
[   27.009932] ieee80211 phy0: change: 0x60
[   27.035960] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   27.042430] br-lan: port 6(wlan1) entered blocking state
[   27.047767] br-lan: port 6(wlan1) entered forwarding state
[   27.107962] ieee80211 phy0: change: 0x100
[   27.208108] ieee80211 phy0: change: 0x100
[   27.216953] ieee80211 phy0: change: 0x62
[   27.379941] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   27.386399] br-lan: port 5(wlan0) entered blocking state
[   27.391749] br-lan: port 5(wlan0) entered forwarding state
[   28.273933] mv88e6085 f1072004.mdio-mii:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   28.282523] br-lan: port 1(lan1) entered blocking state
[   28.287781] br-lan: port 1(lan1) entered forwarding state
```

As you can see, lan3 speed is set to 10Mbps only. After running the following command my upload speed is back to normal (500 - 600 Mbps).

```
ethtool -s lan3 speed 1000 duplex full autoneg on 
```

Please fix this issue, I expect lan3's speed to be 1Gbps as well. I had no such problem with prevois version of OpenWrt.

Kind regards
Georgi

pesa1234, 2022-09-07:
Or remove the flow control .... `ethtool -A lan3 autoneg off rx off`

aquamacho1, 2022-09-07:
Hi @pesa1234, thx for the tip, worked like a charm 👍.

I have WRT1900AC v2, and my device run hotter than before on previous release.
`[    0.005187] mvebu-pmsu: CPU idle is currently broken on Armada 38x: disabling`
I was wondering, can this lead to CPU damage in a long term?  I do not have a fan on my router. Is this something to worry about?

tapper82, 2022-09-08:
@aquamacho1 Your CPU will be OK mate.

pesa1234, 2022-09-21:
Hi, somebody knows if this bug will be solved on the future? Thanks

ynezz, 2022-09-22:
@Ansuel @robimarko @rsalvaterra @tmn505 any idea?

robimarko, 2022-09-22:
Quickly looking at the DSA driver, I dont see what could have broken it, there were new features but nothing suspicious.

I dont have any HW with this switch

Ansuel, 2022-09-22:
Notice flow offload may cause this, this is what happen with r7800

Wasn't enabled by default with the swconfig driver.

BKPepe, 2022-09-22:
Well, I am quite familiar with this issue as it was reported to us by users, who are using Turris routers, but in Turris we did not dig up what's wrong as there were many issues with kernel 5.4/5.10 for us related to switch and DSA, that's we rather use kernel 5.15 even for OpenWrt 21.02 release, which behaves a lot better, but there is still some issue with VLAN filtering.

Reference:
- https://gitlab.nic.cz/turris/os/build/-/issues/310 (there should be more, but I can not find it right now)

tmn505, 2022-09-23:
Also not familiar with this HW.

goshoo, 2022-09-24:
Hey OpenWrt engineers,

It seems that WRT3200ACM router is not available on Amazon, but Linksys WRT1900ACS is available on amazon.de

https://www.amazon.de/-/en/Linksys-WRT1900ACS-Router-Black-Blue/dp/B014MIBLSA/ref=dp_fod_2?pd_rd_i=B014MIBLSA&psc=1

I can make donation for one router (300 EUR) to be bought, Does this make sense to you? If it does, let me know.

laped83, 2022-10-04:
I also got this bug on my linksys WRT1900 acs using openwrt 22.03. Disabling flow control is a workaround.

I still hope that the real issue gets addressed and fixed soon.

pesa1234, 2022-10-04:
> I also got this bug on my linksys WRT1900 acs using openwrt 22.03. Disabling flow control is a workaround.
> 
> I still hope that the real issue gets addressed and fixed soon.

I think that nobody look at this matter... I found this bug from April 2021 on kernel 5.10...

drujd, 2022-10-24:
Turning flow control off will only work around the speed limitation (resulting from devices getting flow control frames they're not supposed to get), but not the underlying issue of everything sent to switch (at least to the CPU port) being broadcasted to every port. There will still be huge performance issues stemming from that regardless the flow control setting...

Turris team fixed this for their 5.15 kernel (or, to be precise, they gave up on trying to find the root cause and just ipgraded to 5.15 which has some DSA stuff fixed)... https://gitlab.nic.cz/turris/os/build/-/issues/336

Until this gets fixed, no device with the same/similar Marvell switch should be listed as supported for 22.03.2...

pesa1234, 2022-10-24:
> Turning flow control off will only work around the speed limitation (resulting from devices getting flow control frames they're not supposed to get), but not the underlying issue of everything sent [...]

I'm agree, it's a good choice.

drujd, 2022-10-25:
Tried today's OpenWrt snapshot. With 5.10, it exhibits the same issue. With 5.15 (testing kernel) everything is OK. So no kernel patches from the Turris project (at least not any that are not already upstreamed in OpenWrt as well) are needed, OpenWrt's 5.15 kernel straight up works.

pesa1234, 2022-10-31:
> Tried today's OpenWrt master build from source. With 5.10, it exhibits the same issue. With 5.15 (testing kernel) everything is OK. So no kernel patches from the Turris project (at least not any that are not already upstreamed in OpenWrt as well) are needed, OpenWrt's 5.15 kernel straight up works.

Yes you are right, kernel 5.15.76 with wrt1900acs no issue at all... Lan LEDs now blinking correctly also...

ynezz, 2022-12-01:
> So, 22.03.x with kernel 5.10.x have this bug, and in my opinion is a big bug....

We've discussed those issues in the developer meeting yesterday and the conclusion was to disable support for those broken devices in 22.03 release. So [I've just sent out such proposal in the patch form for a review](https://lists.infradead.org/pipermail/openwrt-devel/2022-December/039921.html).

BKPepe, 2022-12-01:
Will there be somewhere notes for the developer mettings?  I know there is this page https://openwrt.org/meetings/start , but it was not updated since March 2022.

BKPepe, 2022-12-01:
Thanks, @ynezz! Would OpenWrt be interested in having (at least testing) kernel 5.15 for mvebu, where it works? It runs just fine on all our (= Turris) devices. Unfortunately, the OpenWrt 22.03 release would have different kernel versions, and I don't think that's wanted, though. Basically, we can provide support for such a kernel version if it helps because we are not going to invest time to look up what's broken with kernel 5.10.

robimarko, 2022-12-01:
There is 5.15 for mvebu, but there is no 5.15 support in general in 22.03

ynezz, 2022-12-01:
> Unfortunately, the OpenWrt 22.03 release would have different kernel versions, and I don't think that's wanted, though.

Indeed, IIRC then in the past there were releases having multiple different kernel versions, but it turned out to be a support and maintenance nightmare, so it was decided to support only single kernel version in each release since then.

goshoo, 2022-12-01:
> We've discussed those issues in the developer meeting yesterday and the conclusion was to disable support for those broken devices in 22.03 release. So [I've just sent out such proposal in the patch [...]

I'm afraid I don't understand nature of this issue - is it hardware and/or software?

What are possible options for users/owners of such devices, can they continue using OpenWrt? In case the answer is "yes" - how?

In case of such hardware being buggy or not supported anymore, could you recommend similar or better router with good OpenWrt support which can handle 1Gbps traffic (or even more)?

BKPepe, 2022-12-01:
From my point of view, this is really drastic solution/workaround, but I understand the OpenWrt reasons. I would say that this mostly affects experienced users, but for many people it will run, if they are not using VLANs, etc. But yes, mvebu target or to be specifically Turris devices will be supported and are, but the thing is that the software (in this case, we are talking about Linux kernel) is somehow problematic on version 5.10. This leads to two solutions, which can be done right now:

1) Use the testing kernel 5.15, which is in the master branch and I will prepare PR to switch it there by default
2) Stay on OpenWrt 21.02 release. It's still supported.

There is also the possibility that some volunteer steps in (not me or anyone from Turris, as we are using a newer kernel version, where it works) and look into it.

pesa1234, 2022-12-01:
I'm the last wheel, we say in Italy, but I'm agree to don't use kernel 5.10 on mvebu devices. So, if it is possible to use as standard 5.15 on master is the good choice....

drujd, 2022-12-01:
> I would say that this mostly affects experienced users, but for many people it will run, if they are not using VLANs, etc.

That's the thing, that's wrong. The "switch acts as an ancient-times hub" bug affects everyone, even with default settings (no vlans etc.). Most people just won't notice until the performance impact starts affecting them, but is not working correctly at all...

Those switches are truly broken on 5.10 (and fine on 5.15).

BKPepe, 2022-12-01:
I hear your voice loud and clear. Since you created a duplicate issue and you are affected by this bug, it bothers you, but it is the wrong expectation that everyone has the same setup as you did. If it would affect many people, then the issue would have many comments or upvotes, right? Based on the notes, there are just a few issues with it. 

What is done because of you is that the OpenWrt 22.03 images will not be provided for mvebu. If it is not good or bad, we can discuss it somewhere else, but some people are using it. OpenWrt provides some statistics how many times the image was downloaded, how many packages were downloaded, etc. 

As said, if you are so proactive, you can look into it or use the OpenWrt snapshot with the testing kernel. That's an option, too.

drujd, 2022-12-01:
> but it is the wrong expectation that everyone has the same setup as you did.

No, it is not. If you have the affected switch in your OpenWrt router and kernel 5.10, the bug affects you. With or without VLANs etc. And most importantly, it affects every user running even a completely default configuration.

>,If it would affect many people, then the issue would have many comments or upvotes, right?

Only if the people noticed. Many didn't - yet. That does not mean there is not a glaring issue.

> Based on the notes, there are just a few issues with it.

There is a huge issue with the switch. Big enough that stopping people upgrading to 22.03 is the right thing to do. People with such devices should stay on (downgrade to) the older version of OpenWrt. Just because many didn't notice the problem does not mean there isn't one. 

I am not "proactive", I am building my own image from source / master branch and will keep doing so until there is a fully working nftables-based version of OpenWrt or TurrisOS for Turris Omnia. It's not a big issue for me, just an inconvenience, but again, stopping people from installing 22.03 with these switches is the right thing to do. If not to stop building the images completely, then at least a big warning or something... It IS broken.

drujd, 2022-12-01:
If people are not willing to build their own OpenWrt image from source, they are MORE affected by the bug, not less - I have a fix/workaround (use 5.15 kernel), they don't.

The bug affects (albeit unknowingly) many people, for instance connecting a 10mbit or 100mbit device that negotiates traffic control frames (TVs, Philips HUE bridge, printers...) to the switch will result in a slow down to 100mbit or even 10mbit (in one direction) on _all_ switch ports.

If you only connect 1gbit devices to the switch and/or disable traffic control frames (which is a non-default configuration), you may not notice the issue, but it is still there. There is a reason network switches have replaced the old dumb ethernet hubs...

HolgerBirkmeyer, 2022-12-01:
Dear Jan, thank you for clarifying that. In my humble opinion, this bug affects core functionality of the router and I am somewhat surprised about the discussion: it never occurred to me, that an upvote of the issue was expected, to move that issue forward. I was certain, the severity was clear... 

And yes: It does affect me as well. I fully support the approach, to mark the affected devices as unsupported until the team finds the time, to introduce the new kernel officially at some point. I hope my understanding is correct? I will just stay on 21.0.5 until then. 

Thank you and regards, Holger

codyc1515, 2023-06-02:
I am facing the same issue with the WRT1900ACS and a Philips hue hub. Installing ethtool and then running `ethtool -A lan3 autoneg off rx off` for me also worked. As everyone says, the issue came about in 22.03.

codyc1515, 2023-06-08:
Installed the latest `snapshot` build and it's working okay now. Can we close this?

anton000, 2023-07-08:
WRT1900ACS can find the 23.05.0-rc2 img here. Be sure not to flash the **snapshot**, it doesnt have Luci. 

https://firmware-selector.openwrt.org/?version=23.05.0-rc2&target=mvebu%2Fcortexa9&id=linksys_wrt1900acs