openwrt/openwrt #10633: ath79/tiny latest build does not support ed25519

infinity0, 2022-09-07:
After sysupgrading (using `-n -F` as instructed) to https://downloads.openwrt.org/releases/22.03.0/targets/ath79/tiny/openwrt-22.03.0-ath79-tiny-tplink_tl-wpa8630p-v2.0-eu-squashfs-sysupgrade.bin from a previous 21.02.xx snapshot version, I was locked out of my device. I was able to regain access by booting into failsafe mode and adding a rsa key.

Looks like the root cause is because ed25519 support was dropped from dropbear:

~~~~
# cat /etc/os-release 
NAME="OpenWrt"
VERSION="22.03.0"
[..]
# dropbearkey 
Must specify a key filename
Usage: dropbearkey -t <type> -f <filename> [-s bits]
-t type	Type of key to generate. One of:
		rsa
-f filename    Use filename for the secret key.
~~~~

The culprit may be openwrt/openwrt@44e1e5d153d00915a7e516c9af3f440cbd84cf78 but I am not sure how that disables ed25519 support. The string "ed25519" is only mentioned in a few files in this repo:

~~~~
openwrt$ rg ed25519
package/network/services/dropbear/Config.in
57:		  ssh-ed25519

package/network/services/dropbear/Makefile
71:$(if $(CONFIG_DROPBEAR_ED25519),/etc/dropbear/dropbear_ed25519_host_key)
206:	$(if $(CONFIG_DROPBEAR_ED25519),touch $(1)/etc/dropbear/dropbear_ed25519_host_key)

package/network/services/dropbear/files/dropbear.init
69:	for ktype in ed25519 ecdsa rsa; do
~~~~

@jwmullally any ideas?

jwmullally, 2022-09-07:
This is a general issue with the `dropbear` package when built for small flash devices. It was set here to save 12K: https://github.com/openwrt/openwrt/commit/d0f295837a03f7f52000ae6d395827bdde7996a4 so they would be the best people to weight in.

<https://github.com/openwrt/openwrt/blob/openwrt-22.03/package/network/services/dropbear/Config.in>
```
config DROPBEAR_ED25519
	bool "Ed25519 support"
	default y if !SMALL_FLASH
```

<https://github.com/openwrt/openwrt/blob/openwrt-22.03/target/linux/ath79/tiny/target.mk>
```
FEATURES += small_flash
```

I don't know much about this area, so this is just my opinion. If this was enabled for SMALL_FLASH, something else might need to be disabled (e.g. deprecate old ciphers) to not increase space. Not sure anyone will want to change this for `/openwrt-22.03`, but maybe for `/master`. In general it seems like a good idea to keep the SMALL_FLASH up-to-date with slim but modern features, as soon I'd say most of today's 8M flash units will be tomorrow's SMALL_FLASH targets. (Though I doubt this particular device TL-WPA8630Pv2 with 6M flash will be big enough even for the next ath79/tiny release).

---

> I was locked out of my device

Since you did `sysupgrade -n -F`, you would have been ssh'ing into a clean install with just the auto-generated host signature keys. Can you describe this impact more? Was the device unresponsive, or was ssh & LuCI running, but you couldn't ssh to it? 

It sounds like the image itself is OK. Many Linux distributions (e.g. [Fedora](https://fedoraproject.org/wiki/Changes/StrongCryptoSettings2)) are continuously disabling weaker crypto by default in the ssh client, which might have been what you saw until you manually put in an RSA key. I ran into this with git+ssh access to bitbucket.org recently after upgrading distro's.

infinity0, 2022-09-07:
Ok, thanks for the explanation. Yes the image itself works outside of my issue.

> Since you did `sysupgrade -n -F`, you would have been ssh'ing into a clean install with just the auto-generated host signature keys. Can you describe this impact more? Was the device unresponsive, or was ssh & LuCI running, but you couldn't ssh to it?

To be precise, I did `sysupgrade --create-backup` before upgrading, then did `sysupgrade -n -F`, then logged into the default root shell fine with no password, then I did `sysupgrade --restore-backup` and then was locked out.

I had thought the backup-upgrade-restore loop would be fine (like a regular upgrade), as the message about `-n -F` seemed to suggest it was only necessary because of disk image format issues. IMO it was not clear that the new image would also have significant config changes such as loss of ed25519. So in this regard, the message about `-n -F` could be improved a bit.

jwmullally, 2022-09-08:
I added this message to [the wiki page](https://openwrt.org/toh/tp-link/tl-wpa8630p_v2#installation):

> If you are upgrading to these official ath79/tiny releases from an early ath79/generic development snapshot (i.e. when sysupgrading you see the DEVICE_COMPAT_MESSAGE “The flash erase blocksize has changed to 4k from the 64k in previous versions…”), you should not restore a backup of your previous settings due to subtle differences in packages built for the tiny target (e.g. [No ed25519 support in dropbear](https://github.com/openwrt/openwrt/issues/10633)), and instead start from clean settings using a backup for reference. Any future backup/restores should work fine.)

This device never had ath79/generic "-factory.bin" files released due to the +6M flash size issue, as such I'm not sure its worth any extra patches to the sysupgrade message (and the issue is recoverable with a reset when it happens), so this note in the wiki should be good enough and the issue considered resolved for this particular device. Sorry for any inconvenience caused, I hadn't noticed this in testing.

---

If the OpenWrt devs want to keep this issue open for the general issue of "ath79/tiny latest build does not support ed25519" and solve it for `/master` thats fine by me, but otherwise I'd say this could be closed if everyone is happy.

zorun, 2022-09-09:
Closing, ed25519 has never been available by default in ath79/tiny, so this is expected (at least currently). The behaviour change is simply because this device has been moved from ath79/generic to ath79/tiny.

If you want to propose enabling ed25519 on SMALL_FLASH devices, feel free to start a discussion on the openwrt-devel mailing list or submit a patch.

closed 2022-09-09 by zorun