openwrt/openwrt #10635: 22.03 doesn't work on glinet s1300

NotAgentBilly, 2022-09-07:
I tried sysupgrade / fresh. Neither worked.

I'm an openwrt novice. But I noticed that, for example, /etc/opkg/distfeeds.conf shows feeds from glinet whereas when i install 22.02.3 distfeeds shows feeds from openwrt. So clearly something is busted :man_shrugging:

zorun, 2022-09-09:
Please give more details. "doesn't work" could mean anything.

NotAgentBilly, 2022-09-10:
Well all I can really give is:

Internet works as normal. Luci is missing. when sshing and trying to install luci it has the wrong distfeeds/packages installed (perhaps gli-net defaults as their website is referenced in the distfeeds).

PolynomialDivision, 2022-09-12:
Similar issues are described here:
https://forum.openwrt.org/t/need-people-testing-ipq40xx-dsa-pr/132154/37