openwrt/openwrt #10639: firewall4 may omit all of the jumps to the *_wan chains

smferris, 2022-09-07:
I'm testing 22.03.0 on a router that's not yet deployed, and it's failing to build the firewall properly.  It omits the wan zone/interface from the input, forward, and output chains.  It did build the input_wan, forward_wan, and output_wan chains based on /etc/config/firewall, but they're never used in the ruleset.

```
        chain input {
                type filter hook input priority filter; policy accept;
                iifname "lo" accept comment "!fw4: Accept traffic from loopback"
                ct state established,related accept comment "!fw4: Allow inbound established and related flows"
                ct state invalid drop comment "!fw4: Drop flows with invalid conntrack state"
                tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
                iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
                iifname "br-guest" jump input_guest comment "!fw4: Handle guest IPv4/IPv6 input traffic"
        }

        chain forward {
                type filter hook forward priority filter; policy drop;
                meta l4proto { tcp, udp } flow add @ft
                ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
                ct state invalid drop comment "!fw4: Drop flows with invalid conntrack state"
                iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
                iifname "br-guest" jump forward_guest comment "!fw4: Handle guest IPv4/IPv6 forward traffic"
                jump handle_reject
        }

        chain output {
                type filter hook output priority filter; policy accept;
                oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
                ct state established,related accept comment "!fw4: Allow outbound established and related flows"
                ct state invalid drop comment "!fw4: Drop flows with invalid conntrack state"
                oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
                oifname "br-guest" jump output_guest comment "!fw4: Handle guest IPv4/IPv6 output traffic"
        }
```

At a wild guess, perhaps it's failing to process the wan zone because its eth device is currently down?  If that's the cause, I think it's a security issue, since best practice is to define the firewall rules before the interface goes up to avoid races.  It's not hard to imagine a router booting up after a power outage, and finding the WAN interface down because the equipment providing it hasn't finished booting yet.

```
$ ifstatus wan
{
        "up": false,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "dhcp",
        "device": "eth1",
        "data": {

        }
}
```

If it's not the interface being down, then I'm even more puzzled.  It clearly knows the wan zone exists since it generated input_wan, forward_wan, and output_wan.

Normally I'd troubleshoot further but I wasn't able to find any useful documentation on how to debug ucode.  If fw4 was implemented as a shell script, lua script, or C program, I'd be able to do more to figure out what the problem is.

ruleset.uc has this in it:
```
{% for (let zone in fw4.zones()): for (let rule in zone.match_rules): %}
                {%+ include("zone-jump.uc", { fw4, zone, rule, direction: "input" }) %}
{% endfor; endfor %}
```

From that, I'm assuming that either fw4.zones() is missing "wan", or the match_rules are doing something to exclude it. I have no idea how to get ucode/utpl to tell me fw4.zones() contains, or what the match_rules are.

jow-, 2022-09-07:
> At a wild guess, perhaps it's failing to process the wan zone because its eth device is currently down?

This is correct, it is down, due to that the `l3_device` property is missing and the netdev cannot be resolved. The following patch will fall back to `device` in this case:

```diff
diff --git a/root/usr/share/ucode/fw4.uc b/root/usr/share/ucode/fw4.uc
index 766290e..a7837ed 100644
--- a/root/usr/share/ucode/fw4.uc
+++ b/root/usr/share/ucode/fw4.uc
@@ -541,7 +541,7 @@ return {
                        for (let ifc in ifaces.interface) {
                                let net = {
                                        up: ifc.up,
-                                       device: ifc.l3_device,
+                                       device: ifc.l3_device ?? ifc.device,
                                        physdev: ifc.device,
                                        zone: ifc.data?.zone
                                };
```

Note that for some protocols the l3 device will differ from the device and it might not be known in advance, so your claimed security issue would still occur then (rules would be set up for `.device` but until hotplug triggered firewall reload no rules for `.l3_device`). This has been the case with firewall3 too however, so it wouldn't be a regression.

> If that's the cause, I think it's a security issue, since best practice is to define the firewall rules before the interface goes up to avoid races.

Yes, I agree. If the interfaces are known in advance.

> I have no idea how to get ucode/utpl to tell me fw4.zones() contains, or what the match_rules are.

Inside `{% ... %}` statement blocks you can simply invoke `print()` and the value will be printed as JSON dictionary. Outside `{% ... %}` statement blocks you can simply add an expression block: ``{{ zone.match_rules }}``

smferris, 2022-09-07:
I tried your patch and I now get the jump rules I was expecting to the *_wan chains.

commit cb24be47 (2022-10-14): firewall4: update to latest Git HEAD

4fbf6d7 ruleset.uc: log forwarded traffic not matched by zone policies
c7201a3 main.uc: reintroduce set reload restriction
756f1e2 ruleset: fix emitting set_mark/set_xmark rules with masks
3db4741 ruleset: properly handle zone names starting with a digit
43d8ef5 fw4: fix formatting of default log prefix
592ba45 main.uc: remove uneeded/wrong set reload restrictions
b0a6bff tests: fix testcases
145e159 fw4: recognize `option log` and `option counter` in `config nat` sections
ce050a8 fw4: fall back to device if l3_device is not available in ifstatus

Fixes: # [... cut]
commit 9a599fee (2022-10-14): firewall4: update to latest Git HEAD

4fbf6d7 ruleset.uc: log forwarded traffic not matched by zone policies
c7201a3 main.uc: reintroduce set reload restriction
756f1e2 ruleset: fix emitting set_mark/set_xmark rules with masks
3db4741 ruleset: properly handle zone names starting with a digit
43d8ef5 fw4: fix formatting of default log prefix
592ba45 main.uc: remove uneeded/wrong set reload restrictions
b0a6bff tests: fix testcases
145e159 fw4: recognize `option log` and `option counter` in `config nat` sections
ce050a8 fw4: fall back to device if l3_device is not available in ifstatus

Fixes: # [... cut]
closed 2022-10-14 by openwrt-bot