openwrt/openwrt #10642: cant install mwlwifi，pls help

magellan-99, 2022-09-08:
When I use this code try to install mwlwifi, it tells me "unkown package kmod-mwlwifi":

opkg update
opkg install kmod-mac80211
opkg kmod-mwlwifi

what can I do next?

magellan-99, 2022-09-08:
the plateform is X86_64

obsy, 2022-09-08:
Because mwlwifi depends on TARGET_mvebu only. Do you have this card in x86?

magellan-99, 2022-09-08:
> Because mwlwifi depends on TARGET_mvebu only. Do you have this card in x86?

osby，I didn't know what is TARGET_mvebu. I use a surface pro3 laptop to run  openwrt and want to use its wireless card. how can i add  this wireless driver?

obsy, 2022-09-08:
Show us output of:

lspci -v

First install pciutils

magellan-99, 2022-09-08:
> Show us output of:
> 
> lspci -v
> 
> First install pciutils

lspci -v display:
01:00.0 Ethernet controller: Marvell Technolody Group Ltd. 88W8897 [AVASTAR] 802.11ac Wireless
            Subsystem: SafeNet (wrong ID) Device 045e
            Flags: fast devsel, IRQ 255

obsy, 2022-09-08:
Install kmod-mwifiex-pcie and wpad-openssl

magellan-99, 2022-09-08:
> Install kmod-mwifiex-pcie and wpad-openssl

opkg install kmod-mwifiex-pcie

Multiple packages (libgccl and libgccl) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'kmod-mwifiex-pcie'.
Collected errors:
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.139-1-85ed76a9238319d569d47adb9fef5ca5) for kmod-cfg80211
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.139-1-85ed76a9238319d569d47adb9fef5ca5) for kmod-crypto-hash
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.139-1-85ed76a9238319d569d47adb9fef5ca5) for kmod-crypto-cmac
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.139-1-85ed76a9238319d569d47adb9fef5ca5) for kmod-crypto-null
.
.
.
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.139-1-85ed76a9238319d569d47adb9fef5ca5) for kmod-mwifiex-pcie
* pkg_hash_fetch_best_installation_candidate: Packages for kmod-mwifiex-pcie found, but incompatible with the architectures configured
* opkg_install_cmd: Cannot install package kmod-mwifiex-pcie. 

GOD, I have a compatible problem, how to resolve it?

obsy, 2022-09-08:
https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.10.139-1-85ed76a9238319d569d47adb9fef5ca5/

jow-, 2022-09-08:
This isn't a bug. Please continue the discussion in the OpenWrt forum as this seems more like a general support question.

closed 2022-09-08 by jow-

magellan-99, 2022-09-08:
> https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.10.139-1-85ed76a9238319d569d47adb9fef5ca5/

I renewed download source like this : 
'src/gz  openwrt_kmods  https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.10.139-1-85ed76a9238319d569d47adb9fef5ca5' in web console and updated opkg list. When I try to install 'kmod-mwifiex-pcie'，the same  compatible problem is still in there.

magellan-99, 2022-09-11:
> https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.10.139-1-85ed76a9238319d569d47adb9fef5ca5/

hi,obsy. I change the openwrt version to ImmortalWrt 18.06-k5.4 then all incompatible is gone. finally, I open the surface`s wifi card in openwrt succeful. after spend some time studying how to set up WiFi AP，all things is running good. thank you for your help!