openwrt/openwrt #10645: [22.03.0] 22.03.0 bricks Netgear EX6150v1 ramips MT7621

masterblaster79, 2022-09-08:
Hi folks.

I tried to install 22.03.0 on the Netgear EX6150v1 device and it fails to boot.
sysupgrade was used with the "-n" option to purge settings.

Upon reboot the device just sits there with orange LED being solid (every 30 seconds or so the LED flickers a few ms - I guess it's in a reboot loop).
Also it's not possible to enter failsafe.
The only way to recover is using nmrpflash with the 21.02.3 .chk file.

Unfortunately I do not have UART connected. I opened the device once, but I was unable to identify the correct solder pads and the wiki has no photos..

21.02.3 works flawlessly though.

WatermelonRei, 2022-09-09:
Same problem on HiWiFi HC5962.

seriv, 2022-09-13:
I have the same problem with Netgear WDNR4700, had to debrick by getting back to 21.02.3-apm821xx-nand-netgear_wrdr4700.

kroeckx, 2022-10-14:
Any news about this? Does 22.03.1 fix this?

DragonBluep, 2022-10-15:
I believe it's LZMA ERROR: 064e7e57b483e6879de0facef4f1fce86ec4ad47
Please try to add `$(Device/uimage-lzma-loader)` between line 1393 and 1394: 
https://github.com/openwrt/openwrt/blob/cb24be47ffda12616a0670e9d29a96f9530a19b8/target/linux/ramips/image/mt7621.mk#L1393-L1394

WatermelonRei, 2022-10-15:
> Any news about this? Does 22.03.1 fix this?

nope

masterblaster79, 2022-10-17:
Will applying the changes to mt7621.mk be effective for images created with ImageBuilder or is compilation required?
I tried with ImageBuilder only and it didn't work.

kroeckx, 2022-10-17:
The kernel needs to be compiled with that option.

masterblaster79, 2022-10-17:
Adding `$(Device/uimage-lzma-loader)` fixes the issue.

ynezz, 2022-10-18:
> Same problem on HiWiFi HC5962. Neither sysupgrade nor factory image can boot successfully.

@WatermelonRei are you able to test the above mentioned fix?

closed 2022-10-18 by openwrt-bot

WatermelonRei, 2022-10-18:
Yes, adding `$(Device/uimage-lzma-loader)` fixes HC5962 boot issue.

WatermelonRei, 2022-10-18:
> > Same problem on HiWiFi HC5962. Neither sysupgrade nor factory image can boot successfully.
> 
> @WatermelonRei are you able to test the above mentioned fix?

Would you like to commit this fix for HC5962 again?  I'm a novice at doing these procedure, thank for your help!

ynezz, 2022-10-18:
> > > Same problem on HiWiFi HC5962. Neither sysupgrade nor factory image can boot successfully.
> > 
> > 
> > @WatermelonRei are you able to test the above mentioned fix?
> 
> Would you like to commit this fix for HC5962 again? I'm a novice at doing these procedure, thanks for your help!

It's in b63d6d4730f.

commit 7dd1cab1 (2022-10-18): mt7621: netgear_ex6150: fix reboot loop by using LZMA loader

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of similar devices.

Fixes: #10645
commit 36a808b7 (2022-10-18): mt7621: netgear_ex6150: fix reboot loop by using LZMA loader

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of similar devices.

Fixes: #10645

(cherry picked from commit 7dd1cab1c16f374716b1ee7a5bf99f849b74c1dc)