openwrt/openwrt #10648: ramips: build error:

tiagogaspar8, 2022-09-08:
Hi,

Recently when building for a 5.15 kernel I've been having this error:
```
Packaged contents of /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-lib-lzo into /mnt/sdb/ss_mod/bin/targets/ramips/mt7620/packages/kmod-lib-lzo_5.15.64-1_mipsel_24kc.ipk
rstrip.sh: /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-libphy/lib/modules/5.15.64/libphy.ko: relocatable
Packaged contents of /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-libphy into /mnt/sdb/ss_mod/bin/targets/ramips/mt7620/packages/kmod-libphy_5.15.64-1_mipsel_24kc.ipk
Packaged contents of /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-mii into /mnt/sdb/ss_mod/bin/targets/ramips/mt7620/packages/kmod-mii_5.15.64-1_mipsel_24kc.ipk
rstrip.sh: /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-swconfig/lib/modules/5.15.64/swconfig.ko: relocatable
Packaged contents of /mnt/sdb/ss_mod/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/packages/ipkg-mipsel_24kc/kmod-swconfig into /mnt/sdb/ss_mod/bin/targets/ramips/mt7620/packages/kmod-swconfig_5.15.64-1_mipsel_24kc.ipk
Package kmod-switch-rtl8366-smi is missing dependencies for the following libraries:
of_mdio.ko
make[3]: *** [modules/netdevices.mk:400: /mnt/sdb/ss_mod/bin/targets/ramips/mt7620/packages/kmod-switch-rtl8366-smi_5.15.64-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/mnt/sdb/ss_mod/package/kernel/linux'
time: package/kernel/linux/compile#9.93#6.30#14.75
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/mnt/sdb/ss_mod'
make[1]: *** [package/Makefile:110: /mnt/sdb/ss_mod/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/sdb/ss_mod'
make: *** [/mnt/sdb/ss_mod/include/toplevel.mk:231: world] Error 2
```

This is for an archer C2 device, also, the kernel config seems to be highly out of sync causing build errors too.

Thanks.

mans0n, 2022-09-18:
Strange, I could not reproduce your build error, which config did you use for the build?

tiagogaspar8, 2022-09-18:
Hi!

I actually meant to close this, thanks for reminding me! It got fixed a few days ago with some ramips commit.

Thanks! 😁

closed 2022-09-18 by tiagogaspar8