openwrt/openwrt #10660: On TP Link Archer C50 V4 Wi-Fi is disabled by default

erayrafet, 2022-09-10:
Running 22.03, after a reset, Wi-Fi comes disabled by default. I need Ethernet to configure the router settings. It's tricky with modern laptops not having Ethernet ports.

musashino205, 2022-09-10:
I recommend building a custom image with customized config files.
see: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#custom_files

erayrafet, 2022-09-10:
> I recommend building a custom image with customized config files. see: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem#custom_files

I am very new to OpenWRT, so kinda seems complicated with a very high chance of risk. If I build my own image once, then upgrade with an official build, my default conf with wifi will be gone, right?

castillofrancodamian, 2022-09-10:
Yes.

closed 2022-09-10 by erayrafet

Openwrtfunboy, 2022-09-14:
Buy usb-ethernet adapter. It cost 2$ if u buy used.

Openwrtfunboy, 2022-09-14:
Easiest way for you bro