openwrt/openwrt #10663: TP-Link Archer C50V4 Luci exposes unsupported frequencies for 5GHz Wi-Fi

erayrafet, 2022-09-10:
As the title say, Luci shows support for 5GHz Wi-FI channels higher than 48. Sadly, the hardware doesn't support those and picking anything higher than 48 (52, 100, etc.) results in AC Wi-Fi not initializing/working.
Running version 22.03.0.

[image]
[image]
[image]

erayrafet, 2022-09-10:
Seems like V6 of the router supports those higher frequency bands (has a slightly different chip), but support for it is still WIP. This variant does not.

Shine-, 2022-09-11:
Tested channel 100 on a C50v4 and it's working perfectly fine. 5GHz Hardware (MT7612E) supports these channels. You have probably just forgotten to set your country code, so only "world" channels (36-48) are allowed in your configuration.

erayrafet, 2022-09-11:
> Tested channel 100 on a C50v4 and it's working perfectly fine. 5GHz Hardware (MT7612E) supports these channels. You have probably just forgotten to set your country code, so only "world" channels (36-48) are allowed in your configuration.
 
Well, the stock firmware doesn't allow anything higher, and OpenWRT wouldn't initialize my 5GHz with a higher channel set. Alright, I set the country code to world and nada.
[image]

erayrafet, 2022-09-11:
So, setting the country code to US fixes the "Not associated issue" while the 00 option borks it.

erayrafet, 2022-09-11:
Using channel 165 that can be selected results in the same "Not associated" message.

Shine-, 2022-09-11:
World -> 36-48 only
HW Default -> probably same as "World" for this device
Any other country code -> channels allowed as per regulatrory rules for that country.

Works as designed.

erayrafet, 2022-09-11:
Well, the US seems to allow channel 165, as well as China, the UK, Singapore which I've tested, but it's a no go. At least the issue text can be more descriptive.

Shine-, 2022-09-11:
Perhaps you have left bandwidth at "80MHz"? 165 is a 20MHz channel, so again, works as designed.

erayrafet, 2022-09-11:
Indeed, I've set the channel width to 80 which channel 165 apparently doesn't support. I've learned so many new things tonight.

Shine-, 2022-09-11:
I have to correct myself, 165 is also a 40 (165-169) and 80 MHz (165-177) channel, at least for US regulatory domain. Maybe not be supported by the hardware as such, though.

erayrafet, 2022-09-11:
@Shine- Thank you for the information. Now I do run channel 161 instead of 48. Ping seems very good.

erayrafet, 2022-09-11:
I can even increase the transmit power from the default 20mV to 22 with US set as country code.
Though, it seems like channel 14 for 2.4GHz isn't supported by the hardware.

Shine-, 2022-09-11:
14 -> Japan only.
possible TX power -> also depends on antenna gain

You might be better off asking your questions in the forum instead and close this issue, since none of this seems to be an actual issue with OpenWRT, but rather beginner's usage questions.

erayrafet, 2022-09-12:
Well, the UI could use some changes like hiding channels not supported because of the country code set. Also, getting more descriptive error messages and not like Microsoft with errors like "Something happened." Not a great user experience, even though this is software for experienced users.

erayrafet, 2022-09-12:
At 2.4GHz, channels 13-14 were hidden until I set the country code to Japan. There's definitely an issue here since I can see channels through 165 on 5GHz with country code set to world, driver default, etc. The unsupported channels are always displayed there, but not for 2.4GHz.

Shine-, 2022-09-12:
LuCI lets you select whatever the driver exposes. Also see `iw phy phy1 info` (phy1 is 5G for your device). LuCI doesn't do any sanitizing against the RegDB on its own, why should it. And the MT76 the driver **does** filter available frequencies against the country code that you have set before (unavailable frequencies showing as "disabled" in the `iw` output).

If the driver exposes unsupported frequencies for certain generic country codes (ie. "driver default" or "world"), that's usually improper EEPROM presets, ie. the device vendor's fault, not LuCI's fault.

As I said, better head over to the forum, as what you're experiencing are beginners pains, not actual issues with OpenWRT.

closed 2022-09-12 by erayrafet