openwrt/openwrt #10669: Cant find procd-ujail-console from repos

MikkoWrightson, 2022-09-10:
As far as I can tell the package should exist in the base repo but no amount of updating and banging my head to the wall helps with this. Issue persists with both luci and shell.

```root@OpenWrt:~# opkg install procd-ujail-console
Unknown package 'procd-ujail-console'.
Collected errors:
 * opkg_install_cmd: Cannot install package procd-ujail-console.
```

CleverCompiler, 2022-09-20:
I'm getting this same error.  Were you every able to work around the problem?

MikkoWrightson, 2022-09-22:
Haven't been able to figure it out, gonna switch out to the ext4 version soon and see if it works on that

CleverCompiler, 2022-09-23:
I was able to install this package by installing the version of openwrt listed on this packages page in the package library.