openwrt/openwrt #10672: kernel: status of 5.15 version

PolynomialDivision, 2022-09-11:
The status of 5.15 kernel:

- [x] airoha
- [ ] apm821xx (testing ✓)
- [ ] archs38
- [ ] armvirt (testing ✓)
- [ ] at91
- [ ] ath25
- [ ] ath79 (testing ✓)
- [x] bcm27xx
- [ ] bcm47xx
- [ ] bcm53xx (testing ✓)
- [ ] bcm63xx (testing ✓)
- [ ] bcm4908
- [x] bmips
- [x] gemini
- [x] imx
- [x] ipq40xx
- [ ] ipq806x (testing ✓)
- [ ] kirkwood (testing ✓)
- [ ] lantiq
- [ ] layerscape
- [ ] malta (testing ✓)
- [x] mediatek
- [ ] mpc85xx
- [ ] mvebu (testing ✓)
- [ ] mxs (testing ✓)
- [ ] octeon (testing ✓)
- [ ] octeontx
- [ ] omap (testing ✓)
- [ ] oxnas
- [ ] pistachio
- [ ] qoriq (testing ✓)
- [ ] ramips (testing ✓)
- [ ] realtek
- [ ] rockchip (testing ✓)
- [ ] sunxi (testing ✓)
- [ ] tegra
- [ ] uml
- [ ] x86 (testing ✓)
- [ ] zynq

Testing: 17/38

We don't even have 50% bumped to 5.15. :/ Are those targets unmaintained?

abajk, 2022-09-11:
I think armvirt and malta can be switched to 5.15. These targets are mainly for testing.

abajk, 2022-09-11:
[Tegra](https://patchwork.ozlabs.org/project/openwrt/patch/20220906132954.9585-2-tmn505@terefe.re/) and [mpc85xx](https://patchwork.ozlabs.org/project/openwrt/patch/mailman.18705.1653294067.441203.openwrt-devel@lists.openwrt.org/).

PolynomialDivision, 2022-09-11:
@abajk Thanks. Updated.

hnyman, 2022-09-13:
> think armvirt and malta can be switched to 5.15. These targets are mainly for testing.

Not quite true. You have to pay attention also for the buildbot usage.
Armvirt is used as the SDK by the package buildbot e.g. for popular ipq806x (arm_cortex-a15_neon-vfpv4)

abajk, 2022-09-15:
[Tegra](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=df59c07fe9e32bd5124871daa400a8c5e2e8d96f) and [mpc85xx](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=016c9344824c06f54412f23fa5962156edfc0d39) got the 5.15 kernel. [Armvirt](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=67144f38e7e2bf6e973ffee3043883192fb73764) and [malta](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4ed850fcb18ae894b10caa979bcc249eb92f5fc0) left the testing phase.

PolynomialDivision, 2022-09-26:
Can we pin this issue? (@aparcar @ynezz ?)

Ansuel, 2022-09-26:
Ideally ipq806x should switch with 5.15... I'm testing some cleared patch since i upstreamed lots of stuff... ipq40xx we are waiting for dsa driver and we can switch I think.

john-tho, 2022-09-29:
In 5.15 sunxi: coretexa53:
306861cf3848 ("sunxi/cortexa53: refresh kernel 5.15 config") dropped `CONFIG_RTC_DRV_SUN6I=y`. This caused my device to fail to boot the 5.15 kernel (before 79af0593a30b ("kernel: switch back 5.15 to fw_devlink=permissive")—have not tried without my kconfig changes since). I had to refresh the target & subtarget configs to get that RTC driver option to stick in the a53 kconfig, but it made an awful lot of changes, and I was not comfortable with all of them.
Happy to open a separate issue, or PR.

Not sure if this process was correct:
```
make kernel_menuconfig CONFIG_TARGET=target
 target/linux/sunxi/config-5.15 | 158 ++++++++++++++---------------------------
 1 file changed, 55 insertions(+), 103 deletions(-)
make kernel_menuconfig CONFIG_TARGET=subtarget
 target/linux/sunxi/cortexa53/config-5.15 | 54 --------------------------------
 1 file changed, 54 deletions(-)
# then enabled CONFIG_RTC_DRV_SUN6I in
make kernel_menuconfig CONFIG_TARGET=subtarget
 target/linux/sunxi/cortexa53/config-5.15 | 3 +++
 1 file changed, 3 insertions(+)
```

Would really appreciate someone detailing or providing a link for how this kconfig refresh process should work for OpenWrt, and switching to 5.15 seems like a good time to check and refresh all the subtarget kconfigs.

Cheers

ynezz, 2022-10-08:
>  Happy to open a separate issue, or PR.

Neustradamus, 2022-10-16:
To follow :)

hnyman, 2022-10-17:
ipq806x has been updated to 5.15. (table needs updating)

PolynomialDivision, 2022-10-17:
Already updated :)

CHKDSK88, 2022-10-20:
I prepare 5.15 for layerscape.

pmagid, 2022-11-04:
Anyone know what is holding up ramips (its been listed as "testing" for a while now)?

pesa1234, 2022-11-06:
mvebu
tested on wrt1900acs kernel 5.15
I don't have these issue: 
1- https://github.com/openwrt/openwrt/issues/10997
2- https://github.com/openwrt/openwrt/issues/10628

So, 5.15 no bug

drujd, 2022-11-08:
Unless someone fixes the mv88e6085 (and similar) switch bug for 5.10, mvebu absolutely needs to switch to 5.15 (working well on Turris Omnia). 22.03 is broken (aka the switch is turned into a hub) for devices with these switches due to 5.10.

sjkhsl, 2022-11-23:
So how to switch from 22.03 default 5.10 to 5.15?

hnyman, 2022-11-23:
> how to switch from 22.03 default 5.10 to 5.15?

You can't change kernel in 22.03.
22.03 will stay with 5.10

drujd, 2022-11-23:
> > how to switch from 22.03 default 5.10 to 5.15?
> 
> You can't change kernel in 22.03. 22.03 will stay with 5.10

Which unfortunately means 22.03 is and will remain broken for many devices (unless someone actually fixes it for 5.10)...

PolynomialDivision, 2022-11-23:
> > > how to switch from 22.03 default 5.10 to 5.15?
> > 
> > 
> > You can't change kernel in 22.03. 22.03 will stay with 5.10
> 
> Which unfortunately means 22.03 is and will remain broken for many devices (unless someone actually fixes it for 5.10)...

What do you mean? You can backport necessary fixes also for 5.10 kernel?

Ansuel, 2022-11-23:
just to give some solid point
- stable release will stick to the current kernel, won't be changed to point release 
- it's hard to get backport fixes for the testing kernel if that is present for the target
- it's possible that stable gets patches backported from master with enough reason

CHKDSK88, 2022-11-27:
Layerscape testing 5.15 merged: https://github.com/openwrt/openwrt/commit/23b9b01cdd612bfa46620cd505285707b164e891.

msdos03, 2022-12-01:
> 

I'm surprised that someone mentions this issue. It happens on my kirkwood device with mv88e6xxx too(the switch turns to bridge mode automatic). After turning to 5.15, it's OK until now. Kirkwood target should be switched to 5.15 kernel as soon as possible too.

PalebloodSky, 2022-12-04:
Welp now it's reverted. On Wrt32x and it was a clear improvement to these bugs. So apparently we won't be getting these get fixes for mvebu for now.

Ansuel, 2022-12-04:
@PalebloodSky it will come soon enough... we notice some problem with some packages and the change was not correct... So it's easier to revert and push the correct commit later.

satmandu, 2022-12-28:
bcm2711 needs a patch to fix ethernet working on 5.15 using the default Broadcom ethernet interface: https://github.com/openwrt/openwrt/issues/11608#issuecomment-1364676840

PolynomialDivision, 2023-01-03:
I still have those two PRs ready to be merged:
- https://github.com/openwrt/openwrt/pull/10881
- https://github.com/openwrt/openwrt/pull/10885

PolynomialDivision, 2023-01-04:
Only 4 targets left that don't have 5.15 testing support:
- archs38
- ath25
- bcm47xx
- pistachio

abajk, 2023-01-04:
[Uml](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=50d20db6a8fd2aaed3322578c6e5a2960a8d3f4d) already uses the 5.15 kernel.

Falcon-118, 2023-01-21:
> just to give some solid point
> 
>     * stable release will stick to the current kernel, won't be changed to point release
>     * it's possible that stable gets patches backported from master with enough reason

Just curious (I came from a different IT world)...  how can it still be considered a "stable release" with such a glaring fault?
Especially with a fix already available.   Seems there is more concern for the process than the product...  isn't the product "enough reason"?

Don't take that wrong, Chris Christian...  I certainly applaud YOUR efforts.  Maybe it was just a choice of words.  I'd certainly do worse in Italian.

Ansuel, 2023-01-21:
Again the fact that the problem is not present in 5.15 means that it's totally possible to backport the patch to 5.10. Stable means that we don't introduce major changes in the release. If we would start introducing these kind of big changes then it makes no sense to have all the release system and the feature frezee with branching from master and we would just tag stuff from master and be done with it.

Again if the community care enough kernel bisect are difficult but not impossible. Once the affecting patch is found we would be more than happy to backport it to openwrt-22.03

champtar, 2023-01-21:
@Falcon-118 Turris Omnia works fine for me with 5.10 and bridge vlan filtering (IE not default config), but same config doesn't work in 5.15, that's why stable release don't get massive changes

PolynomialDivision, 2023-02-01:
@robimarko What should we do with the ath25? Can we now just ignore it?

PolynomialDivision, 2023-02-03:
Yey, after merging https://github.com/openwrt/openwrt/commit/18e6df83bb92dc095da237302de82ef6b4bc62bf we have all testing kernels (ath25 is dead?) on 5.15.

sss132919, 2023-02-23:
the linux kernel add 6.1 version to the longterm ,  is the openwrt.org will upgrade its kernel?

robimarko, 2023-02-23:
6.1 generic support will only be able to get merged after next stable is branched off, currently that should happen at the end of March

Borromini, 2023-02-23:
@aparcar Ramips has been migrated to 5.15 see https://github.com/openwrt/openwrt/commit/77d061167cf8a7edb3ea7cbc9926f25c8ac8fe1e, so you can check that one off the list.

melroy89, 2023-03-01:
Hopefully Banana Pi BPI-R3 will be supported soon with 23.x release!

udanfacy22, 2023-03-07:
At91 has left the testing phase. See acb1891.

aparcar, 2023-03-14:
Just eight more to go...

Falcon-118, 2023-03-14:
> Just eight more to go...

And then what?   Is this moving to a 23.x RC?

My thanks to all of you.  I know this was an unexpected failure you are having to work through.  Progress has been quick.

udanfacy22, 2023-03-21:
> > Just eight more to go...
> 
> And then what? Is this moving to a 23.x RC?

It looks like it, yes. However, before they get to release candidates, first they'll have to branch from master. Sometimes there can be can a gap of several months between between branch creation and the first RC. It just depends.

Here's what I've been able to find about what they're working on:

1. https://openwrt.org/meetings/20230124
2. https://openwrt.org/meetings/20230221
3. https://openwrt.org/docs/guide-developer/releases/goals/23.xx

> I know this was an unexpected failure you are having to work through. 

I'm not a developer. Out of mild curiosity, what are you referring to here?

Falcon-118, 2023-03-22:
> 
> > I know this was an unexpected failure you are having to work through.
> 
> I'm not a developer. Out of mild curiosity, what are you referring to here?

Thanks for the links!  interesting reading.
I was referring to the network switch problem discovered in the Kernel.  It seems to have caught everyone by surprise.  But, it's found, It appears to be fixed and we move forward.

Openwrtfunboy, 2023-04-12:
What are the advantages for routers in the kernel 5.15?

melroy89, 2023-04-12:
> What are the advantages for routers in the kernel 5.15?

Well.. A lot. I took the time to answer your question.

- Lot of ARM chip improvements/added support for chips like Allwinner, Rockchip, Amlogic, Qualcomm, MediaTek, NXP, etc. As wel as improved RISC-V support (anybody?);
- Better kernel driver/modules support. For example Realtek RTL8188EU wifi driver;
- Lot of networking improvements various places from bridge networks, iwlwifi, ethtool, IPv6, multi-path TCP, wwan, you name it;
- Better schedular;
- Improved memory management;
- Improved power management;
- Improver storage capabilities as well as file systems (some people might use the router as a simple NAS as well). Like NTFS, XFS, EXT4, Btrfs, NFS and even SMB 3 server ..;
- CPU frequency scaling improvements for qcom and mediatek;
- RTC/Clock improvements (various rockchip and qcom supported chips);
- Pin controller support/drivers;
- And much more various improvements!!!

Borromini, 2023-04-13:
@gentoo-root Let's stay on-topic which is tracking 5.15 status in master, not discussing its merits or its effects on particular devices.

For anything broken on QCA stuff, you usually have to thank them for that. Its attitude versus FOSS is well known by now.

rmilecki, 2023-04-13:
We have to update kernel used by OpenWrt because of:
1. Development<br>All new drivers and features appear in new kernels. There is limited backporting effort we can handle. It's impossible to backport a lot of net features to few-years-old kernels.
2. Support<br>Old kernel releases stop being support at some point. They stop receiving security fixes.

So updating kernel (usually every release) it a must-do.

Of course regressions do happen. If your hardware is affected you can:
1. Help<br>Debug / bisect / report / fix it.
2. Use older release<br>We usually support older OpenWrt releases for few years

Ansuel, 2023-05-03:
@aparcar we can proceed with branching i think

aparcar, 2023-05-03:
Yes feel free to do so

Neustradamus, 2023-05-22:
Please note that @Ansuel has added Kernel 6.1 support:
- https://github.com/openwrt/openwrt/pull/11011

Djfe, 2023-05-29:
@PolynomialDivision should we close this?
Everything in the first post has been checked

closed 2023-05-29 by aparcar