openwrt/openwrt #10684: 22.03 BRICKS ROUTER!!!

Openwrtfunboy, 2022-09-12:
Hello! When updating with OpenWRT 21.02 to 22.03 via Web/Cli, I ran into the fact that the device became inaccessible, I reset the settings through the Reset button, but this did not help. Then I flash through the TFTP 22.03 carefully checked sha256sum, but the device was still inaccessible. After I reflash through TFTP 21.02 and the device finally start works as it should, as before. I believe that this is a CRITICAL bug and should be fixed as soon as possible!!!
Device: ZyXEL Keenetic Lite Rev.B

Openwrtfunboy, 2022-09-12:
[image]

Hurricos, 2022-09-14:
Too little RAM (32MB).

Openwrtfunboy, 2022-09-14:
> Too little RAM (32MB).

I wouldn't update if there wasn't an update for my device. So remove 22.03, remove the device from the supported ones, than break the users of the device. Imagine: a person wants to try OWRT for the first time, puts it on the device, but OWRT does not work AT ALL. What emotions will he get? What will he think? It is necessary to either fix or remove this device from the supported 22.03.

Openwrtfunboy, 2022-09-14:
There is so much negativity in the world, and situations like this create it.

Openwrtfunboy, 2022-09-14:
Please! Delete this version ASAP from device page!!! Until this get fixed.

https://openwrt.org/toh/hwdata/zyxel/zyxel_keenetic_lite_rev._b

Openwrtfunboy, 2022-09-14:
I can flash with TFTP, and I have the skills, but many users will simply say that the router is broken and throw it away. This is wrong and sad.

Hurricos, 2022-09-14:
True, we should disable this from building. The toh page you linked does say that:

> 19.07 will be the last official build for 4/32 devices. After 19.07, no further OpenWrt images will be built for 4/32 devices. See [OpenWrt on 4/32 devices](https://openwrt.org/supported_devices/openwrt_on_432_devices) what you can do now.
>  2) OpenWrt support for this device will end after 2019.

I think the device might work if not given luci / uhttpd to deal with, but it should be built with different flags as it stands.

Openwrtfunboy, 2022-09-14:
Yes. But builds still comings. Also I have many 8/32 devices that works perfect. So I think this is bug. BTW 21.02 works perfect.

Openwrtfunboy, 2022-09-14:
> True, we should disable this from building. The toh page you linked does say that:
>
>
> I think the device might work if not given luci / uhttpd to deal with, but it should be built with different [...]

I'm leaning that this is a bug because 21.02 works just fine. Also I have other 8/32 devices that work very well on 22.03 for their purposes

Openwrtfunboy, 2022-09-14:
> True, we should disable this from building. The toh page you linked does say that:
>
>
> I think the device might work if not given luci / uhttpd to deal with, but it should be built with different [...]

Btw, initial support version was 21.02 lol

CaiMiao, 2022-09-16:
for these newly compatible 4/32 devices, i suggest providing builds one-time (if that build is confirmed working) and drop support immediately after that. it's their freedom of choice to backport to a older version, however.

Openwrtfunboy, 2022-10-02:
issue still issue. Will openwrt will do anything?