openwrt/openwrt #10687: iwinfo reporting mesh unencrypted (none) instead of WPA3-SAE

quantumbeat, 2022-09-12:
In 22.03.0, 'sae' encrypted mesh (802.11s) is displayed unencrypted

[image]

while iwinfo phy scan shows mesh encrypted (all mesh points against each other)

Cell 20 - Address: XX:XX:XX:XX:XX:XX
          ESSID: "Meshname"
          Mode: Mesh Point  Channel: 100
          Signal: -61 dBm  Quality: 49/70
          Encryption: WPA3 SAE (CCMP)
          HT Operation:
                    Primary Channel: 100
                    Secondary Channel Offset: no secondary
                    Channel Width: unknown
          VHT Operation:
                    Channel Width: unknown
                    Center Frequency 1: 100
                    Center Frequency 2: 0

iwinfo phy info instead shows unencrypted (and probably in turn displayed in luci like this)

phy1      ESSID: "Meshname"
          Access Point: XX:XX:XX:XX:XX:XX
          Mode: Mesh Point  Channel: 100 (5.500 GHz)
          Center Channel 1: 100 2: unknown
          Tx-Power: 8 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

This issue reported against luci might be a duplicate: https://github.com/openwrt/luci/issues/5954 
This forum thread might be related: https://forum.openwrt.org/t/openwrt-22-03-mesh-not-encrypted/136420/5

quantumbeat, 2022-09-12:
I can test, would compile from $branch + flash, if helpful, let me know

Ansuel, 2022-09-13:
@quantumbeat can you confirm the mesh is working? I notice on my testing setup that when the mesh is up with no peer the encryption is none but as soon as another mesh connects, the encryption is correctly displayed as WPA3

Ansuel, 2022-09-13:
@quantumbeat also can you compile a firmware with the wpa-cli package and can you give me the output of `wpa_cli status`

mjsage, 2022-09-13:
I see the same behaviour in 22.x, the mesh points are both connected.  
In 19.07 Luci as I recall showed open when the mesh point was not connected and   WPA3-SAE when connected.
I believe in 18.x it showed encryption: open in luci connected, or not 

# iwinfo phy1 info
phy1      ESSID: "mymesh1"
          Access Point: XXXXXXXXXX
          Mode: Mesh Point  Channel: 13 (2.472 GHz)
          Center Channel 1: 11 2: unknown
          Tx-Power: 18 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 168C:002D 168C:209A [Atheros AR9287]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

# iwinfo phy0 info
phy0      ESSID: "mymesh2"
          Access Point: XXXXXXXXXXXXXX
          Mode: Mesh Point  Channel: 48 (5.240 GHz)
          Center Channel 1: 42 2: unknown
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 168C:003C 0000:0000 [Qualcomm Atheros QCA9880]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

(There's also a bug in wireless scan/ channel analysis in Luci timing out if a mesh point is enabled on the radio, but scan works fine from iwinfo.)

quantumbeat, 2022-09-14:
> @quantumbeat can you confirm the mesh is working? I notice on my testing setup that when the mesh is up with no peer the encryption is none but as soon as another mesh connects, the encryption is correctly displayed as WPA3

yes it works and displays "none" even though another mesh point is connect.
Besides the display, it works flawlessly

quantumbeat, 2022-09-14:
> @quantumbeat also can you compile a firmware with the wpa-cli package and can you give me the output of `wpa_cli status`

yes, but I need a couple days, probably the weekend

Ansuel, 2022-09-14:
I really need the wpa_cli output as that is the source of our data

quantumbeat, 2022-09-17:
> @quantumbeat also can you compile a firmware with the wpa-cli package and can you give me the output of `wpa_cli status`

root@node:~# wpa_cli status
Selected interface 'mesh0'
bssid=00:00:00:00:00:00
freq=5500
ssid=Meshname
id=0
mode=mesh
pairwise_cipher=UNKNOWN
group_cipher=UNKNOWN
key_mgmt=UNKNOWN
wpa_state=COMPLETED
address=9c:3d:cf:d3:6e:14
uuid=b3fe0116-8178-53de-953b-1465088e12bd

quantumbeat, 2022-09-23:
Anything I could do here? @Ansuel

dzenc, 2022-09-28:
I am seeing the same behavior as @quantumbeat ; and I can confirm that my wpa_cli output is similar: 00's for bssid, UNKNOWN for ciphers.  And Luci says the mesh is unencrypted.  Please let me know if I can provide any additional info.

dzenc, 2022-09-30:
FYI @quantumbeat, I can see that @Ansuel is currently deep in the middle of debugging an intermittent crash https://github.com/openwrt/openwrt/pull/10703 (Thanks!)

I'm sure he'll get back to this after that is resolved.

McGiverGim, 2022-10-04:
@Ansuel I've observed the same with the Xiaomi AX3600. I know it is not an official target, but you own one and maybe can use it to reproduce the problem.

dimitris-personal, 2022-10-20:
FWIW: I can reproduce this with a mesh between two nodes running 22.03.2:

- Netgear R7800 (using -ct atheros drivers/firmware, wpad-wolfssl)
- Buffalo WZR-HP-AG300H (wpad-wolfssl)

Encryption is set at WPA3/SAE on the mesh.  With matching keys, traffic flows normally.

With mismatched keys (just one character appended on the Netgear side), the Buffalo is still shown as "associated" on the UI.  However, traffic is not being received from it on the Netgear interface (ping: ARP requests are sent but no replies).  Setting the password back to matching restores traffic.

In all cases encryption is shown as "none" in the UI.

`iw scan` results for the mesh, captured from the Buffalo side:

```
BSS xx:xx:xx:xx:xx:xx
	TSF: 2726092801 usec (0d, 00:45:26)
	freq: 2412
	beacon interval: 100 TUs
	capability: (0x0010)
	signal: -43.00 dBm
	last seen: 750 ms ago
	SSID: 
	RSN:	 * Version: 1
		 * Group cipher: CCMP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: SAE
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
	HT capabilities:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-31
	HT operation:
		 * primary channel: 1
		 * secondary channel offset: no secondary
		 * STA channel width: 20 MHz
	MESH ID: foo
	VHT capabilities:
		VHT Capabilities (0x339b79b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformer
			SU Beamformee
			MU Beamformer
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
	VHT operation:
		 * channel width: 0 (20 or 40 MHz)
		 * center freq segment 1: 1
		 * center freq segment 2: 0
		 * VHT basic MCS set: 0xffff
```

ffries, 2022-11-02:
I can report the same issue.

iwinfo phy1 scan
reports mesh0 is encrypted with WPA3-SAE

But Luci shows that mesh0 is unencrypted.

jow-, 2022-11-02:
As stated above, `wpa_cli` reports:

> pairwise_cipher=UNKNOWN
> group_cipher=UNKNOWN
> key_mgmt=UNKNOWN

The iwinfo utility acquires it's information from the wpa-supplicant control socket as well. So if WPA supplicant itself is not reporting the crypto correctly, iwinfo will fail too.

dhewg, 2022-11-14:
I ran into this too, this fixes it for me:
http://patchwork.ozlabs.org/project/openwrt/patch/20221113203346.3470333-1-a.heider@gmail.com/

dhewg, 2022-11-15:
Is anyone able to test that patch and check if it fixes the problem?
A `Tested-by` tag would be helpful! If it's easier I can provide an OpenWrt branch containing that patch.

dhewg, 2022-11-18:
The fix is part of https://github.com/openwrt/openwrt/pull/11280
Please test :)

dhewg, 2022-11-25:
Just stumbled upon 398cbb76fa88f2d5dcc05b0ecc24ced1bbcad9ca, which may be the cause of this regression.

Ansuel, 2022-12-12:
Totally missed this... Can someone summarize?

McGiverGim, 2022-12-13:
@Ansuel as far as I understand, there is a patch that seems to fix the issue: http://patchwork.ozlabs.org/project/openwrt/patch/20221113203346.3470333-1-a.heider@gmail.com/

This patch is part of this PR: https://github.com/openwrt/openwrt/pull/11280 but I'm not too sure if it will be merged into OpenWrt or in iwinfo. @dhewg can you give more information?

dhewg, 2022-12-13:
> Totally missed this... Can someone summarize?

We talked about this and the patch fixing this ~3 weeks ago? That's `[PATCH 09/39] nl80211: fix wpa supplicant ctrl socket permissions` from #11280. But that's just one issue with iwinfo, there're many more problems which that PR fixes

closed 2022-12-15 by openwrt-bot

dimitris-personal, 2023-01-09:
Thanks!  LGTM on 22.03.3.

dimitris-personal, 2025-01-09:
This has regressed in [24.10.0-rc5](https://forum.openwrt.org/t/openwrt-24-10-0-rc5-fifth-release-candidate/220854/46).

dimitris-personal, 2025-02-04:
Built the [24.10.0 tag](https://github.com/openwrt/openwrt/releases/tag/v24.10.0) from source, this still reproduces on Buffalo WZR-HP-AG300H (ath79) and Netgear WAX206 (mt7622).

chrdev, 2025-04-13:
OpenWrt 24.10.0 r28427-6df0e3d02a on OpenWrt One has the same issue.

dimitris-personal, 2025-04-15:
Still present on 24.10.1.

joe-sydney, 2025-08-09:
Also on 24.10.2. Will anyone look into this since this particular issue # is closed? Or do we need to open a new issue?

chrdev, 2025-08-16:
Fixed on 24.10.2 for filogic devices.

I was wrong to post that source by a skim. It's not like that to fix. Sorry.