openwrt/openwrt #10692: SSL certificate checking fails for downloads (at least with wget)

hnyman, 2022-09-12:
@Ansuel  @roysjosh 

The new practice introduced by 90c6e3aed of checking SLL certs by default at source downloads seems to fail at least for some downloads.

I noticed that yesterday when I tried to build new versions after kernel bumps.
The download completes, but gets somehow rejected without any clear error message. Just "Download failed".
There is also a strange error line:  `http://: Invalid host name.`  (from wget?)

Toggling the new option DOWNLOAD_CHECK_CERTIFICATE to disabled makes downloads to work again. (and no invalid host message)

I first noticed this with kernel, but then tested with "nano", and downloading its sources fail similarly.

Note that my Ubuntu buildhost has wget, but does not have curl.
This might also be something related the options passed to wget.
I do have ca-certificates installed.

------

See below. Download first fails twice, then I toggle the option in config, and then it suddenly succeeds.

```
perus@ub2204:/Openwrt/r7800$ make download V=s
...
make[2]: Leaving directory '/Openwrt/r7800/toolchain/gdb'
time: toolchain/gdb/download#0.00#0.00#0.01
make[2]: Entering directory '/Openwrt/r7800/toolchain/kernel-headers'
mkdir -p /Openwrt/r7800/dl
SHELL= flock /Openwrt/r7800/tmp/.linux-5.10.142.tar.xz.flock -c '  	/Openwrt/r7800/scripts/download.pl "/Openwrt/r7800/dl" "linux-5.10.142.tar.xz" "3f47ebdb9afe152a0c32c1157336ef13fa5cc08ac6d884dfc1f6ddc2b7dba268" "" "@KERNEL/linux/kernel/v5.x"    '
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=-  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
http://: Invalid host name.
--2022-09-12 16:56:17--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.245.176, 2a04:4e42:3a::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.245.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 120490684 (115M) [application/x-xz]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  0% 27.8M 4s
    50K .......... .......... .......... .......... ..........  0%  155M 2s
   100K .......... .......... .......... .......... ..........  0% 29.8M 3s
   150K .......... .......... .......... .......... ..........  0%  122M 2s
...
117550K .......... .......... .......... .......... .......... 99%  503M 0s
117600K .......... .......... .......... .......... .......... 99% 29.4M 0s
117650K .......... ......                                     100% 17.5M=5.2s

2022-09-12 16:56:23 (22.2 MB/s) - written to stdout [120490684/120490684]

FINISHED --2022-09-12 16:56:23--
Total wall clock time: 5.4s
Downloaded: 1 files, 115M in 5.2s (22.2 MB/s)
Download failed.
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=-  https://download.xs4all.nl/ftp.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
http://: Invalid host name.
--2022-09-12 16:56:23--  https://download.xs4all.nl/ftp.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
Resolving download.xs4all.nl (download.xs4all.nl)... failed: Name or service not known.
wget: unable to resolve host address 'download.xs4all.nl'
Download failed.
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=-  https://mirrors.mit.edu/kernel/linux/kernel/v5.x/linux-5.10.142.tar.xz
http://: Invalid host name.
--2022-09-12 16:56:23--  https://mirrors.mit.edu/kernel/linux/kernel/v5.x/linux-5.10.142.tar.xz
Resolving mirrors.mit.edu (mirrors.mit.edu)... 18.7.29.125
Connecting to mirrors.mit.edu (mirrors.mit.edu)|18.7.29.125|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 120490684 (115M) [application/octet-stream]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  0%  331K 5m56s
    50K .......... .......... .......... .......... ..........  0%  552K 4m44s
   100K .......... .......... .......... .......... ..........  0% 1013K 3m48s
   150K .......... .......... .......... .......... ..........  0% 1.33M 3m12s
...
117550K .......... .......... .......... .......... .......... 99% 8.65M 0s
117600K .......... .......... .......... .......... .......... 99% 8.44M 0s
117650K .......... ......                                     100% 21.7M=12s

2022-09-12 16:56:35 (9.48 MB/s) - written to stdout [120490684/120490684]

FINISHED --2022-09-12 16:56:35--
Total wall clock time: 13s
Downloaded: 1 files, 115M in 12s (9.48 MB/s)
Download failed.
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=-  http://ftp.nara.wide.ad.jp/pub/kernel.org/linux/kernel/v5.x/linux-5.10.142.tar.xz
http://: Invalid host name.
--2022-09-12 16:56:35--  http://ftp.nara.wide.ad.jp/pub/kernel.org/linux/kernel/v5.x/linux-5.10.142.tar.xz
Resolving ftp.nara.wide.ad.jp (ftp.nara.wide.ad.jp)... 203.178.137.175, 2001:200:0:1::800:21
Connecting to ftp.nara.wide.ad.jp (ftp.nara.wide.ad.jp)|203.178.137.175|:80... connected.
...

perus@ub2204:/Openwrt/r7800$ diff .config.old  .config
192c192
< CONFIG_DOWNLOAD_CHECK_CERTIFICATE=y
---
> # CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set

perus@ub2204:/Openwrt/r7800$ make download V=s
make[1]: Entering directory '/Openwrt/r7800/scripts/config'
...
make[2]: Leaving directory '/Openwrt/r7800/toolchain/gdb'
time: toolchain/gdb/download#0.00#0.00#0.00
make[2]: Entering directory '/Openwrt/r7800/toolchain/kernel-headers'
mkdir -p /Openwrt/r7800/dl
SHELL= flock /Openwrt/r7800/tmp/.linux-5.10.142.tar.xz.flock -c '  	/Openwrt/r7800/scripts/download.pl "/Openwrt/r7800/dl" "linux-5.10.142.tar.xz" "3f47ebdb9afe152a0c32c1157336ef13fa5cc08ac6d884dfc1f6ddc2b7dba268" "" "@KERNEL/linux/kernel/v5.x"    '
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=- --no-check-certificate https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
--2022-09-12 16:59:06--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.245.176, 2a04:4e42:3a::432
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.245.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 120490684 (115M) [application/x-xz]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  0% 14.2M 8s
    50K .......... .......... .......... .......... ..........  0% 11.1M 9s
   100K .......... .......... .......... .......... ..........  0% 64.3M 7s
   150K .......... .......... .......... .......... ..........  0% 55.1M 6s
...
117550K .......... .......... .......... .......... .......... 99% 18.3M 0s
117600K .......... .......... .......... .......... .......... 99% 36.1M 0s
117650K .......... ......                                     100%  130M=5.1s

2022-09-12 16:59:11 (22.3 MB/s) - written to stdout [120490684/120490684]

make[2]: Leaving directory '/Openwrt/r7800/toolchain/kernel-headers'
time: toolchain/kernel-headers/download#0.62#0.87#5.67
make[2]: Entering directory '/Openwrt/r7800/toolchain/binutils'
make[2]: Nothing to be done for 'download'.
...

hnyman, 2022-09-12:
I tried changing the empty '' into a space ' ' on the wget command, and that also causes failure, but differently.
```
-               : (qw(wget --tries=5 --timeout=20 --output-document=-), $check_certificate ? '' : '--no-check-certificate', shellwords($ENV{WGET_OPTIONS} || ''), $url)
+               : (qw(wget --tries=5 --timeout=20 --output-document=-), $check_certificate ? ' ' : '--no-check-certificate', shellwords($ENV{WGET_OPTIONS} || ''), $url)
```
But that causes again a different error regarding the invalid host.
```
 http://%20/
Resolving   ( )... failed: Name or service not known.
wget: unable to resolve host address ' '
```
To me it now looks like the empty '' parameter gets handled wrongly by wget / passed wrongly to wget. 
It gets interpreted as a host and the SSL cert check is possibly done against that, instead of the actually download host.

```
+ wget --tries=5 --timeout=20 --output-document=-   https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
--2022-09-12 17:29:43--  http://%20/
Resolving   ( )... failed: Name or service not known.
wget: unable to resolve host address ' '
--2022-09-12 17:29:43--  https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
Resolving mirror.csclub.uwaterloo.ca (mirror.csclub.uwaterloo.ca)... 129.97.134.71, 2620:101:f000:4901:c5c:0:f:1055
Connecting to mirror.csclub.uwaterloo.ca (mirror.csclub.uwaterloo.ca)|129.97.134.71|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1593804 (1.5M) [text/plain]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  3%  419K 4s
    50K .......... .......... .......... .......... ..........  6%  425K 3s
...
  1500K .......... .......... .......... .......... .......... 99%  608M 0s
  1550K ......                                                100%  177M=0.6s

2022-09-12 17:29:51 (2.40 MB/s) - written to stdout [1593804/1593804]

FINISHED --2022-09-12 17:29:51--
Total wall clock time: 7.0s
Downloaded: 1 files, 1.5M in 0.6s (2.40 MB/s)
Download failed.

```

ynezz, 2022-09-12:
Can you try this fix?

```diff
diff --git a/scripts/download.pl b/scripts/download.pl
index ab1801aad5ae..9f88de7381f6 100755
--- a/scripts/download.pl
+++ b/scripts/download.pl
@@ -82,8 +82,14 @@ sub download_cmd($) {
 	}
 
 	return $have_curl
-		? (qw(curl -f --connect-timeout 20 --retry 5 --location), $check_certificate ? '' : '--insecure', shellwords($ENV{CURL_OPTIONS} || ''), $url)
-		: (qw(wget --tries=5 --timeout=20 --output-document=-), $check_certificate ? '' : '--no-check-certificate', shellwords($ENV{WGET_OPTIONS} || ''), $url)
+		? (qw(curl -f --connect-timeout 20 --retry 5 --location),
+			shellwords($check_certificate ? '' : '--insecure'),
+			shellwords($ENV{CURL_OPTIONS} || ''),
+			$url)
+		: (qw(wget --tries=5 --timeout=20 --output-document=-),
+			shellwords($check_certificate ? '' : '--no-check-certificate'),
+			shellwords($ENV{WGET_OPTIONS} || ''),
+			$url)
 	;
 }
```

Ansuel, 2022-09-12:
With nano I have 
```
make[2]: Entering directory '/home/ansuel/openwrt-external/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/ansuel/openwrt-external/openwrt/scripts/config'
time: target/linux/prereq#0.12#0.00#0.11
make[1]: Entering directory '/home/ansuel/openwrt-external/openwrt'
make[2]: Entering directory '/home/ansuel/openwrt-external/openwrt/feeds/packages/utils/nano'
mkdir -p /home/ansuel/openwrt-external/openwrt/dl
SHELL= flock /home/ansuel/openwrt-external/openwrt/tmp/.nano-6.4.tar.xz.flock -c '      /home/ansuel/openwrt-external/openwrt/scripts/download.pl "/home/ansuel/openwrt-external/openwrt/dl" "nano-6.4.tar.xz" "4199ae8ca78a7796de56de1a41b821dc47912c0307e9816b56cc317df34661c0" "" "@GNU/nano"    '
+ curl -f --connect-timeout 20 --retry 5 --location  https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
curl: (3) URL using bad/illegal format or missing URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1556k  100 1556k    0     0   853k      0  0:00:01  0:00:01 --:--:--  853k
make[2]: Leaving directory '/home/ansuel/openwrt-external/openwrt/feeds/packages/utils/nano'
time: package/feeds/packages/nano/tiny/download#0.24#0.03#1.95
make[1]: Leaving directory '/home/ansuel/openwrt-external/openwrt'
```

Ansuel, 2022-09-12:
With @ynezz fix
```
make[2]: Entering directory '/home/ansuel/openwrt-external/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/ansuel/openwrt-external/openwrt/scripts/config'
make[1]: Entering directory '/home/ansuel/openwrt-external/openwrt'
make[2]: Entering directory '/home/ansuel/openwrt-external/openwrt/feeds/packages/utils/nano'
mkdir -p /home/ansuel/openwrt-external/openwrt/dl
SHELL= flock /home/ansuel/openwrt-external/openwrt/tmp/.nano-6.4.tar.xz.flock -c '      /home/ansuel/openwrt-external/openwrt/scripts/download.pl "/home/ansuel/openwrt-external/openwrt/dl" "nano-6.4.tar.xz" "4199ae8ca78a7796de56de1a41b821dc47912c0307e9816b56cc317df34661c0" "" "@GNU/nano"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1556k  100 1556k    0     0   809k      0  0:00:01  0:00:01 --:--:--  810k
make[2]: Leaving directory '/home/ansuel/openwrt-external/openwrt/feeds/packages/utils/nano'
time: package/feeds/packages/nano/tiny/download#0.17#0.07#2.05
make[1]: Leaving directory '/home/ansuel/openwrt-external/openwrt'
```

hnyman, 2022-09-12:
Thanks.
wget downloads seems to work with the fix from @ynezz 

```
perus@ub2204:/Openwrt/r7800$ patch -p 1 -i p.patch 
patching file scripts/download.pl

perus@ub2204:/Openwrt/r7800$ make package/nano/download V=s
make[2]: Entering directory '/Openwrt/r7800/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Openwrt/r7800/scripts/config'
make[1]: Entering directory '/Openwrt/r7800'
make[2]: Entering directory '/Openwrt/r7800/feeds/packages/utils/nano'
mkdir -p /Openwrt/r7800/dl
SHELL= flock /Openwrt/r7800/tmp/.nano-6.4.tar.xz.flock -c '  	/Openwrt/r7800/scripts/download.pl "/Openwrt/r7800/dl" "nano-6.4.tar.xz" "4199ae8ca78a7796de56de1a41b821dc47912c0307e9816b56cc317df34661c0" "" "@GNU/nano"    '
Can't exec "curl": No such file or directory at /Openwrt/r7800/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --output-document=- https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
--2022-09-12 19:47:46--  https://mirror.csclub.uwaterloo.ca/gnu/nano/nano-6.4.tar.xz
Resolving mirror.csclub.uwaterloo.ca (mirror.csclub.uwaterloo.ca)... 129.97.134.71, 2620:101:f000:4901:c5c:0:f:1055
Connecting to mirror.csclub.uwaterloo.ca (mirror.csclub.uwaterloo.ca)|129.97.134.71|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1593804 (1.5M) [text/plain]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  3% 65.2K 23s
    50K .......... .......... .......... .......... ..........  6%  402K 13s
...
  1500K .......... .......... .......... .......... .......... 99% 85.1M 0s
  1550K ......                                                100%  281M=1.3s

2022-09-12 19:47:48 (1.17 MB/s) - written to stdout [1593804/1593804]

make[2]: Leaving directory '/Openwrt/r7800/feeds/packages/utils/nano'
time: package/feeds/packages/nano/full/download#0.14#0.03#2.62
make[1]: Leaving directory '/Openwrt/r7800'
```

I did not test curl. (but ansuel seems to have already done that)

commit 50a48faa (2022-09-13): scripts/download.pl: fix downloads with wget

Several users of wget for downloads (curl is not available in the
system) have reported broken download functionality:

 wget --tries=5 --timeout=20 --output-document=-  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
 http://: Invalid host name.

Thats all happening due to '' was passed as an argument, which got later
expanded to http://.

In the context of a list constructor '' is not nothing, it is an empty
string element.  So fix it by using () as it will yield "nothing" and
thus not introduce an empty string element.

Fixes: [... cut]
commit 81388f74 (2022-09-16): scripts/download.pl: fix downloads with wget

Several users of wget for downloads (curl is not available in the
system) have reported broken download functionality:

 wget --tries=5 --timeout=20 --output-document=-  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
 http://: Invalid host name.

Thats all happening due to '' was passed as an argument, which got later
expanded to http://.

In the context of a list constructor '' is not nothing, it is an empty
string element.  So fix it by using () as it will yield "nothing" and
thus not introduce an empty string element.

Fixes: [... cut]
commit c07c565e (2022-09-21): scripts/download.pl: fix downloads with wget

Several users of wget for downloads (curl is not available in the
system) have reported broken download functionality:

 wget --tries=5 --timeout=20 --output-document=-  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
 http://: Invalid host name.

Thats all happening due to '' was passed as an argument, which got later
expanded to http://.

In the context of a list constructor '' is not nothing, it is an empty
string element.  So fix it by using () as it will yield "nothing" and
thus not introduce an empty string element.

Fixes: [... cut]
closed 2022-09-13 by jow-