openwrt/openwrt #10695: udhcpc: no lease, failing

lansky123, 2022-09-13:
1、version  Xiaomi Mi Router 4A Gigabit Edition 22.03.0-rc6 (r19590-042d558536)
2、 append the follow to dnsmasq.conf.
          conf-dir=/etc/dnsmasq.d/,*.conf
3、 ls dnsmasq.d/
        dnsmasq_list.conf 
4、 cat dnsmasq.d/dnsmasq_list.conf
server=/zzcartoon.com/8.8.8.8
ipset=/zzcartoon.com/gfwlist
server=/zzcloud.me/8.8.8.8
ipset=/zzcloud.me/gfwlist
server=/zzux.com/8.8.8.8
ipset=/zzux.com/gfwlist
 
root@OpenWrt:/etc/dnsmasq.d# killall dnsmasq
root@OpenWrt:/etc/dnsmasq.d# /etc/init.d/dnsmasq restart
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
udhcpc: no lease, failing

nslookup www.google.com
nslookup: write to '127.0.0.1': Connection refused
nslookup: write to '::1': Connection refused
;; connection timed out; no servers could be reached

dave14305, 2022-09-13:
See https://github.com/openwrt/openwrt/issues/10625

lansky123, 2022-09-14:
@dave14305 delete conf-dir in dnsmasq.conf and add to dhcp. But "udhcpc: no lease, failing"  is not fited . nslookup is ok.

lansky123, 2022-09-14:
delete conf-dir in dnsmasq.conf and add to dhcp. But "udhcpc: no lease, failing"  is not fited . nslookup is ok.

lansky123, 2022-09-14:
root@OpenWrt:/etc/config# /etc/init.d/firewall start
Reference error: left-hand side expression is null
In [anonymous function](), file /usr/share/ucode/fw4.uc, line 2294, byte 72:
  called from function [arrow function] (/usr/share/ucode/fw4.uc:703:65)
  called from function foreach ([C])
  called from function [anonymous function] (/usr/share/ucode/fw4.uc:703:66)
  called from function render_ruleset (/usr/share/firewall4/main.uc:54:24)
  called from anonymous function (/usr/share/firewall4/main.uc:138:28)

 `            ipset = filter(this.state.ipsets, s => (s.name == rule.ipset.name))[0];`
  Near here ----------------------------------------------------------------------^

Unable to read firewall state - do you need to start the firewall?

firewall
config rule
        option name 'gotovpn_ipset'
        option dest '*'
        option target 'MARK'
        option set_mark '1'
        option src 'lan'
        option ipset 'gfwlist dest'
        list proto 'all'

lansky123, 2022-09-15:
Can the "gfwlist ipset" be maked by ipset-dns? ipset is string in firewall.

lansky123, 2022-09-15:
config	ipset
	option	name		'gfwlist_1'
	option	match		'dest_ip'
	option	storage		'hash'
	option	enabled		'1'
	list	entry		'42.56.0.0/16'

config rule
	option name 'vpn_ipset'
	option dest '*'
	option target 'MARK'
	option set_mark '1'
	option src 'lan'
	option ipset 'gfwlist_1 dest'
	list proto 'all'
issue 1:
root@OpenWrt:/etc/config# /etc/init.d/firewall restart
Section @ipset[0] (gfwlist_1) option 'storage' is not supported by fw4

Issu 2:
config	ipset
	option	name		'gfwlist_1'
	option	match		'dest_ip'
	#option	storage		'hash'
	option	enabled		'1'
	list	entry		'42.56.0.0/16'

config rule
	option name 'vpn_ipset'
	option dest '*'
	option target 'MARK'
	option set_mark '1'
	option src 'lan'
	option ipset 'gfwlist_1 dest'
	list proto 'all'

/etc/init.d/firewall restart  OK
ipset list .There is not gfwlist_1

lansky123, 2022-09-15:
Issu 2:
config ipset
   option name 'gfwlist_1'
   option match 'dest_ip'
  #option storage 'hash'
  option enabled '1'
  list entry '142.250.4.106'

config rule
   option name 'vpn_ipset'
   option dest '*'
   option target 'MARK'
   option set_mark '1'
   option src 'lan'
   option ipset 'gfwlist_1 dest'
   list proto 'all'

/etc/init.d/firewall restart OK
ipset list .There is not gfwlist_1

dave14305, 2022-09-15:
`nft list sets` would be the new nft syntax to list sets.

closed 2022-09-16 by lansky123