openwrt/openwrt #10701: [22.03] Unable to build for MVEbu (Linksys WM3200ACM): hunk #1 FAILED

emandtf, 2022-09-13:
Few months ago I successfully compiled against 21.02 version but there are weeks that I try to build 22.03 without success.

I'm following build guide at: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
When I exec the command "_make -j2 V=s kernel_menuconfig_" I got this error:

`
make[1]: Entering directory '/home/emandt/src/openwrt/scripts/config'
make[1]: 'conf' is up to date.
make[1]: Leaving directory '/home/emandt/src/openwrt/scripts/config'
make[2]: Entering directory '/home/emandt/src/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/emandt/src/openwrt/scripts/config'
make[1]: Entering directory '/home/emandt/src/openwrt'
+ mkdir -p /home/emandt/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi
+ cd /home/emandt/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/emandt/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp
touch /home/emandt/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/.prepared
make[2]: Entering directory '/home/emandt/src/openwrt/toolchain/binutils'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/emandt/src/openwrt/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[2]: Entering directory '/home/emandt/src/openwrt/toolchain/gdb'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/emandt/src/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[2]: Entering directory '/home/emandt/src/openwrt/toolchain/fortify-headers'
make[2]: Entering directory '/home/emandt/src/openwrt/toolchain/gcc/initial'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/emandt/src/openwrt/toolchain/fortify-headers'
time: toolchain/fortify-headers/compile#0.00#0.01#0.02
/home/emandt/src/openwrt/staging_dir/host/bin/sed -i -e 's,TARGET_CROSS=.*,TARGET_CROSS=arm-openwrt-linux-muslgnueabi-,' /home/emandt/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/info.mk
/home/emandt/src/openwrt/staging_dir/host/bin/sed -i -e 's,GCC_VERSION=.*,GCC_VERSION=11.2.0,' /home/emandt/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/info.mk
. /home/emandt/src/openwrt/include/shell.sh; xzcat /home/emandt/src/openwrt/dl/gcc-11.2.0.tar.xz | tar -C /home/emandt/src/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gcc-11.2.0-initial/.. -xf - --exclude-from='/home/emandt/src/openwrt/toolchain/gcc/initial/../exclude-testsuite' --exclude=gcc/ada/*.ad* --exclude=libjava
[ ! -d ./src/ ] || cp -fpR ./src/* /home/emandt/src/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gcc-11.2.0-initial

Applying ../patches/11.2.0/001-v11.3.0-ranger-Fix-up-fold_using_range-range_of_address-PR10.patch using plaintext: 
patching file gcc/gimple-range.cc
The next patch would create the file gcc/testsuite/gcc.c-torture/execute/pr103255.c,
which already exists!  Applying it anyway.
patching file gcc/testsuite/gcc.c-torture/execute/pr103255.c
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file gcc/testsuite/gcc.c-torture/execute/pr103255.c.rej
Patch failed!  Please fix ../patches/11.2.0/001-v11.3.0-ranger-Fix-up-fold_using_range-range_of_address-PR10.patch!
make[2]: *** [Makefile:44: /home/emandt/src/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gcc-11.2.0-initial/.prepared] Error 1
make[2]: Leaving directory '/home/emandt/src/openwrt/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#7.52#3.11#7.74
    ERROR: toolchain/gcc/initial failed to build.
make[1]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[1]: Leaving directory '/home/emandt/src/openwrt'
make: *** [/home/emandt/src/openwrt/include/toplevel.mk:230: toolchain/install] Error 2`

emandtf, 2022-10-08:
Changing GCC version (from 11.2 to 12.1) doesn't change anything......I don't understand how all of you can build it. I just pulled from git source and executed the build command without change anything....

damnms, 2022-10-14:
same error here on fedora 37 with 22.03.1.
iirc i was able to build 22.03 but not 22.03.1

karlp, 2022-10-25:
I had this as well, updating from 22.03 "ealier" to 22.03.2  f3ffb04a43..00d7702796 to be precise.  A make dirclean fixed it.