openwrt/openwrt #10707: mt7620: disconnected due to excessive missing ACKs

HiGarfield, 2022-09-14:
On mt7620, when there is a lot of co-channel interference, "disconnected due to excessive missing ACKs" appears, and then the wireless cannot be connected (but the SSID can be seen). The wifi will remain unconnectable until the wifi is restarted.

`Wed Sep 14 15:18:17 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:18 2022 daemon.info hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: disconnected due to excessive missing ACKs
Wed Sep 14 15:18:18 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:19 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:20 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:24 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:25 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:26 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:27 2022 daemon.notice hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response
Wed Sep 14 15:18:48 2022 daemon.info hostapd: wlan0: STA XX:XX:XX:XX:XX:XX IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`

dangowrt, 2022-09-14:
@sgruszka any idea?

sgruszka, 2022-09-15:
No, unfortunately not .

HiGarfield, 2022-09-16:
The same problem was reported by 
https://forum.openwrt.org/t/ieee-802-11-disconnected-due-to-excessive-missing-acks/70025

DragonBluep, 2023-10-05:
@HiGarfield Can you help to test if enabling watchdog can have some help?
```
--- a/drivers/net/wireless/ralink/rt2x00/rt2800lib.c
+++ b/drivers/net/wireless/ralink/rt2x00/rt2800lib.c
@@ -31,7 +31,11 @@
 #include "rt2800lib.h"
 #include "rt2800.h"
 
+#ifdef CONFIG_SOC_MT7620
+static bool modparam_watchdog = true;
+#else
 static bool modparam_watchdog;
+#endif
 module_param_named(watchdog, modparam_watchdog, bool, S_IRUGO);
 MODULE_PARM_DESC(watchdog, "Enable watchdog to detect tx/rx hangs and reset hardware if detected");
 
```

HiGarfield, 2023-10-06:
> @HiGarfield Can you help to test if enabling watchdog can have some help?
>
> ```
> --- a/drivers/net/wireless/ralink/rt2x00/rt2800lib.c
> +++ b/drivers/net/wireless/ralink/rt2x00/rt2800lib.c
> @@ -31,7 [...]

I have tried it earlier. The issue is not resolved.

See:
https://github.com/HiGarfield/lede-17.01.4-Mod/blob/master/package/kernel/mac80211/patches/rt2x00/9999-007-rt2800-enable-watchdog-by-default.patch

DragonBluep, 2023-10-06:
Now I am using a simple script `check_radio.sh` to restart the wifi. 
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh

err_msg='did not acknowledge authentication response'
if [ "$(logread -e $err_msg | wc -l)" -gt 1 ]; then
	phy=`expr $(ls -l /sys/class/ieee80211 | wc -l) - 1`
	[ $phy -gt 1 ] && phy=1
	wifi down radio$phy && \
	sleep 2 && \
	wifi up radio$phy && \
	sleep 20 && \
	/etc/init.d/log restart
fi
```
This issue may be caused by DMA busy. I am still debugging it. However, I can't reproduce it quickly.

HiGarfield, 2023-10-06:
> Now I am using a simple script `check_radio.sh` to restart the wifi.
>
> ```
> #!/bin/sh
>
> # A simple script to workaround MT7620 wireless hang issue via
> # crontab schedule:
> # * * * * * [...]

By placing dozens of high-power wireless access points on the same channel around the wireless router and connecting dozens of high-load clients to the wireless router, the problem can be easily reproduced.

HiGarfield, 2023-10-10:
> @HiGarfield Could you please test this patch? [999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch](https://github.com/openwrt/openwrt/files/12848259/999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch)

In my code, there is no `CONFIG_NET_RALINK_MT7620`. I modified patch. 

See:
https://github.com/HiGarfield/lede-17.01.4-Mod/blob/master/package/kernel/mac80211/patches/rt2x00/9999-007-rt2800-enable-watchdog-by-default.patch
and
https://github.com/HiGarfield/lede-17.01.4-Mod/blob/master/package/kernel/mac80211/patches/rt2x00/9999-016-wifi-rt2x00-various-experimental-improvements-for-MT.patch

I have built the firmware and installed it to a mt7620 wireless router. I will try for several days to see the result.

DragonBluep, 2023-10-10:
Please try this new version. I found why watchdog didn't work.
[999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch](https://github.com/openwrt/openwrt/files/12857654/999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch)

DragonBluep, 2023-10-10:
> Please try this new version. I found why watchdog didn't work. [999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch](https://github.com/openwrt/openwrt/files/12857654/999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch)

This patch causes too much watchdog reset. I changed the check times in this one.
[999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch](https://github.com/openwrt/openwrt/files/12858331/999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch)

DragonBluep, 2023-10-11:
@HiGarfield It seems that the check threshold is still small. Though iperf3 pressure test is fine, I can still catch reset event (only one time) when I started some P2P download schedules. If kernel throw a lot of restart log you can increase the threshold to suppress them, unit is 100ms.
`return queue->wd_count > 32;`

I‘m testing `64` now, but this number may be too aggressive. 40~50 should be enough.

DragonBluep, 2023-10-11:
I just found it restart again. This time seems to recover the real hang state.
```
Wed Oct 11 19:58:04 2023 daemon.notice hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Wed Oct 11 19:58:06 2023 daemon.notice hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Wed Oct 11 19:58:06 2023 daemon.notice hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Wed Oct 11 19:58:09 2023 kern.warn kernel: [38734.029366] ieee80211 phy1: rt2800_watchdog: Warning - Watchdog TX hung detected
Wed Oct 11 19:58:09 2023 kern.info kernel: [38734.036958] ieee80211 phy1: Hardware restart was requested
Wed Oct 11 19:58:09 2023 kern.warn kernel: [38734.389369] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
Wed Oct 11 19:58:09 2023 kern.warn kernel: [38734.739351] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 1 failed to flush
Wed Oct 11 19:58:10 2023 kern.warn kernel: [38735.089346] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush
Wed Oct 11 19:58:10 2023 kern.warn kernel: [38735.439354] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 3 failed to flush
```

HiGarfield, 2023-10-12:
With the patch above applied, the problem still exists without any reset.

```
Thu Oct 12 07:09:01 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Thu Oct 12 07:09:57 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Thu Oct 12 07:10:53 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Thu Oct 12 07:10:53 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Thu Oct 12 07:11:03 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
Thu Oct 12 07:11:10 2023 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
```

DragonBluep, 2023-10-12:
> With the patch above applied, the problem still exists. No reset occurred.
> 
> ```
> Thu Oct 12 07:09:01 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
> Thu Oct 12 07:09:57 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
> Thu Oct 12 07:10:53 2023 daemon.notice hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
> Thu Oct 12 07:10:53 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
> Thu Oct 12 07:11:03 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
> Thu Oct 12 07:11:10 2023 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
> ```

It seems that the watchdog check threshold is too large to catch the hang state. MT7620 may need some additional checks.
[999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch](https://github.com/openwrt/openwrt/files/12882462/999-wifi-rt2x00-various-experimental-improvements-for-MT7620.patch)

HiGarfield, 2023-10-13:
> It seems that the watchdog check threshold is too large to catch the hang state. MT7620 may need some additional checks. [...]

After applying the patch, I find the following log.

```
Fri Oct 13 21:13:22 2023 daemon.info hostapd: wlan0: STA b6:64:d0:e4:b2:59 IEEE 802.11: disconnected due to excessive missing ACKs
Fri Oct 13 21:13:22 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:64:d0:e4:b2:59
Fri Oct 13 21:13:27 2023 daemon.notice hostapd: Beacon request: b6:64:d0:e4:b2:59 is not connected
```

DragonBluep, 2023-10-28:
@HiGarfield Hi! I pushed some new patches to https://github.com/openwrt/openwrt/pull/13786. This time I get better result. At least it works for me. 

I noticed you are using very old mac80211 backport. So please make sure you have backported all necessary upstream patches otherwise it won't work at all.  I have verified that they cannot work on OpenWrt 19.07. Some patches are missing...

> After applying the patch, I find the following log.
>
> ```
> Fri Oct 13 21:13:22 2023 daemon.info hostapd: wlan0: STA b6:64:d0:e4:b2:59 IEEE 802.11: disconnected due to excessive missing ACKs
> Fri Oct 13 [...]

I didn't catch this `Beacon request:` log on OpenWrt main branch. It should have been fixed upstream.

HiGarfield, 2023-10-30:
> @HiGarfield Hi! I pushed some new patches to #13786. This time I get better result. At least it works for me.
>
> I noticed you are using very old mac80211 backport. So please make sure you have [...]

Thanks for your efforts. I have backported upstream patches related to rt2x00. Now I have applied your new patches. I will test for several days to see the result.

HiGarfield, 2023-11-05:
> @HiGarfield Hi! I pushed some new patches to #13786. This time I get better result. At least it works for me.
>
> I noticed you are using very old mac80211 backport. So please make sure you have [...]

After applying patches in https://github.com/openwrt/openwrt/pull/13786, I got
```
Sat Nov  4 21:22:16 2023 kern.warn kernel: [147132.933397] ieee80211 phy0: rt2800_watchdog_hung: Warning - Watchdog TX hung detected
Sat Nov  4 21:22:16 2023 kern.info kernel: [147132.949219] ieee80211 phy0: Hardware restart was requested
Sat Nov  4 21:22:16 2023 kern.warn kernel: [147133.059246] ieee80211 phy0: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush
Sat Nov  4 21:22:21 2023 kern.warn kernel: [147136.281046] ieee80211 phy0: rt2800_rxiq_calibration: Warning - RXIQ calibration FAIL
Sat Nov  4 21:22:23 2023 kern.warn kernel: [147136.959641] ieee80211 phy0: rt2800_watchdog_hung: Warning - Watchdog TX hung detected
Sat Nov  4 21:22:23 2023 kern.info kernel: [147136.975474] ieee80211 phy0: Hardware restart was requested
Sat Nov  4 21:22:23 2023 kern.warn kernel: [147137.086517] ieee80211 phy0: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
Sat Nov  4 21:22:24 2023 kern.warn kernel: [147137.202228] ieee80211 phy0: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush
Sat Nov  4 21:22:28 2023 kern.warn kernel: [147140.423461] ieee80211 phy0: rt2800_rxiq_calibration: Warning - RXIQ calibration FAIL
```

DragonBluep, 2023-11-06:
> After applying patches in #13786, I got
>
> ```
> Sat Nov  4 21:22:16 2023 kern.warn kernel: [147132.933397] ieee80211 phy0: rt2800_watchdog_hung: Warning - Watchdog TX hung detected
> Sat Nov  4 21:22:16 [...]

It is expected. Due to some queues getting stuck (this is the cause of the issue), flushing queue will fail in most cases. Has your device returned to normal state after these logs?

HiGarfield, 2023-11-06:
> It is expected. Due to some queues getting stuck (this is the cause of the issue), flushing queue will fail in most cases. Has your device returned to normal state after these logs?

The wireless network can be connected at this moment. I will go on my testing.

Alternativend, 2024-02-22:
Are there any news on this? I have a few dozen Asus RT-AX53U which are basically broken now, because they wont even run for a single day. After a few hours they keep "crashing" like that.

DragonBluep, 2024-02-22:
> Are there any news on this? I have a few dozen Asus RT-AX53U which are basically broken now, because they wont even run for a single day. After a few hours they keep "crashing" like that.

@Alternativend Sorry, but this issue only focuses on MT7620.

@HiGarfield BTW, has the latest patches solved this problem for you? If so, maybe we can close this issue.

HiGarfield, 2024-02-23:
> @Alternativend Sorry, but this issue only focuses on MT7620.
>
> @HiGarfield BTW, has the latest patches solved this problem for you? If so, maybe we can close this issue.

Thank you for providing the patch to address the issue. The previous issues have been successfully resolved. However, there are still some persistent problems affecting the connection quality, including slow speed and significant packet loss. Furthermore, a new issue has arisen, which involves the frequent triggering of the watchdog.

closed 2024-02-23 by HiGarfield

DragonBluep, 2024-02-23:
> The previous issues have been successfully resolved. However, there are still some persistent problems affecting the connection quality, including slow speed and significant packet loss.

Reducing the tx power should be helpful.

> Furthermore, a new issue has arisen, which involves the frequent triggering of the watchdog.

I noticed the watchdog patches on your repo are outdated. I disabled the useless hang check watchdog on the final version. Maybe you can try sync with upstream version again.

Alternativend, 2024-02-23:
So no chances for MT7621 to solve this?

I will try to reduce tx power as well next.

DragonBluep, 2024-02-23:
> So no chances for MT7621 to solve this?
> 
> I will try to reduce tx power as well next.

@Alternativend Please report it to https://github.com/openwrt/mt76 or Linux wireless mailing list.

HiGarfield, 2024-02-23:
> Reducing the tx power should be helpful.
>
>
> I noticed the watchdog patches on your repo are outdated. I disabled the useless hang check watchdog on the final version. Maybe you can try sync with [...]

OK. Thanks. I will try it later.

HiGarfield, 2024-02-24:
> Reducing the tx power should be helpful.
>
>
> I noticed the watchdog patches on your repo are outdated. I disabled the useless hang check watchdog on the final version. Maybe you can try sync with [...]

Sorry, I have tried to apply new patches, but failed due to huge difference between mac80211 4.xx and 6.xx。

namiltd, 2024-08-12:
Unfortunately the watchdog problem still occurs for mt7620 (tested on Archer C5 v4 on current snapshot).
```
[89618.690112] ieee80211 phy1: rt2800_watchdog_dma_busy: Warning - Watchdog TX DMA busy detected
[89618.699505] ieee80211 phy1: Hardware restart was requested
[89618.886519] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
```
I will probably have to use the reset script described above.

DragonBluep, 2024-08-12:
> Unfortunately the watchdog problem still occurs for mt7620 (tested on Archer C5 v4 on current snapshot).
> 
> ```
> [89618.690112] ieee80211 phy1: rt2800_watchdog_dma_busy: Warning - Watchdog TX DMA busy detected
> [89618.699505] ieee80211 phy1: Hardware restart was requested
> [89618.886519] ieee80211 phy1: rt2x00queue_flush_queue: Warning - Queue 0 failed to flush
> ```
> 
> I will probably have to use the reset script described above (check_radio.sh) or something similar I wonder why iieee80211_restart_hw() does not work and does not restart itself.

These logs are just some warning. The ieee80211_restart_hw() should have been executed correctly.

namiltd, 2024-08-12:
Unfortunately the wifi remains off.

namiltd, 2024-08-21:
A script that helps:
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh

err_msg='phy1: Hardware restart was requested'
if [ "$(logread -e $err_msg | wc -l)" -gt 1 ]; then
	wifi down radio1 && \
	sleep 2 && \
	wifi up radio1 && \
	sleep 20 && \
	/etc/init.d/log restart
fi
```

namiltd, 2024-09-21:
Now script is more universal:
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh

err_msg='phy0: Hardware restart was requested'
(logread 2>/dev/null | grep -q "$err_msg") && phy0=true

err_msg='phy1: Hardware restart was requested'
(logread 2>/dev/null | grep -q "$err_msg") && phy1=true

if [ "$phy0" == "true" ] || [ "$phy1" == "true" ]; then
    if [ "$phy0" == "true" ]; then
        wifi down radio0
    fi
    if [ "$phy1" == "true" ]; then
        wifi down radio1
    fi

    sleep 2

    if [ "$phy0" == "true" ]; then
        wifi up radio0
    fi
    if [ "$phy1" == "true" ]; then
        wifi up radio1
    fi

    sleep 20
    /etc/init.d/log restart
fi
```

DragonBluep, 2024-09-29:
@namiltd You can try increase these number.
https://github.com/torvalds/linux/blob/3efc57369a0ce8f76bf0804f7e673982384e4ac9/drivers/net/wireless/ralink/rt2x00/rt2800lib.c#L1331-L1332

namiltd, 2024-12-29:
The same script but run every 30 seconds and log restarts:
```
#!/bin/sh

# A simple script to workaround MT7620 wireless hang issue via
# crontab schedule:
# * * * * * /sbin/check_radio.sh 
# * * * * * sleep 30 && /sbin/check_radio.sh

minilog=`logread 2>/dev/null | grep "Hardware restart was requested"`

(echo $minilog | grep -q "phy0: Hardware restart was requested") && phy0=true 

(echo $minilog | grep -q "phy1: Hardware restart was requested") && phy1=true

if [ "$phy0" == "true" ] || [ "$phy1" == "true" ]; then
#    logread >> /tmp/logread.log

    /etc/init.d/log restart

    [ "$phy0" == "true" ] && date >>/tmp/radio0_restart.log
    [ "$phy1" == "true" ] && date >>/tmp/radio1_restart.log

    [ "$phy0" == "true" ] && wifi down radio0
    [ "$phy1" == "true" ] && wifi down radio1

    sleep 2

    [ "$phy0" == "true" ] && wifi up radio0
    [ "$phy1" == "true" ] && wifi up radio1
fi
```

kofec, 2024-12-31:
Maybe running the script in the background from /etc/rc.local with a loop and a sleep