openwrt/openwrt #10735: ZyXEL NBG6817 MAC addresses changes randomly

NilsRo, 2022-09-15:
I facing the issue with OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.213.35964-87836ca on ZyXEL NBG6817 (and also 21.03.0) that the max addresses changes randomly after reboot. 
[image]
[image]

My actual network configuration file and I opened a discussion https://forum.openwrt.org/t/mac-adresses-changes-randomly/136999
Also I found a similar issue but different device and also correction, but perhaps they are linked: https://github.com/openwrt/openwrt/issues/9766#issuecomment-1248124941

I cannot see a similar behavior with my Netgear R7800 devices which have the same architecture. 

config interface 'loopback'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'
        option device 'lo'

config globals 'globals'
        option ula_prefix 'fd11:b2f0:a80f::/48'

config interface 'lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ipaddr '192.168.177.1'
        option device 'br-lan'

config interface 'wan'
        option proto 'dhcp'
        option broadcast '1'
        option delegate '0'
        option device 'eth0.2'

config interface 'wan6'
        option proto 'dhcpv6'
        option reqprefix 'auto'
        option reqaddress 'try'
        option device 'eth0.2'
        option auto '0'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option vid '1'
        option ports '2 3 4 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '0t 5'
        option vid '2'

config interface 'VPN'
        option device 'tap_vpn'
        option proto 'none'

config route
        option netmask '255.255.255.0'
        option target '192.168.178.0'
        option interface 'lan'
        option gateway '192.168.177.6'

config interface 'IoT'
        option proto 'static'
        option ipaddr '192.168.175.1'
        option netmask '255.255.255.0'
        option device 'br-iot'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1.1'
        list ports 'tap_vpn'

config switch_vlan
        option device 'switch0'
        option vlan '3'
        option ports '1 6t'
        option vid '3'

config route
        option interface 'lan'
        option target '192.168.180.0'
        option netmask '255.255.255.0'
        option gateway '192.168.177.6'

config interface 'ELLERAU'
        option device 'vpn_ellerau'
        option proto 'dhcp'

config device
        option type 'bridge'
        option name 'br-iot'
        list ports 'eth1.3'

config interface 'GUEST'
        option proto 'static'
        option ipaddr '192.168.176.1'
        option netmask '255.255.255.0'
        option device 'br-guest'

config device
        option type 'bridge'
        option name 'br-guest'
        option bridge_empty '1'

NilsRo, 2022-09-16:
Found a matching MAC here but it is the same for another router I own...

mtd_get_mac_binary 0:SBL1 0x6170
8c:59:00:2a:34:5b

So could it be right?

Ansuel, 2022-10-25:
@NilsRo can you compile a custom image and test your implementation?

NilsRo, 2022-10-25:
> @NilsRo can you compile a custom image and test your implementation? 

Actually I have no solution as with ucidef_set_interface_macaddr I cannot set a MAC to eth0 and eth1. I have to set this in the network configuration manually. So I we need a function for ethX devices I ask in the forum for without response.