openwrt/openwrt #10737: lscpu seg faults

efahl, 2022-09-16:
This is on a Belkin RT3200 running snapshot 20614. Not tested elsewhere.  strace shows it's almost there, then fails.  Same segfault with any parameter combination that I've tried (other than help and version).

```
$ cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r20614-c27279dc26"
OPENWRT_BOARD="mediatek/mt7622"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r20614-c27279dc26"

$ opkg install lscpu
Package lscpu (2.38-2) installed in root is up to date.

$ strace lscpu
execve("/usr/bin/lscpu", ["lscpu"], 0x7fe684d110 /* 15 vars */) = 0
set_tid_address(0x7f8f939248)           = 6160
brk(NULL)                               = 0x2980f000
brk(0x29811000)                         = 0x29811000
mmap(0x2980f000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x2980f000
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libsmartcols.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libsmartcols.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libsmartcols.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=98307, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0`R\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 167936, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7f8f879000
mmap(0x7f8f8a0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x17000) = 0x7f8f8a0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=73744, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0-\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 143360, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7f8f856000
mmap(0x7f8f877000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x11000) = 0x7f8f877000
close(3)                                = 0
mprotect(0x7f8f8a0000, 4096, PROT_READ) = 0
mprotect(0x7f8f877000, 4096, PROT_READ) = 0
mprotect(0x428000, 4096, PROT_READ)     = 0
ioctl(1, TIOCGWINSZ, {ws_row=60, ws_col=200, ws_xpixel=2200, ws_ypixel=1200}) = 0
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8f92e000
openat(AT_FDCWD, "/sys/devices/system/cpu", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
openat(3, "kernel_max", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8f92d000
fcntl(4, F_SETFD, FD_CLOEXEC)           = 0
read(4, "1\n", 1024)                    = 2
lseek(4, -1, SEEK_CUR)                  = 1
close(4)                                = 0
munmap(0x7f8f92d000, 4096)              = 0
openat(3, "possible", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8f92d000
fcntl(4, F_SETFD, FD_CLOEXEC)           = 0
read(4, "0-1\n", 1024)                  = 4
close(4)                                = 0
munmap(0x7f8f92d000, 4096)              = 0
openat(3, "present", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
[... 90 lines cut ...]
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
read(5, "", 1024)                       = 0
close(5)                                = 0
munmap(0x7f8f92d000, 4096)              = 0
openat(3, "cpu0/topology/book_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(3, "cpu0/topology/drawer_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
faccessat(3, "cpu1/topology/thread_siblings", F_OK) = 0
openat(3, "cpu1/topology/thread_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 5
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8f92d000
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
read(5, "", 1024)                       = 0
close(5)                                = 0
munmap(0x7f8f92d000, 4096)              = 0
openat(3, "cpu1/topology/core_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 5
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8f92d000
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
read(5, "", 1024)                       = 0
close(5)                                = 0
munmap(0x7f8f92d000, 4096)              = 0
openat(3, "cpu1/topology/book_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(3, "cpu1/topology/drawer_siblings", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x10} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

ynezz, 2022-09-16:
@efahl IMO it's not OpenWrt issue, so I would report it [upstream|https://github.com/util-linux/util-linux/issues].

neheb, 2022-09-16:
if I had to guess, printf issue :)

efahl, 2022-09-16:
Got it, thanks.  Will report upstream.  https://github.com/util-linux/util-linux/issues/1810

ynezz, 2022-09-16:
The issue is still there, if you don't mind I would rather close this once it's fixed.

neheb, 2022-09-17:
tried to reproduce on armvirt. No dice:

```
root@OpenWrt:/# lscpu
Architecture:            aarch64
  CPU op-mode(s):        32-bit, 64-bit
  Byte Order:            Little Endian
CPU(s):                  2
  On-line CPU(s) list:   0,1
Vendor ID:               ARM
  Model name:            Cortex-A53
    Model:               4
    Thread(s) per core:  1
    Core(s) per cluster: 2
    Socket(s):           -
    Cluster(s):          1
    Stepping:            r0p4
    BogoMIPS:            125.00
    Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
Vulnerabilities:
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Not affected
  Spec store bypass:     Not affected
  Spectre v1:            Mitigation; __user pointer sanitization
  Spectre v2:            Not affected
  Srbds:                 Not affected
  Tsx async abort:       Not affected
```

Maybe some extra settings are needed.

efahl, 2022-09-18:
Not much to see without a debug build/symbol tables, strace at least gives a sense of what was going on when it crashed.

```
(gdb) r
Starting program: /usr/bin/lscpu
Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7f9367c in ?? ()

(gdb) bt
#0  0x0000007ff7f9367c in ?? ()
#1  0x0000007ff7fffe70 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

neheb, 2022-09-18:
Can you build with debug symbols and try again?

ynezz, 2022-09-22:
Just tracked it down and proposed [upstream fix](https://github.com/util-linux/util-linux/pull/1821).

ynezz, 2022-09-22:
> Maybe some extra settings are needed.

You need 5.15 kernel to reproduce this.

efahl, 2022-09-30:
Kernel fix thread https://lore.kernel.org/lkml/20220922113217.GA90426@meh.true.cz/T/#u

Upstream utility fix in https://github.com/util-linux/util-linux/commit/d2081ec7a1ba302aa3f4414c007e782e08a32f6e

efahl, 2022-10-03:
Tested in SNAPSHOT, r20856-2b4f12e55b, kernel 5.15.71 with lscpu 2.38 and this bug has been fixed.

closed 2022-10-03 by efahl