openwrt/openwrt #10738: busybox ntpd is not Y2036-ready

mlichvar, 2022-09-16:
As a new feature in openwrt-22.03, all archs were switched to 64-bit time_t in order to support timestamps after 2038. I'm not sure if the expectation is that some devices on 22.03 will actually be used for so long, but there is an issue with the default NTP client that it doesn't handle overflow in the NTP timestamp which will happen in 2036. That's the beginning of the next NTP era.

There is a trivial patch for busybox to support synchronization until 2106 (not accepted upstream yet):
https://www.mail-archive.com/busybox@busybox.net/msg28352.html

ynezz, 2022-09-16:
@mlichvar thanks! BTW whats the status of testing/deployment of this fix within RH?

mlichvar, 2022-09-16:
There is no deployment of the fix that I know of. I think it's safe, but of course I could be wrong. This was just an effort to minimize the global NTP breakage and possibly flooding of public servers that will happen in 2036.

When I saw that 22.03 was advertised as Y2038-ready, I thought you would at least want to be aware of this NTP-specific Y2036 issue.

ynezz, 2022-09-16:
@mlichvar this is above my head, so I've asked @arndb for help and he kindly looked at it and we've got `Reviewed-by: Arnd Bergmann <arnd@arndb.de>`, so I'm going to prepare a fix with that unless someone else beats me to it. IMO this might probably help with your upstream fix as well.

arndb, 2022-09-16:
It may be worth pointing out here that the upstream openntpd project from which the busybox version was originally forked uses a slightly different change, see 
https://github.com/openntpd-portable/openntpd-openbsd/commit/0e2d622180

The only practical difference is that openntpd treats times between 1968 and 1970 as valid, while the proposed patch here wraps them to the 2104 to 2016 range. Linux does not support setting the time anything before 1970, so @mlichvar's approach is probably better here.

mlichvar, 2022-09-16:
Hm, looking at the openntpd change, I think we might need to update `d_to_lfp()` and `d_to_tv()` too. IIRC I tested it as a server, but maybe it was relying on undefined behavior in casting of double to uint32.

mlichvar, 2022-09-16:
I did some tests and confirmed that the clock stepping was broken after 2038 and the server timestamp conversion was inaccurate (in the sub-second fraction) after 2036.

Here is the new patch:
https://www.mail-archive.com/busybox@busybox.net/msg28497.html

I tested both client and server before 2036, between 2036 and 2038, and after 2038 on Fedora/glibc with binaries compiled with `-m32 -D_FILE_OFFSET_BITS=64 -D_TIME_BITS=64`.