openwrt/openwrt #10744: [22.03] kernel RTC_HCTOSYS broken on 32-bit platforms

ynezz, 2022-09-16:
Quoting from [linux-arm-kernel](https://lore.kernel.org/linux-arm-kernel/CAKYb531CyL8XRVRcRN30cC3xRgsd-1FzXUeS7o2LiZqALJ42qw@mail.gmail.com/):

> I'm working to fix the y2038 issue for an existing sama5d3-based product. This involves updating the kernel and glibc to appropriate versions (5.10 and 2.35.1 respectively) and I got things running up to a state where, from userspace, both date and hwclock commands have no issue accepting dates beyond 2038. However, even with the RTC_HCTOSYS and RTC_HCTOSYS_DEVICE options configured correctly, the RTC driver fails to initialize the system clock at bootup.

Proposed [[PATCH] [RFC] rtc: y2038: remove broken RTC_HCTOSYS workaround](https://lore.kernel.org/lkml/20220908115337.1604277-1-arnd@kernel.org).

neheb, 2022-09-17:
```
> git grep RTC_HCTOSYS
target/linux/generic/config-5.10:CONFIG_RTC_HCTOSYS=y
target/linux/generic/config-5.10:CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
target/linux/generic/config-5.15:CONFIG_RTC_HCTOSYS=y
target/linux/generic/config-5.15:CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
```

hmm.

neheb, 2022-09-17:
Based on that patch, this is an issue with CONFIG_COMPAT_32BIT_TIME

```
> git grep CONFIG_COMPAT_32BIT_TIME | wc -l
77
```

What's confusing here is that openwrt uses musl 1.2.x, which uses 64-bit time_t. I'm curious what the connection is.