openwrt/openwrt #10749: fw4: race condition with dnat rules (config redirect) during boot, leading to such rules being persistently broken

zorun, 2022-09-17:
I have an issue with dnat rules during boot on 22.03.0 (with fw4): if a packet matching a dnat rule hits the router just after it booted, the dnat rule will never correctly work from this remote host. That is, the port redirect will be persistently broken, no traffic from the remote host will ever go through, it will always be reject with an ICMP error.

When I'm lucky after a router reboot, this bug doesn't trigger. It's a race condition that only triggers when remote traffic hits the dnat rule of the router while fw4 is still initializing (which means a window of less than 10 seconds, see detailed analysis below)

Assuming I have the following redirect rule:

```
config redirect
        option name             wireguard-to-lan
        option src              wan
        option src_dport        50000:50099
        option proto            udp
        option dest             lan
        option dest_ip          192.168.1.142
```

Remote host `203.0.113.8` is sending regular traffic that matches this rule (it's a wireguard tunnel between the remote host and the LAN device `192.168.1.142`)

When the bug triggers after a reboot, this is what I see from the remote host, which is unexpected:

```
12:23:08.592193 IP 203.0.113.8.38817 > $WAN_IP.50014: UDP, length 4
12:23:08.646537 IP $WAN_IP > 203.0.113.8: ICMP $WAN_IP udp port 50014 unreachable, length 40
```

This is the conntrack state on the router:

```
# conntrack -L
udp      17 59 src=203.0.113.8 dst=$WAN_IP sport=38817 dport=50014 packets=10 bytes=320 [UNREPLIED] src=$WAN_IP dst=203.0.113.8 sport=50014 dport=38817 packets=0 bytes=0 mark=0 use=1
```

This is also unexpected, because `$WAN_IP` appears on both side of the conntrack state. Clearly, no dnat rule was used to build this conntrack entry, otherwise we would get the LAN IP address on the right side. Since this wrong conntrack entry stays around and is refreshed every time a new packet matches, it explains why connectivity stays broken forever.

zorun, 2022-09-17:
I believe this is because fw4 builds its ruleset gradually: it starts mostly empty, and then fills up afterwards when interfaces come up (probably via hotplug triggers or ubus notifications, I'm not familiar enough with the design).

Just after boot, this is the state of nftables:

```
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy accept;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
	}

	chain prerouting {
		type filter hook prerouting priority filter; policy accept;
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		ct status dnat accept comment "!fw4: Accept port redirections"
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		ct status dnat accept comment "!fw4: Accept port forwards"
		jump accept_to_lan
	}

	chain helper_lan {
	}

	chain accept_from_lan {
	}

	chain accept_to_lan {
	}

	chain input_wan {
		meta nfproto ipv4 udp dport 68 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCP-Renew"
		icmp type echo-request counter packets 0 bytes 0 accept comment "!fw4: Allow-Ping"
		meta nfproto ipv4 meta l4proto igmp counter packets 0 bytes 0 accept comment "!fw4: Allow-IGMP"
		meta nfproto ipv6 udp dport 546 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCPv6"
		ip6 saddr fe80::/10 icmpv6 type . icmpv6 code { mld-listener-query . no-route, mld-listener-report . no-route, mld-listener-done . no-route, mld2-listener-report . no-route } counter packets 0 bytes 0 accept comment "!fw4: Allow-MLD"
		icmpv6 type { destination-unreachable, time-exceeded, echo-request, echo-reply, nd-router-solicit, nd-router-advert } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Input"
		icmpv6 type . icmpv6 code { packet-too-big . no-route, parameter-problem . no-route, nd-neighbor-solicit . no-route, nd-neighbor-advert . no-route, parameter-problem . admin-prohibited } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Input"
		ct status dnat accept comment "!fw4: Accept port redirections"
		jump reject_from_wan
	}

	chain output_wan {
		jump accept_to_wan
	}

	chain forward_wan {
		icmpv6 type { destination-unreachable, time-exceeded, echo-request, echo-reply } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Forward"
		icmpv6 type . icmpv6 code { packet-too-big . no-route, parameter-problem . no-route, parameter-problem . admin-prohibited } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Forward"
		meta l4proto esp counter packets 0 bytes 0 jump accept_to_lan comment "!fw4: Allow-IPSec-ESP"
		udp dport 500 counter packets 0 bytes 0 jump accept_to_lan comment "!fw4: Allow-ISAKMP"
		ct status dnat accept comment "!fw4: Accept port forwards"
		jump reject_to_wan
	}

	chain accept_to_wan {
	}

	chain reject_from_wan {
	}

	chain reject_to_wan {
	}

	chain dstnat {
		type nat hook prerouting priority dstnat; policy accept;
	}

	chain srcnat {
		type nat hook postrouting priority srcnat; policy accept;
	}

	chain dstnat_lan {
	}

	chain srcnat_lan {
	}

	chain dstnat_wan {
		meta nfproto ipv4 udp dport 50000-50099 counter packets 0 bytes 0 dnat ip to 172.23.184.119:50000-50099 comment "!fw4: wireguard-to-lan"
	}

	chain srcnat_wan {
		meta nfproto ipv4 masquerade comment "!fw4: Masquerade IPv4 wan traffic"
	}

	chain raw_prerouting {
		type filter hook prerouting priority raw; policy accept;
	}

	chain raw_output {
		type filter hook output priority raw; policy accept;
	}

	chain mangle_prerouting {
		type filter hook prerouting priority mangle; policy accept;
	}

	chain mangle_postrouting {
		type filter hook postrouting priority mangle; policy accept;
	}

	chain mangle_input {
		type filter hook input priority mangle; policy accept;
	}

	chain mangle_output {
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
	}
}
```

Some interesting things to note:

- the `input` and `filter` chains are mostly empty
- the dnat rule is here, but nothing points to it yet (we are missing the `chain dstnat` with `type nat hook prerouting` that dispatches to dnat rules depending on the input interface)
- so, if a packet comes from the WAN interface at this point (e.g. the wireguard case I described above), it will not match any nft rules. It will thus hit the local network stack of the router, and since UDP port 50014 is not bound locally by any process, the inbound packet will be rejected with ICMP unreacheable, and the wrong conntrack entry will be created.

If this is correct, then this explains the behaviour I'm seeing.

zorun, 2022-09-17:
Now the second question is: why is the race condition window so large? When the WAN interface goes up, fw4 should almost immediately add the missing rules and all should go well.

From what I can tell, it's a combination of several factors:

- fw4 takes time to generate the nft ruleset (around 2 seconds on my device for a simple `fw4 print`)
- fw4 handles "notifications" about interfaces going up serially, which means it will regenerate the whole ruleset several times in a row, each time with new rules for the interface it was notified about.

In my setup, I have two WAN interfaces, which means that it takes at least 6s before the rules matching the second WAN interface are created (2s for LAN, 2s for WAN1, 2s for WAN2). In the meantime, WAN2 has already gone up and started receiving packets, triggering the "ICMP unreachable / wrong conntrack entry" problem. My dual-WAN setup probably makes the problem worse (larger race condition window), but I believe even a simple LAN+WAN setup has the same problem.

jow-, 2022-09-17:
Please try setting the global input policy (within the defaults section) to `drop` and see if this solves your issue.

perrin4869, 2022-09-27:
I am not sure if this is the same issue I'm facing, but also I am getting rejected packets to a certain IP address behind a wireguard VPN. Seems like it could be the same issue as this. How can I confirm this?

```
OpenWrt 22.03.0, r19685-512e76967f
```

zorun, 2022-09-28:
@jow- it does fix the issue, thanks! (with `REJECT` in the UCI config, it's translated to nft `drop`)

Should it become the default?

jow-, 2022-09-29:
Yes, I think we should make it the default.

commit ef597b02 (2022-11-01): firewall: config: drop input traffic by default

This is necessary with firewall4 to avoid a hard-to-diagnose race
condition during boot, causing DNAT rules not to be taken into account
correctly.

The root cause is that, during boot, the ruleset is mostly empty, and
interface-related rules (including DNAT rules) are added incrementally.
If a packet hits the input chain before the DNAT rules are setup, it can
create buggy conntrack entries that will persist indefinitely.

This new default should be safe because firewall4 explicitly accepts
authorized traffic and rejects the rest.  Thus, in nor [... cut]
closed 2022-11-01 by openwrt-bot