openwrt/openwrt #10752: The heap memory performance (malloc/free) is significantly degraded in musl 1.2 (compared to 1.1)

asbai, 2022-09-18:
We have two test nodes as follow:
**Node 1**: OpenWrt 19.07.3 - mips32r2 (MT7621) running musl (mipsel-sf) **1.1.24**.
**Node 2**: OpenWrt 22.03.0-rc4 - armv7-a (IPQ4019) running musl (armhf) **1.2.3**.

We ran the following simple tests on the above two nodes:
```C
for (size_t i=0; i<10485760; i+=128)
{
	void* p = malloc(i);
	malloc_usable_size(p);
	free(p);
}
```

It took 6.73s on node 1 and 10.65s on node 2. Obviously, node 2 is much slower than node 1.

PLEASE NOTE: This is measured against the fact that node 2 (P&W R619AC w/ IPQ4019) is much more powerful hardware than node 1 (Newifi-D2 w/ MT7621). So the malloc performance degradation of musl 1.2 is obviously larger than the above values.

neheb, 2022-09-18:
what's the significance of this?

asbai, 2022-09-18:
> what's the significance of this?

malloc in musl 1.2 is saveral times slower than the old one. Is this significance?

neheb, 2022-09-18:
why would it be?

asbai, 2022-09-18:
> why would it be?

According to their [changelog](https://musl.libc.org/releases.html), they replaced the old dlmalloc to a new implementation (mallocng) in version 1.2.1. I believe this caused the problem.

I did a quick code review and found that the old dlmalloc still seems to exist in musl 1.2.3 (malloc/oldmalloc folder, new one in /malloc/mallocng folder). I guess we can switch back to the old dlmalloc until this is fixed?

neheb, 2022-09-18:
Why is the speed of malloc relevant?

asbai, 2022-09-18:
> Why is the speed of malloc relevant?

malloc is the infrastructure of almost all c/c++ programs and often have wide-ranging effects on their performance. I suggest you just google this question for more info.

neheb, 2022-09-18:
Here's Mr. Musl's response:

```
[07:21] <dalias> i really dislike misuse of malloc benchmarks that are just hammering malloc because they amplify the % difference by having no load
[07:22] <dalias> it can be meaningful to do that if you're going to apply the results as part of a real analysis involving % of application time spent in malloc
[07:22] <dalias> but nobody does that
[07:22] <dalias> it's just "zomg hyperwastefullolgetpwnedmalloc is 1000% faster!!!"
```

There's no issue here.

asbai, 2022-09-18:
> Here's Mr. Musl's response:
>
> ```
> [07:21] <dalias> i really dislike misuse of malloc benchmarks that are just hammering malloc because they amplify the % difference by having no load
> [07:22] <dalias> [...]

I don't agree with this statement. One of the most obvious proofs: tcmalloc (developed by google), jemalloc and other well-recognized allocators do have obvious performance improvements in a large number of tests including this test. 

This shows that a good allocator can achieve an excellent balance of fragmentation control, space efficiency and allocation performance, rather than a lame state like this.

neheb, 2022-09-18:
Then use that. musl will not be changed to accommodate your needs.

One potential benefit that was noticed in the upgrade to 1.2 was lower memory footprint, probably as a result of the new allocator. That's a benefit.

nbd168, 2022-09-18:
> We have two test nodes as follow: **Node 1**: OpenWrt 19.07.3 - mips32r2 (MT7621) running musl (mipsel-sf) **1.1.24**. **Node 2**: OpenWrt 22.03.0-rc4 - armv7-a (IPQ4019) running musl (armhf) [...]

Please correct me if I'm misreading what you wrote here, but my interpretation is this:
You're comparing two setups that have different architectures, different SoC and different musl versions (and likely also different gcc versions) and you're attributing the observed performance difference entirely to musl malloc changes?
That makes no sense to me at all. 
Let's also look at your claim (presented as fact) that node 2 is "much more powerful hardware" than node 1:
MT7621 has 2 cores (4 virtual threads) running at 800 MHz, while IPQ4019 has 4 cores running at 717 MHz.
IPQ4019 might have more processing power because it has more cores, but your test is single-threaded and the IPQ4019 runs at a lower clock speed than MT7621.

I you insist that there is a malloc performance regression, please at least run the old/new comparison on the same hardware...

asbai, 2022-09-18:
> MT7621 has 2 cores (4 virtual threads) running at 800 MHz, while IPQ4019 has 4 cores running at 717 MHz.
> IPQ4019 might have more processing power because it has more cores, but your test is single-threaded and the IPQ4019 runs at a lower clock speed than MT7621.

Yes, we fully consider that this is a simple single-core test, but we not only consider the frequency, but also compare the difference between the ARMv7-a instruction set and the mips32r2. Therefore, we believe that the IPQ4019 has a considerable improvement over the MT7621 even in single-core performance.

If more consistent testing can effectively promote the improvement of musl's allocator, then we are willing to rebuild an environment. But at the moment it seems that is not the case.

tapper82, 2022-09-18:
Run the same test on one of the routers you have with the old and the new builds of OpenWrt. It's not that hard. To flash a old build then run the test then flash a new build then run the test again should take about 20 minutes.

champtar, 2022-09-18:
Or recompile musl with `--with-malloc=oldmalloc`, because changing the kernel, gcc, architecture and musl version at the same makes your test inconclusive

asbai, 2022-09-18:
OK, We completed a consistent test and initially analyzed the main factors that affect the performance of mallocng.

We built two different new tests, both running on the node 2 (w/ musl 1.2.3).

TEST1:
```C
const size_t nBASE = 10485760;
void* pTest = realloc(NULL, nBASE);

for (UINT i=0; i<10485760; ++i)
{
	pTest = realloc(pTest, nBASE+i);
}
```

and TEST2:
```C
const size_t nBASE = 10485760;
void* pTest = realloc(NULL, nBASE);

for (UINT i=0; i<10485760; ++i)
{
	malloc_usable_size(pTest ); // <---- the ONLY difference line
}
```

It can be seen that there is **only one line** of code difference between TEST1 and TEST2 (the `realloc` call in TEST1 is replaced by a `malloc_usable_size` call in TEST2).

The result is that TEST1 execution consumes **2.4s**, while TEST2 consumes **1.6s**.

Considering that realloc itself contains a complete malloc_usable_size (refer to [here](https://git.musl-libc.org/cgit/musl/tree/src/malloc/mallocng/realloc.c?h=v1 .2.3#n12) and [here](https://git.musl-libc.org/cgit/musl/tree/src/malloc/mallocng/malloc_usable_size.c?h=v1.2.3#n7)), **So actually most (66.7%) of the `realloc` time is spent doing `malloc_usable_size`.**

It's very unusual because `malloc_usable_size` or the equivalent API is very efficient in the implementations we often use. Such as: glibc, tcmalloc, msw crt (_msize), mac os x (malloc_size) and musl 1.1 etc.

In implementations such as: glibc, tcmalloc, msw crt (_msize), mac os x (malloc_size), and musl 1.1, even on low-end embedded processors, the consumption of `malloc_usable_size` per 10 million calls is mostly not more than a few hundred milliseconds.

For example, on Node 1 (MT7621), 10485760 (over 10 million) times calls to `malloc_usable_size` to musl 1.1 also took only **288ms**.

In addition, this very slow slab size acquisition algorithm also needs to be called every time free (see [here](https://git.musl-libc.org/cgit/musl/tree/src/malloc/mallocng/free .c#n105)). So we believe it should be the main reason for malloc/free and realloc performance degradation in version 1.2.

If we can improve its speed and make it close to implementations like tcmalloc (tcmalloc can also accurately return the size of the size class to which the chunk belongs), it should significantly improve the performance of mallocng (at least in single-threaded scenarios) .

jow-, 2022-09-19:
I suggest to continue this discussion upstream at https://musl.libc.org/support.html - OpenWrt ships musl libc allocators as-is and will not deviate from upstream defaults.

closed 2022-09-19 by jow-