openwrt/openwrt #10759: dnsmasq: require update to support nftset

lvqier, 2022-09-19:
Sicnce 22.03.0, iptables was replaced by nftables and ipset in firewall was migrated to nftset. But current version of dnsmasq does not support nftset. There's no approach to make things work like filtering traffic with IP sets by DNS.

First version of dnsmasq with nftset support is 2.87, which is still in release candidate.

ChemistScript, 2022-09-20:
iptables-nft is not working?

dave14305, 2022-09-21:
> iptables-nft is not working?

dnsmasq cannot populate nftables sets until 2.87 is released with this commit:
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=47aefca5e405b4b6627ef952fdc42e61b1baa770;hp=981fb037102306a4ca683f14c8469db4d5e27233

ChemistScript, 2022-09-21:
I thought iptables-nft able to populate nft sets.

harvey-git, 2022-09-24:
In order to solve the same problem, I wrote a script and put it in crontab to run, the following is the content of the script

`#!/bin/sh 
i=1;
numset=$((`ipset list|wc -l`-8));
while [ $i -le $numset ] ;
 do ipset list|tail -$i|head -1|sed 's|^|{|'|sed 's|$|}|'|xargs nft add element apric gfwzone  ;
i=$((i+1)) ; 
done
`

KeyofBlueS, 2022-09-26:
dnsmasq 2.87 has been released https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=tag;h=refs/tags/v2.87 :+1:

xabolcs, 2022-09-26:
Related PR: #10820

981213, 2022-11-16:
dnsmasq is already bumped to 2.87. Closing.

closed 2022-11-16 by 981213