openwrt/openwrt #10760: ddns service not starting on reboot

shigedayo, 2022-09-19:
sorry I am writing using Google Translate.

I can start the process manually from web interface.
But it doesn't start automatically on reboot.

/etc/init.d/ddns
boot() {
return 0
}

I added
/etc/init.d/ddns
boot() {
/usr/lib/ddns/dynamic_dns_updater.sh -- start
return 0
}

Thank you

PaTTeeL, 2023-04-30:
yes, my openwrt seems with the same problem!
```
root@OpenWrt:~# service ddns enable
root@OpenWrt:~# ls -l /etc/rc.d/*ddns*
lrwxrwxrwx    1 root     root            14 May  1 16:05 /etc/rc.d/K10ddns -> ../init.d/ddns
lrwxrwxrwx    1 root     root            14 May  1 16:05 /etc/rc.d/S95ddns -> ../init.d/ddns
root@OpenWrt:~# uci show ddns                                                                                                               
ddns.global=ddns                                                                                                                            
ddns.global.ddns_dateformat='%F %R'                                                                                                         
ddns.global.ddns_loglines='250'                                                                                                             
ddns.global.ddns_rundir='/var/run/ddns'                                                                                                     
ddns.global.ddns_logdir='/var/log/ddns'                                                                                                     
ddns.global.cacert='/etc/ssl/certs'                                                                                                         
ddns.freedns_afraid_org=service                                                                                                             
ddns.freedns_afraid_org.service_name='afraid.org-basicauth'                                                                                 
ddns.freedns_afraid_org.use_ipv6='1'                                                                                                        
ddns.freedns_afraid_org.enabled='1'                                                                                                         
ddns.freedns_afraid_org.lookup_host='mydomain.my.to'                                                                                         
ddns.freedns_afraid_org.domain='mydomain.my.to'                                                                                              
ddns.freedns_afraid_org.username='username'                                                                                                  
ddns.freedns_afraid_org.password='password'                                                                                               
ddns.freedns_afraid_org.use_https='1'                                                                                                       
ddns.freedns_afraid_org.cacert='/etc/ssl/certs'                                                                                             
ddns.freedns_afraid_org.ip_source='interface'                                                                                               
ddns.freedns_afraid_org.ip_interface='pppoe-wan'                                                                                            
ddns.freedns_afraid_org.interface='pppoe-wan'                                                                                               
ddns.freedns_afraid_org.use_syslog='2'                                                                                                      
ddns.freedns_afraid_org.check_unit='minutes'                                                                                                
ddns.freedns_afraid_org.force_unit='minutes'                                                                                                
ddns.freedns_afraid_org.retry_unit='seconds'
```
And when router starts, `logread -e ddns` shows nothing at all.
I used to add `* * * * * /etc/init.d/ddns start` and the ddns service restarts every minutes which makes the router crash quickly. Now I'm using `/etc/init.d/ddns start` in /etc/rc.local to make the ddns service work.

System Info
`
Phicomm K3
ARMv7 Processor rev 0 (v7l)
bcm53xx/generic
OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-23.069.50944-cd8bea9
5.10.161
`

Arnie97, 2024-07-26:
After some investigation, I find that the behavior was deliberate, and was introduced in commit https://github.com/openwrt/packages/commit/e5613b306f890da5ca2e6401bc587d67815dfc93.  The DDNS services are expected to be started by hotplug events, but for some reason it doesn't work on my device (OpenWrt 22.03.5 r20134-5f15225c1e, ramips-mt7620-phicomm_k2).

So just revert this patch by delete the following function from `/etc/init.d/ddns`:

```
boot() {
        return 0
}
```

mkrle, 2024-08-22:
> but for some reason it doesn't work on my device
Good find, helped me a lot, thanks. Maybe doublecheck your interface parameter is correct (regarding hotplug not starting ddns).

shiromiso, 2024-10-07:
Hello all, any news on this? Still happens for me on ASUS TUF-AX4200 23.05.5.
I am hesitant to modify any files manually as those changes would be overwritten by upgrades.

Perhaps it is related to the default PPPoE config? I am using PPPoE for IPv4 and DHCPv6 client for IPv6.
DDNS is IPv6 only.

Thanks

qianxuu, 2024-12-06:
I encountered the same issue. Based on my testing, the IP address source in the advanced settings needs to remain set to the default "Network." If it is set to "Interface," DDNS will not start automatically on boot. After reverting it to the default setting, the auto-start on boot works normally.

mstigge, 2024-12-29:
I just ran into this as well, and the suggestion of @mkrle was important. Even more so, LuCI offers `eth1` and `@wan`, but what you _actually_ need to configure at `ddns.whateverhere.interface` is `wan`. Only then did things start to work automatically upon restart. Notice the absence of `@`. That's not available in LuCI and you need to do that by hand. (Perhaps something to be improved in LuCI at some point, alternatively in the ddns script.)

christaikobo, 2025-01-16:
For me, when using `wan` or `@wan`, DDNS just reports `Device "@wan" does not exist.`

christaikobo, 2025-01-16:
OK the situation seems quite bizarre to me.

In the `Advanced Settings` tab, you are allowed to pick a `interface`, which corresponds to `ip_interface` in the config file. Then there is `Event Network` next line which corresponds to `interface` in the config file and it is just automatically set by the script to be the same as the former. What a tongue twister. 

When I pick `wan` or `@wan` as interface in LuCI, DDNS just reports device `wan` or `@wan` does not exist, because it indeed does not exist.  `wan` or `@wan` is an interface, not a device. 

The correct way to do it is manually set `ip_interface` (corresponds to `interface` in LuCI) to `eth1` and `interface` (corresponds to `Event Network` in LuCI) to `wan` in the config file, only then it will work and continue to work automatically after reboot. 

I remember encountering this several years back and I didn't bother to figure it out. Now I finally figure it out, and honestly it feels stupid to me. Or maybe I'm stupid and missed something glaringly obvious, in which case please enlighten me!

mstigge, 2025-01-20:
Thanks for the background.

As "IP address source" I didn't use "Interface" at the time but "URL", so `ip_interface` wasn't relevant.

So, what I was referring to 3 comments ago was the "Event Network" setting, for which I needed to use the `wan` alias (manually configured in `/etc/config/ddns`). I suppose that when the system comes up, the bringup of the `wan` interface would then trigger the `ddns_update` script to run. That's anyway the earliest time for that to make sense since any earlier and you couldn't determine your public IP anyway.

I _think_ that I also have a setup somewhere which is using the "Interface" setting for address source with `eth1` for both `interface` as well as `ip_interface`. I haven't tried the reboot behavior in a while though.

regunakyle, 2025-05-18:
Bumping this as this just occurred to me.
BTW, the `START DDNS` button seems to do nothing, only the `RESTART DDNS` button starts the DDNS service properly.

tristan-k, 2025-06-04:
Can someone post an example `/etc/config/ddns` which will fix the issue and explain how? I'm also struggeling with the same issue.

juronja, 2025-07-15:
My ddns also does not autostart. And from the [sample config](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/samples/ddns.config_sample) it looks like my config should work with network as source, but it does not.

I tried with the network as the source and the interface as source. See the config below.

I am using the TP-Link TL-WR1043ND v2 with the default wan interface (didn't tweak it). But in my tplink model, the wan is attached to Device: Software VLAN: "eth0.2" so I had to config this via uci.

Do I miss something? I don't know what this hotplug mechanic is, but where could I check if the ddns really listens to wan up/down events?

```
config service 'duckdns'
        option service_name 'duckdns.org'
        option use_ipv6 '0'
        option domain 'redacted'
        option username 'redacted'
        option password 'redacted'
        option ip_source 'interface'
        option use_syslog '2'
        option check_interval '20'
        option check_unit 'minutes'
        option force_interval '72'
        option force_unit 'hours'
        option interface 'wan'
        option lookup_host 'redacted'
        option enabled '1'
        option ip_interface 'eth0.2'

config service 'cloudflare'
        option service_name 'cloudflare.com-v4'
        option use_ipv6 '0'
        option domain 'redacted'
        option username 'Bearer'
        option password 'redacted'
        option ip_source 'network'
        option use_syslog '2'
        option check_unit 'hours'
        option force_unit 'hours'
        option interface 'wan'
        option lookup_host 'redacted'
        option use_https '1'
        option cacert '/etc/ssl/certs'
        option enabled '1'
        option check_interval '2'
        option force_interval '72'
        option ip_network 'wan'
```

sqliuchang, 2025-07-22:
This bug has been fixed recently.
[Why OpenWrt DDNS does not start on boot](https://7ji.github.io/networking/2025/06/05/why-openwrt-ddns-does-not-start-on-boot.html)
https://github.com/openwrt/luci/commit/e6b26d1e07c41dfd520a2e8ef67ec14b6118f288