openwrt/openwrt #10761: configure maximum A-MSDU size on openwrt

kamran-nishat, 2022-09-19:
We have Dlink-822 APs with OpenWRT on it. We want to restrict the maximum A-MSDU size. It is an option in hostapd.conf. But in OpenWRT this file is generated using a script. 

I believe there should be an easier way to configure these hostapd options on OpenWRT.

namidairo, 2022-09-20:
https://openwrt.org/docs/guide-user/network/wifi/basic#n_capabilities

Already configurable via UCI. Not really a whole lot of options as hostapd just looks for `[MAX-AMSDU-7935]`.

Unless you meant MPDU length, which is also configurable.

kamran-nishat, 2022-09-20:
Thanks for your reply. Yes, we want to restrict the length of MPDU to a certain max value. How can I configure that?

MartijnPW, 2023-03-08:
I was wondering the same thing... And I'm quite a beginner when it comes to OpenWRT. 
Does anyone know how to easily configure MPDU length? Thanks!