openwrt/openwrt #10771: Only first configured wifi network gets dhcp address - config sequence is important

otrebor67, 2022-09-20:
I have four configured wifi networks which serve different purposes and have each their own vlan and ip address range.
Only the first configured wifi actually serves the ip adress by dhcp. All others won't get an address and therefore disconnect.
The effect is independent of the default network setting, but the sequence of the wireless config file is relevant.
I see the clients connecting on the LuCi wireless page, but they disconnect as they do not get a valid ip address.
I have tried shuffling around the config blocks of each networks settings without changing anything. In a reproducible way, only the first one works. No change on the client side is necessary. Once connected, the client network flow is as expected.
I'm using 22.03 (new installation). This is a regression from 19.07 as this has worked before (i've skipped 21.02).
[dhcp.txt](https://github.com/openwrt/openwrt/files/9611121/dhcp.txt)
[wireless.txt](https://github.com/openwrt/openwrt/files/9611122/wireless.txt)

jow-, 2022-09-21:
Please provide your /etc/config/network file as well

otrebor67, 2022-09-21:
sure.
[network.txt](https://github.com/openwrt/openwrt/files/9619476/network.txt)
thanks for the fast response.

otrebor67, 2022-09-22:
I did as you suggested. The symptoms are unchanged.
[ifstatus.wlan194.txt](https://github.com/openwrt/openwrt/files/9629174/ifstatus.wlan194.txt)
[ifstatus.wlan196.txt](https://github.com/openwrt/openwrt/files/9629175/ifstatus.wlan196.txt)
[ifstatus.wlan197.txt](https://github.com/openwrt/openwrt/files/9629176/ifstatus.wlan197.txt)
[ifstatus.wlan198.txt](https://github.com/openwrt/openwrt/files/9629177/ifstatus.wlan198.txt)
[ifstatus.wlan199.txt](https://github.com/openwrt/openwrt/files/9629178/ifstatus.wlan199.txt)

patrakov, 2022-09-23:
This looks like a duplicate of https://github.com/openwrt/openwrt/issues/7795, even though the symptoms are different (they have DEVICE_CLAIM_FAILED, you don't).

Please create an empty bridge for each of the WiFi interfaces, and configure the static IP and DHCP on the bridge instead of the WiFi device itself. In Wireless Configuration, associate each wireless device with the bridge created for it. This way, the startup race between creating a WiFi interface and starting to use it is avoided.

otrebor67, 2022-09-25:
I'm not sure if I fully understood what you suggested, but I gave it a try.
The bottom line is, that it makes things actually worse as no wifi interface gets an ip address from dhcp.
[brctl-show.txt](https://github.com/openwrt/openwrt/files/9640598/brctl-show.txt)
[dhcp.txt](https://github.com/openwrt/openwrt/files/9640599/dhcp.txt)
[ifstatus.wlan194.txt](https://github.com/openwrt/openwrt/files/9640600/ifstatus.wlan194.txt)
[ifstatus.wlan196.txt](https://github.com/openwrt/openwrt/files/9640601/ifstatus.wlan196.txt)
[ifstatus.wlan197.txt](https://github.com/openwrt/openwrt/files/9640602/ifstatus.wlan197.txt)
[ifstatus.wlan198.txt](https://github.com/openwrt/openwrt/files/9640603/ifstatus.wlan198.txt)
[ifstatus.wlan199.txt](https://github.com/openwrt/openwrt/files/9640604/ifstatus.wlan199.txt)
[network.txt](https://github.com/openwrt/openwrt/files/9640605/network.txt)
[wireless.txt](https://github.com/openwrt/openwrt/files/9640606/wireless.txt)

gpsa, 2023-03-07:
> This looks like a duplicate of #7795, even though the symptoms are different (they have DEVICE_CLAIM_FAILED, you don't).
>
> Please create an empty bridge for each of the WiFi interfaces, and configure [...]

This worked like a breeze! 

Just for helping whoever that's struggling with that, here's how to configure it:

At `Network -> Interfaces -> Devices (tab) -> Add device configuration…`
[image]

At `Network -> Interfaces -> Interfaces (tab) -> Select your new Interface (or create one)` and under `Device`, select the newly created bridge interface:
  
[image]

At `Network -> Wireless -> [Your wifi] -> Interface Configuration -> General Setup -> Network`, select the interface you edited/created above.
[image]

mtompkins, 2023-05-06:
Thank you all for the above.

Hurricos, 2023-10-28:
> This looks like a duplicate of #7795, even though the symptoms are different (they have DEVICE_CLAIM_FAILED, you don't).
>
> Please create an empty bridge for each of the WiFi interfaces, and configure [...]

This solution will not work for interfaces which cannot be bridged -- for example, STA or IBSS.

Hurricos, 2023-10-28:
And, the real issue comes out. OP's issue was likely different from mine -- maybe related to the behavior of their dnsmasq? -- but *my* case was that directly described by Jo [above](https://github.com/openwrt/openwrt/issues/10771#issuecomment-1254214568) documented [here](https://openwrt.org/docs/guide-user/network/network_configuration), section "Common options", option `device`:

> :!: Do not specify wireless interfaces as their names and behavior can be dynamic and unpredictable, instead assign wireless interfaces to bridges using the network option in [wireless configuration](https://openwrt.org/docs/guide-user/network/wifi/basic#wi-fi_interfaces).
This option may be empty or missing if only a wireless interface references this network or if the protocol is pptp, pppoa, 6in4, etc.

In my case, pointing at the `device` in the `/etc/config/network` `interface` meant that netifd failed waiting for my very slow IBSS VIF bringup:

```
config interface 'MESH'   
        option proto 'static'
        option netmask '255.192.0.0'
        option device 'mesh0'       
        option ipaddr '100.78.68.27'
```
        
Removing the 'mesh0' wifi-iface fixes my issue. I suspect netifd then waits for the wifi-iface bring-up to ask it to configure the interface, not sure. We'll see how it goes.

Hurricos, 2023-10-29:
Reordering the interfaces solved my issue. Unfortunately it seems it all comes down to timing.

danpawlik, 2023-10-29:
@Hurricos I saw similar issue described here: https://github.com/openwrt/openwrt/issues/13598 . Is it related?