openwrt/openwrt #10777: BUG? dnsmasq 2.86 crash (Selective DNS forwarding) formalism

pesa1234, 2022-09-21:
Dear all, on previous version of dnsmasq 2.85 for Selective DNS forwarding I wrote this line on /etc/config/DHCP

`list server '/amazonaws.com/llnwi.net/aiv-cdn.net/footprint.net/aiv-delivery.net/amazonvideo.com/amazon.com/cloudfront.net/media-amazon.com/primevideo.com/a2z.com/amazon.fr/amazon.it/paypal.com/1.1.1.1'`

Dnsmasq 2.86 crash.

It is necessary to split example:
`list server '/browserleaks.com/1.1.1.1'
	list server '/browserleaks.org/1.1.1.1'
	list server '/amazonaws.com/1.1.1.1'
	list server '/llnwi.net/1.1.1.1'
	list server '/aiv-cdn.net/1.1.1.1'
	list server '/footprint.net/1.1.1.1'
	list server '/aiv-delivery.net/1.1.1.1'
	list server '/amazonvideo.com/1.1.1.1'
	list server '/amazon.com/1.1.1.1'`

In this way dnsmasq 2.86 don't crash

ynezz, 2022-09-22:
@ldir-EDB0 any idea?

ynezz, 2022-09-22:
> `list server '/amazonaws.com/llnwi.net/aiv-cdn.net/footprint.net/aiv-delivery.net/amazonvideo.com/amazon.com/cloudfront.net/media-amazon.com/primevideo.com/a2z.com/amazon.fr/amazon.it/paypal.com/1.1.1.1'`

@pesa1234 I've just tried to reproduce this in Qemu x86/64 running 22.03.0 release running Dnsmasq version 2.86, but it doesn't crash here.

```
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain amazonaws.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain llnwi.net
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain aiv-cdn.net
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain footprint.net
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain aiv-delivery.net
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain amazonvideo.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain amazon.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain cloudfront.net
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain media-amazon.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain primevideo.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain a2z.com
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain amazon.fr
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain amazon.it
Thu Sep 22 07:14:54 2022 daemon.info dnsmasq[1]: using nameserver 1.1.1.1#53 for domain paypal.com
```

On what platform/device are you experiencing this issue? Are you using official firmware image or custom build one?

pesa1234, 2022-09-22:
Hi, official firmware image with dnsmasq-full and ipset with this package https://docs.openwrt.melmac.net/pbr/

ynezz, 2022-09-22:
> Hi, official firmware image 

Ok, can you tell us, which one is that exeactly?

> with dnsmasq-full and ipset with this package https://docs.openwrt.melmac.net/pbr/

I'm not able to reproduce it even with dnsmasq-full:

```
root@OpenWrt:/# opkg list-installed  | grep dnsmasq
dnsmasq-full - 2.86-14
```

> ipset with this package https://docs.openwrt.melmac.net/pbr/

We need some isolated and minimal reproducer first, can you try to provide that? It seems, that sole dnsmasq UCI setting is not enough for triggering this issue.

pesa1234, 2022-09-22:
`root@WRT1900ACS:~# opkg list-installed | grep dnsmasq
dnsmasq-full - 2.86-14
`
Issue:

DNSMASQ ipset settings and selective DNS forward setting with that formalist
https://forum.openwrt.org/t/vpn-policy-based-routing-web-ui-discussion/10389/1780

Please have a look here, solution:
https://forum.openwrt.org/t/vpn-policy-based-routing-web-ui-discussion/10389/1792

ynezz, 2022-09-22:
> Please have a look here, solution:

Ok, so it's not crashing, dnsmasq probably wont start with some invalid configuration generated by 3rd party package pbr:

```
Wed Sep 21 02:59:48 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 9 seconds since last crash
```

So IIUC this is not a bug in dnsmasq and we can close this.

closed 2022-09-22 by ynezz

pesa1234, 2022-09-25:
@ynezz if you have adblock package installed... 

With latest adblock 4.1.4 with dnsmasq-full 2.86 on openwrt 22.03

Please try these settings.
add a domain in whitelist example `amazon.com`
then in dhcp and DNS option add the same domain on DNS forwardings

`list server '/amazon.com/1.1.1.1'`

Reload Adblock, restart dnsmasq.

Open browser and go to amazon.com

After some second you have that `Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 50 seconds since last crash`