openwrt/openwrt #10781: No WiFi in Banana Pi BPI-R3

reddwarf69, 2022-09-22:
Booting from a SD card with https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz (607c6240db596f7154f3e9598a8157d894fd60bd44db65c092fbfd7c8199f1d7 | 20955.0 KB | Wed Sep 21 16:58:55 2022... commit 1722e23ffcf9038114142a2129c29eb3cdec8ff9 if I understood it right) into a Banana Pi BPI-R3 v1.1, I get this error message at boot
```
[   10.127699] mt7986-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20220113162701a
[   10.127699] 
[   10.247829] mt7986-wmac 18000000.wmac: WM Firmware Version: ____000000, Build Time: 20220113162756
[   10.324188] mt7986-wmac 18000000.wmac: WA Firmware Version: DEV_000000, Build Time: 20220113163034
[   30.565222] mt7986-wmac 18000000.wmac: Message 00000002 (seq 4) timeout
[   30.571834] mt7986-wmac 18000000.wmac: Failed to start WA firmware
[   30.578095] mt7986-wmac: probe of 18000000.wmac failed with error -110
```
And I end up without WiFi interfaces.

eckinga, 2022-09-22:
Hello,

same snapshot same problem.

reddwarf69, 2022-09-29:
https://github.com/openwrt/mt76/issues/702

closed 2022-09-29 by reddwarf69