openwrt/openwrt #10784: 5GHz on NBG6716 on ath79 (regression of #655)

fundef1, 2022-09-22:
Hi, 
the QCA988x has been broken on (some) Zyxel NBG6716 since 22.03 / 21 , worked fine up until 19 (ar71xx).

the root cause is that the flash doesn't contain any cal-data, but is extracted anyway.
* flash layout doesn't match the contents: i.e. mtd partition contains garbage
* (and in my case it reports only half the physically available size)
This isn't the case for all NBG6716s  there are multiple variants out there.

Previously this was worked-around by checking the first few bytes and then not extracting cal-data but letting the driver default to using the cal-data on the PCI-E card.
see [#655](https://github.com/openwrt/openwrt/pull/655)

As a temporaty workaround: with 22.03 it suffices to delete ath10k hotplug script, and the generated cal-data for the QCA988x to work fine again.

fundef1, 2023-05-25:
Hi,  yes the cal file is present, (as well as the hotplug script creating it)

both files here: [nbg6716.tar.gz](https://github.com/openwrt/openwrt/files/11568699/nbg6716.tar.gz)

ps: also see this thread on the [forum](https://forum.openwrt.org/t/17-01-1-2-3-qca988x-ath10k-5ghz-firmware-crashed-zyxel-nbg6716/1911/34)  the most likely conclusion was that this type of NBG6716 doesn't have the cal data in a partition but on the wifi card itself.  In any case the link in the forum points to a dump from mtd2 which did not contain any calibration data.

Djfe, 2023-05-26:
yes, it contains nothing.
Please send me the output of
`cat /proc/mtd`
and a log from right after booting
`logread`

Djfe, 2023-05-26:
So I uploaded a patch that you can test :) #12732
If you need me to compile an image for you, then please tell me.

Out of curiosity: Do you own the variant with the 128MiB or the 256MiB flash chip?

fundef1, 2023-05-26:
My variant is the one with the 256MiB flash.

```
root@AP3:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "zyxel_rfsd"
mtd1: 00200000 00020000 "romd"
mtd2: 00100000 00020000 "header"
mtd3: 07b00000 00020000 "firmware"
mtd4: 00400000 00020000 "kernel"
mtd5: 07700000 00020000 "ubi"
mtd6: 00040000 00010000 "u-boot"
mtd7: 00010000 00010000 "u-boot-env"
mtd8: 00010000 00010000 "art"
mtd9: 00fa0000 00010000 "nbu"

```
logread right after boot : [logread.txt](https://github.com/openwrt/openwrt/files/11574948/logread.txt)
Note: this is with the script/cal file removed so 5Ghz is enabled.

if you want me test the patch you'll need to compile an image.
But if i'm not mistaken the patch only affects the hotplug script... 
it would be much less hassle for me (and for you) to just test that file, no?

that said, getting the full flash size back would be appreciated ;-)