openwrt/openwrt #10790: Superfluous flow control on system port

brada4, 2022-09-23:
OpenWrt 22.03.0, r19685-512e76967f
TP-Link Archer C7 v5

Hardware pause is enabled on "system" switch port and the eth0 shows it actually does pushback to LAN at WAN at once, freshly rebooted, practically no load. Other interfaces are gigabit, WAN is 150Mbps, but no pause negotiated  thus it originates in switch. Endpoint TCP stacks would be much better at handling even packet drops in place of unintentionally bidirectional PAUSE.
ethtool -a is not available, so supposedly NIC driver ag71xx should at least in this condition default to no pauses, or better do it totally to match rest of network world.

```
Port 0:
        link: port:0 link:up speed:1000baseT full-duplex txflow rxflow 
Port 1:
        link: port:1 link:up speed:1000baseT full-duplex auto
```

```
NIC statistics:
     Tx Byte: 2395596
     Tx Pause Frames Honored: 156
```

brada4, 2022-09-28:
I will not edit original post, I tried to monitor `ethtool -S eth0` and it seems that those stats fields just show random values in all non-zeroes, or they roll over doing just one SSH session and trying to play back output every second resets them.

brada4, 2022-10-14:
If I set up different switch vlans for different ports iperf -c does well, 600-700Mbit just like over switch alone (probably running longer than default test would give closer full saturation, CPU usage like 15% while load slightly over 1.0 ) but --bidir makes it down to 20Mbits and no CPU usage visible. Same running iperf one piece on the router - output gets clamped to 20-some Mbit with --bidir. What I suspect is that flow control implemented in switch chip makes it act like half-duplex - i.e having traffic for the port it adds some pause frames for no reason.

hmh, 2022-12-28:
When auto-negotiation breaks on fast-ethernet and gigabit-ethernet, it often ends up working as if it was 100-HD (fast-ethernet, half-duplex).  So, please test for that, see if you can get to the CPE's status page and look at what it says there.  Or try to replace the gigabit ethernet cable between the openwrt router and the CPE.

You want to be sure that's not the source of so many pause frames.

Once you did that, if it is really the pause frames, it would be nice to send an email to the development ML, and make sure to mention it is an issue with the kernel driver (and its file, etc) since you seem to have tracked down things further in #11372...    Otherwise, the bug might linger here on github with no action :-(

brada4, 2022-12-28:
CPE does not support any flow control. It is on CPU port, switch port 0 and beams to port on their own wlans unrelated to vlan 1/2 and isolated by all means

hmh, 2022-12-29:
Does the *CPE* report that the connection is 1000BASE-T ?  (I mean reading that information in the CPE console/web page, *not* in the openwrt router) 

That would rule out any half-duplex shenanigans due to broken auto-negotiation...

brada4, 2022-12-30:
Problem appears putting physical ports in separate VLAN-s and bridging through CPU port.
From irrelevant things - i am pretty sure PON remains ovesubscribed if I disconnect uplink port and experience slownes with no CPE involves.

brada4, 2022-12-30:
i.e:
```
[router] eth0.1 - vlan1 - "internet"
              br-lan  - vlan2 - "intranet", wifis
             [ifconfig up]     - vlan4 - physical4 -> this gets pause frames counted when traffic is present below
             [br-test]            - vlan5 - physical2
             [br-test]            - vlan6 - physical3
```

brada4, 2023-01-08:
Test on arbitrary port with e1000e card in HP laptop - if I enable flow control linux side, respective swconfig port reflects it, as well as disabling flow control.
When I connect card directly to CPE - flow control does not get negotiated, tried all combinations, with resets between, CPE menu is in cloud, it just shows some basic counters for optical wire's other end, nothing past some siemens and version number from this side of CPE.

I cannot find a way to change mii anounce bits on the switch like `ethtool -s eth0 autoneg on advertise 0x020` i.e i strip out flow bits on CPU switch port.
If I remove the flow control bits from CPU side as above - ethtool eth0 shows all flow control is off, while in ethtool -S the "pause frames honored" number nibbles around (others increment logically) and switch stays reporting 
`port:0 link:up speed:1000baseT full-duplex txflow rxflow `

brada4, 2023-01-24:
OpenWRT 19.07 - 200Mbps, bufferbloat grade A no remarks at waveform
OpenWRT 20, 22 - 120Mbps, bufferbloat grade B 3 remarks on right side
OpenWRT 22 ethtool -G eth0 tx 8 120Mbps, bufferbloat grade A no remarks

brada4, 2023-04-22:
It is fq_codel applied on physical port , i.e all VLAN-s together, not only to upstream VLAN.
My internet has good latency characteristics, setting qdisc to noqueue makes "switching via CPU" to 500Mbps, no matter flow control setting.