openwrt/openwrt #10791: Static route through auto-detected network gateway created incorrectly

patrakov, 2022-09-23:
This is on OpenWRT 22.03 on Linksys E8450.

I have an LTE connection as a backup, with NCM protocol:

```
config interface 'lte'
	option proto 'ncm'
	option pdptype 'IPV4V6'
	option ipv6 'auto'
	option peerdns '0'
	option metric '90'
	option apn 'internet'
	option username 'lte'
	option password 'lte'
	option delay '5'
	option device '/dev/ttyUSB1'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	list dns '2001:4860:4860::8888'
	list dns '2001:4860:4860::8844'
```

I want a particular host to be routed through LTE no matter what. I tried to create a static route, and I have read this help in Luci about the Gateway field:

> Specifies the network gateway. If omitted, the gateway from the parent interface is taken if any, otherwise creates a link scope route. If set to 0.0.0.0 no gateway will be specified for the route

OK, so I want the gateway to be taken from the parent interface, because I don't know what it would be after a reboot due to the dynamic nature of LTE. So I leave this empty.

This creates the following entry:

```
config route
	option target '172.94.88.1/32'
	option interface 'lte'
```

However the created route is:

```
172.94.88.1 dev wwan0 proto static scope link metric 90 
```

...which is a link-scope route, which says "arp for 172.94.88.1 on wwan0, don't use any gateway". Which is wrong, according to the help text. The expected result is:

```
172.94.88.1 via 100.108.220.254 dev wwan0 proto static src 100.108.220.253 metric 90 
```

...with IP addresses to be used instead of 100.108.220.253 and 100.108.220.254 being autodetected from the existing default route.

I have also tried `option interface 'lte_4'`, same effect.

brada4, 2022-09-24:
On the serial line you dont have to specify gateway, what hits the wire reaches other end.

patrakov, 2022-09-24:
Correct - but wwan0 is not a serial PPP line. It is a virtual Ethernet interface emulated by the modem.

sdanfa, 2024-05-18:
Hi. This is a serious bug that is affecting all releases. May you fix the bug? Thank you.

eldorel, 2024-05-23:
I just ran into this bug as well. 
Seeing that it's been open for a year and a half, I assume I should just figure out another option for this project.

microka, 2024-08-31:
Why hasn't this been solved?

eldorel, 2024-11-03:
> Why hasn't this been solved?

over 2700 issues reported in this repository, and only 41 members on the project == long wait for replies/fixes for some issues. 

This one is a little bit obscure (not massively though), so I expect it to persist until more SOHO routers reliably support redundant/loadbalanced connections and connection/client pinning.

BK8000L, 2025-03-28:
still in 24.10.0, very easy to reproduce
[image]
wan gateway ip is 192.168.1.254 which is not added in the actual route

yevgeny-z, 2026-01-16:
Faced with same issue.

Workaround for me is `network.wan.customroutes` for DHCP proto interfaces
https://openwrt.org/docs/guide-user/network/ipv4/configuration#protocol_dhcp
https://github.com/openwrt/openwrt/blob/openwrt-24.10/package/network/config/netifd/files/lib/netifd/dhcp.script#L33-L35

```
uci set network.wan.customroutes="<ip1>/32 <ip2>/32 <ip3>/32"
uci commit network
ifup wan
```